HiddenRamblings/TagMo #390: kind of a n00b here, but what does this mean?

Sel1432, 2022-01-08:
"Attempt to invoke virtual method 'void com.hiddenramblings.tagmo.nfctech.NTAG215.connect()' on a null object reference"

Does this mean that the chips i'm trying to use are no good, incompatible, or what? I bought ones specifically for amiibo stuff...

AbandonedCart, 2022-01-08:
It could mean a few things. Incompatible tags, not holding them close enough or long enough to the NFC antenna, or the device doesn't support reading or writing the technology type.

Sel1432, 2022-01-08:
It will briefly flash mifare classic detected but then immediately give that error after doing so.

AbandonedCart, 2022-01-08:
Oh, then likely the wrong tags. You need mifare ultralight (NTAG215).

closed 2022-01-09 by AbandonedCart