HiddenRamblings/TagMo #395: Pressing delete all banks no longer progresses action

Miss1sle, 2022-01-19:
Newest release no longer progresses past the screen where you press the button to erase all banks. 

https://user-images.githubusercontent.com/97257924/150206351-b0612198-3c07-4928-8822-9916b524fe9a.mp4

[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7900275/tagmo_logcat.txt)

closed 2022-01-19 by AbandonedCart

AbandonedCart, 2022-01-19:
I assume the last fix resolved everything. It was automatically closed by the commit.

Miss1sle, 2022-01-20:
Yeah, it all worked. Though it only got to about 12 of the 90 before the N2 crashed and stopped saying it was a tag.

Putting it back on its regular base on a pc fixed it.

Maybe tagmo deletes the banks too fast and needs a pause between each bank?

AbandonedCart, 2022-01-20:
Did you have empty banks?

Miss1sle, 2022-01-20:
I did not. They were all full.

I've deleted the rest by hand as I didn't want to risk a permanent bricking of the N2.

AbandonedCart, 2022-01-20:
It won't brick. That's why it stops working in TagMo. Safety feature.

Miss1sle, 2022-01-20:
I'm happy to give it another go if you can give me a test to do.  Number of banks, which you want active, if you want any empty.

commit 183d32a4 (2022-01-19): Add missing category for erasing banks

Fixes #395

Apparently this got removed as a duplicate