HiddenRamblings/TagMo #396: Error code when using bulk delete

Miss1sle, 2022-01-22:
Hey,

Didn't know if I should make a new case or open the old one.

Wrote 20 banks and it worked great.
Set bank 1 to be active then tried the bulk erase, paused saying erasing bank 1 but then displayed the following error.

[image]
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7918645/tagmo_logcat.txt)

Miss1sle, 2022-01-22:
The tag will no longer read on TagMo, showing the above error.

Placing it back on the base with the official software restores its functionality.

AbandonedCart, 2022-01-24:
It simply can't delete the active bank. For some reason, "normal" mifare processing can't handle that and the way they do it is unstable. It would break virtually everything else. The best solution is to simply not delete that bank via bulk.

Miss1sle, 2022-01-24:
So far as I can see, bulk delete always starts at the top?

Can we make it so it starts at the last bank and works up? That way you can ae active bank to 1 and start at the bottom.

Or am I misunderstanding the issue?

AbandonedCart, 2022-01-24:
The changes were already pushed. It can't delete slot 0 during bulk

Miss1sle, 2022-01-24:
Oh ok good idea. Do I have to set the first slot as active manually?

I'll be home to give it a test shortly.