HiddenRamblings/TagMo #399: [N2 Elite] Set Bank Count button doesn't fit on screen

vaguerant, 2022-02-01:
When I pull up the bottom drawer, the Set Bank Count button below the number roller doesn't appear, presumably because it's hanging off the bottom of the screen.

[image]
[image]

My screen is fairly low-res, at 720\*1280, a 9:16 aspect ratio.

Oddly, I have been able to get the button to appear sometimes, but I don't understand how or why. Occasionally, when pulling up the drawer, the image of the current selected Amiibo would not display, leaving plenty of room for the Set Bank Count button. However, I can't seem to reproduce this at all now, no idea how I managed it previously.

AbandonedCart, 2022-02-01:
Are "View Scale Mods" enabled? If not, you may also want to try that. 3 dots -> Display options

vaguerant, 2022-02-01:
Thanks, just tried that option but it didn't help--it actually made everything slightly *larger*:

[image]

It wasn't any more or less usable, besides the bank count wheel looking a bit cramped down there.

AbandonedCart, 2022-02-02:
https://github.com/HiddenRamblings/TagMo/commit/64731e1424bc567b95838d68a946608cdb48f138 should size the image with the rest of the layout considered. When that view was made, it used the existing code for what is the large card from the browser. That was written at a time when specifying sizes was more common practice, which was also before issues like this one were as noticeable because everyone had the same size phones.

vaguerant, 2022-02-02:
Just updated to 64731e1, unfortunately not resolved.

View Scale Mods OFF:
[image]

View Scale Mods ON:
[image]

AbandonedCart, 2022-02-02:
That's unfortunate. It will take some time to decide the best course on this one. The back button results you found were not a fluke, though. If you don't have a bank selected, the default is to not include one in the menu. Not selecting a bank or going back from not viewing one will show just the menu. Fortunately, the option you are missing is not meant for single banks, so not having an image there shouldn't prevent it from working.

vaguerant, 2022-02-06:
Something else to note is that the app also supports rotation, so it's very easy to get even lower-height aspect ratios by putting the phone into landscape orientation. The N2 Elite menu as it exists now is clearly designed around vertical layouts. I don't know how/if you'd want to solve this, whether making the menu usable on even lower-height screens or perhaps just forcing vertical orientation when the N2 Elite menu is up.

AbandonedCart, 2022-02-06:
It's hard to both fit all the things people demand to see on one screen and still maintain support for these archaic 8-bit phones that a lot seem to throw at it.

vaguerant, 2022-02-06:
Yeah, that's understandable. Although, while I'm sure the tech inside is archaic, my phone itself is actually not that old. It was released in June 2020 and runs Android 10, so it's only about 18 months old. But also yeah, it's an extremely budget model that I got explicitly because it was cheap and had NFC support.

Back to the actual issue at hand, and maybe it's a bit hacky, but couldn't you basically just pillarbox the menu as much as is needed to have it fit the screen? Basically throw empty borders on the left/right (which are collapsed on more modern aspect ratio phones), so that the actual area taken up by the N2 Elite drawer is ~2:1 or whatever the aspect ratio is that it's roughly designed around? Obviously this would mean that the shorter your screen height, the smaller everything gets, but that seems reasonable and expected.

AbandonedCart, 2022-02-06:
I think making the picker horizontal would be easier

closed 2022-02-06 by AbandonedCart

vaguerant, 2022-02-06:
Not really improved by an appreciable amount, unfortunately. I can see the very top of the button and it is pressable, but if I didn't know what it was I probably wouldn't know it was a button.

[image]

AbandonedCart, 2022-02-06:
That was just the setup.

vaguerant, 2022-02-07:
Might be worth keeping the issue open, then.

AbandonedCart, 2022-02-07:
It's resolved, though.

vaguerant, 2022-02-08:
I'm on e8bff9e and it still looks identical to [this comment](https://github.com/HiddenRamblings/TagMo/issues/399#issuecomment-1027402048). The counter is sideways now, but that only allows a thin gray sliver of button to show at the bottom.

Also, while it's far less of an issue, pulling up the drawer in landscape mode it's completely unusable since the image is the *only* thing that fits there. The solution currently is probably just "don't use landscape", but the app doesn't enforce that in any way.

AbandonedCart, 2022-02-08:
That's not he newest commit.

vaguerant, 2022-02-08:
It's the latest on the experimental branch. Just switched back to the latest commit on master (79016a5) and it looks exactly the same.

AbandonedCart, 2022-02-08:
It can't possibly look the same. The buttons were removed.

vaguerant, 2022-02-08:
Sorry about the cropped screenshots, on a video call at the moment, but on current master:

[image]

[image]

AbandonedCart, 2022-02-08:
Not sure how that's possible. The buttons were removed.

vaguerant, 2022-02-08:
Aha, I see if I *tap on* an Amiibo from the main N2 Elite screen (immediately after a scan, from the bank list), I get the reduced Amiibo-only version and can "Switch to N2 Elite Options". Where things are getting confused is that I'm scanning my N2 Elite then immediately pulling up the drawer from the bank list screen, which still has the old kitchen-sink menu.

I'd say when pulled up from there, you should get the N2 Elite options rather than specific-slot options, since you're on a list screen to begin with.

AbandonedCart, 2022-02-08:
Not sure how you're pulling up a menu with an item having not selected an item.

vaguerant, 2022-02-08:
> Not sure how you're pulling up a menu with an item having not selected an item.

1. Scan N2 Elite
2. Press ︿ button at bottom right

AbandonedCart, 2022-02-08:
... that would explain why every report has directions on how to fix it. What I meant was that I forgot the active slot didn't pass through the filter method.

vaguerant, 2022-02-09:
One (hopefully) last thing. Taking the following steps:

1. Scan N2 Elite
2. Tap bottom drawer button, opening Amiibo Details
3. Tap "Switch to N2 Elite Options"

[image]

The section of the bank list in the background which was previously obscured by the Amiibo Details menu is now blank when switching to the N2 Elite Options.

And personally, I still think the bank list screen should default to the N2 Elite options rather than Amiibo Details when pulling up the drawer. Usability wise, you can get to Amiibo Details in a single press already by tapping on any Amiibo bank while getting to N2 Elite Options takes two presses. But I'm just one user, so I'm fine to just deal with it if you prefer the setup as-is.

AbandonedCart, 2022-02-09:
In terms of usability, you should not be deleting and replacing the entire device regularly. More often, you will want to interact with the active bank or individual slots. It was built around direct input from the users.

commit 1d5c58e0 (2022-02-06): Implement more extensive NumberPicker view

Fixes #399