HiddenRamblings/TagMo #403: The possibility to download the thumbnails with one click

HumanixSaektati, 2022-02-07:
Hello, I use the TagMo for the first time and the problem I found is that it is not possible to load the thumbnails with one click.

I have to scroll down manually each time to load the thumbnails.

Additionally, I miss a function to backup and restore the thumbnails with a few clicks in the app. The idea here is that a local backup and restore function should be added and in the UI should also be displayed how up-to-date the thumbnail database is compared to the upstream.

AbandonedCart, 2022-02-07:
It seems you may not have visited the app settings yet. There is no need to swipe down to load anything when opening the app, but slow phones will load slowly.

In settings you will find that there is an option to choose how images are loaded. Setting "always" will refresh the images on every load, but only if you have a data connection of some sort available.

A bit further down is a database option. You can sync, load, and reset. Furthermore, you will receive a notification when the upstream has been updated.

Correct me if I'm wrong, but that is everything you just requested already available.

HumanixSaektati, 2022-02-07:
> In settings you will find that there is an option to choose how images are loaded. Setting "always" will refresh the images on every load, but only if you have a data connection of some sort available.

This does not work as expected, I have this enabled and have over 400 bin files. I then went to the main view and only the first visible ones were loaded, the rest were not loaded. The rest only loaded when I scrolled down step by step. This is exactly what I don't want, I find it better to have it in the settings a function where I can download with one click all available thumbnails, this should also work in the background, I don't necessarily have to have the app in the foreground.

AbandonedCart, 2022-02-07:
Your issue is with Android as a platform. You are asking for a setting that undoes the core design of a list.

HumanixSaektati, 2022-02-07:
> Your issue is with Android as a platform. You are asking for a setting that undoes the core design of a list.

The problem currently is that it is very user unfriendly and I always and always have to scroll down manually to get the thumbnails for the particular bin file. Therefore it would be very great if you could make this easier.

Is it possible to create something like this on the Android platform for the TagMo app?

AbandonedCart, 2022-02-07:
It's not unfriendly. Your device is slow. To put it bluntly, the reason you load so little content at a time is that your device is struggling to find the memory. To do what you are asking would require adding each item without putting them into any sort of management wrapper. Doing that would crash before you got through the first half of your collection.

HumanixSaektati, 2022-02-07:
Can you reproduce it on your side it, what exactly happens on a fast Android device? 

You install the TagMo app for the first time and select the path where the bin files are (more than 400 bin files). Data connection is on. Then you do nothing on the display and TagMo automatically downloads all the thumbnails in the next few seconds and you don't have to scroll down anything?

AbandonedCart, 2022-02-07:
https://mega.nz/file/04oj2KZZ#8PSKhOVC3UkWCervtWF4NEQyfMPS2RFPNFHE9UnzjcE

HumanixSaektati, 2022-02-07:
Thanks for the video but are the thumbnails already downloaded on your fast Android device because it looks just like mine when I have all the thumbnails already downloaded and then can scroll down smoothly offline. 

I searched with File Manager app where the thumbnails are stored and for me it is 200 MB of many small files. Therefore I am skeptical whether it can be so fast on a fast Android device, unfortunately I do not have a current Android device from 2021 or 2022.

AbandonedCart, 2022-02-07:
The thumbnails aren't already downloaded. I have Wifi 6. I figured you would have some convoluted argument for how it was invalid, though, so allow me to show you the install, too.

AbandonedCart, 2022-02-07:
https://mega.nz/file/8tIQmDpa#YfIXU_dSa_ZndtSmnIr6uA-WTsmRcb5UZjHkKnRGfyQ

I had to IMMEDIATELY jump into a MASSIVE scroll to barely get to see the images populating. If your issue is that you want to perform an install, open the app, immediately scroll to the 400th item and not see any sort of loading at all, then I can't really help you. It's an unrealistic request.

HumanixSaektati, 2022-02-07:
In the video it looks like you have to scroll for the thumbnails to download. If you just install the app and do nothing the thumbnails will still download on your fast Android device? 

> If your issue is that you want to perform an install, open the app, immediately scroll to the 400th item and not see any sort of loading at all, then I can't really help you. It's an unrealistic request.
That's not what I'm looking for, I don't want to scroll.

AbandonedCart, 2022-02-07:
[image]

It's the same as the two other videos, but this time I didn't scroll. The scrolling does not start the load.

> it looks just like mine when I have all the thumbnails already downloaded and then can scroll down smoothly offline.

That is the purpose of caching the images. It's so that slow devices, such as yours, and fast devices, such as mine, can have a somewhat similar experience after the initial load. As for how quickly that load happens, that depends on the device.

 The processor and RAM have an impact on the speed. It's like carrying water from one pool to another. If you have a 2 gallon bottle, you will be done a lot faster than a half pint carton and the carton will probably lose a little on the way.

HumanixSaektati, 2022-02-07:
> It's the same as the two other videos, but this time I didn't scroll. The scrolling does not start the load.

I'm starting to get it. You opened the app and didn't scroll. The thumbnails are dowloaded at that moment. In your video I can see that because of the display size of your Android device and the view setting, the first 12 thumbnails were assigned for the respective bin files.

Now when you scroll down in the app, the thumbnails for the bin files are assigned. This is not a download of the thumbnails because you have Wi-Fi 6 and probably already have all the thumbnails downloaded.

I found the problem, it is not immediately obvious in the TagMo app if the thumbnail is being downloaded at this moment or the already downloaded thumbnails are assigned for the respective bin file.

Can you improve it so that the differences between "Download" and "Assignment" can be seen visually?

AbandonedCart, 2022-02-07:
That's not an improvement. Quite the opposite, it is a complete destruction.

HumanixSaektati, 2022-02-07:
Why is it a destruction?

HumanixSaektati, 2022-02-07:
The idea could be that first a scan process is displayed, then comes the download bar for few bin files, maybe a text field is enough. The assignment doesn't need an extra visualization, because when the user sees that the thumbnails are added, it is the process "Assignment".

AbandonedCart, 2022-02-07:
There is a ScrollView, a ListView, and a RecyclerView. The ScrollView allows you take make a regular layout scrollable. All of the content is loaded at once and it isn't available until all of the content loads. This was quick to crash and slow to load. The ListView was a step forward by making each item load individually and allowing them to all have a common set of parameters and actions. This was faster to load, but still crashed without a lot of work to optimize it. The next step up from that was a RecyclerView, which is the most popular of such optimized ListView descendants. It allows removing views that are no longer visible in memory and only caching enough information for when they came back into view again. There are also a mess of additional optimizations, some of which can be configured. One such optimization is recycling all of the views all of the time. In TagMo, this has been disabled in favor of stable IDs, which means that rather than each view being completely replaced, each view is treated as its own unique instance. In terms of performance, this just adds a larger index, but doesn't actually keep the physical view in memory any longer. It's very little overhead to actually increase performance and reduce code.

What you're asking I do is break down everything back to that core ScrollView design for what is literally a minute or two on the initial install at the cost of the next X hours, days, weeks, or months of use.

closed 2022-02-07 by AbandonedCart