HiddenRamblings/TagMo #404: Tagmo won't recognise n2 elite?

NuclearPineapple, 2022-02-14:
Hi there, apologies if i've missed something obvious, but i can't seem to get Tagmo to recognise my N2 elite at all. It reads "Attempt to invoke virtual method"void.com.hiddenramblings.tagmo.nfctech.ntag215.connect()" on a null object reference".

I'm too much of an utter novice to know what that means. For reference, i own a bunch of real amiibo that scan just fine. Using the latest version of the app and using a One plus Nord 2 5g phone. Any pointers would be appreciated!

AbandonedCart, 2022-02-14:
Go to settings and make sure you enabled it.

NuclearPineapple, 2022-02-14:
Made sure it was enabled, unfortunately, same error. Any other ideas?

AbandonedCart, 2022-02-14:
Use the log option in the extended menu, per https://github.com/HiddenRamblings/TagMo/wiki#reporting-issues

NuclearPineapple, 2022-02-14:
TagMo - #c00e608

--------- beginning of crash
02-14 19:36:17.310 22402 22402 E AndroidRuntime: FATAL EXCEPTION: main
02-14 19:36:17.310 22402 22402 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 22402
02-14 19:36:17.310 22402 22402 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{com.hiddenramblings.tagmo/com.hiddenramblings.tagmo.TagDataActivity}: android.view.InflateException: Binary XML file line #19 in com.hiddenramblings.tagmo:layout/activity_tag_data: Binary XML file line #19 in com.hiddenramblings.tagmo:layout/activity_tag_data: Error inflating class TextView
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3800)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3968)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2389)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:264)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:8335)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:632)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: Caused by: android.view.InflateException: Binary XML file line #19 in com.hiddenramblings.tagmo:layout/activity_tag_data: Binary XML file line #19 in com.hiddenramblings.tagmo:layout/activity_tag_data: Error inflating class TextView
02-14 19:36:17.310 22402 22402 E AndroidRuntime: Caused by: android.view.InflateException: Binary XML file line #19 in com.hiddenramblings.tagmo:layout/activity_tag_data: Error inflating class TextView
02-14 19:36:17.310 22402 22402 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.graphics.NinePatch.hasAlpha()' on a null object reference
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.graphics.drawable.NinePatchDrawable.getOpacity(NinePatchDrawable.java:534)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.View.computeOpaqueFlags(View.java:18966)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.View.setBackgroundDrawable(View.java:24238)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at androidx.appcompat.widget.AppCompatTextView.setBackgroundDrawable(AppCompatTextView.java:147)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.View.setBackground(View.java:24131)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.View.<init>(View.java:6118)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.widget.TextView.<init>(TextView.java:1018)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.widget.TextView.<init>(TextView.java:1012)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at androidx.appcompat.widget.AppCompatTextView.<init>(AppCompatTextView.java:108)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at androidx.appcompat.widget.AppCompatTextView.<init>(AppCompatTextView.java:103)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at androidx.appcompat.app.AppCompatViewInflater.createTextView(AppCompatViewInflater.java:201)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at androidx.appcompat.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:121)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at androidx.appcompat.app.AppCompatDelegateImpl.createView(AppCompatDelegateImpl.java:1566)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at androidx.appcompat.app.AppCompatDelegateImpl.onCreateView(AppCompatDelegateImpl.java:1617)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.tryCreateView(LayoutInflater.java:1065)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:995)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:959)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1138)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1099)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1141)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1099)
02-14 19:36:17.310 22402 22402 E AndroidRuntime: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:680)
[... cut ...]
02-14 20:00:12.312 28646 28646 I InputTransport: Destroy ARC handle: 0xb4000077333453c0
02-14 20:00:12.335 28646 28646 I InputTransport: Create ARC handle: 0xb40000773358bc80
02-14 20:00:14.389 28646 28646 I InputTransport: Destroy ARC handle: 0xb40000773358bc80
02-14 20:00:14.714 28646 28646 I InputTransport: Destroy ARC handle: 0xb4000076fdd32520
02-14 20:00:19.115 28646 28646 W AutofillManager: setSaveUiState(92905830, false) called on existing session 92905830; cancelling it
02-14 20:00:20.795 28646 28646 I InputTransport: Create ARC handle: 0xb4000076fdeb2fc0

AbandonedCart, 2022-02-14:
What version of Android do you have?

NuclearPineapple, 2022-02-14:
Apparently, Android 11

AbandonedCart, 2022-02-14:
That's unexpected. It seems the issue is with a text view. Looking at the specific view, it's just the word amiibo with a trademark symbol. I'm guessing it may be the symbol, but it's not one that usually comes up as incompatible.

AbandonedCart, 2022-02-17:
Have recent changes resolved this?

closed 2022-02-21 by AbandonedCart