HiddenRamblings/TagMo #407: Error Message: Attempt to invoke virtual Method

ASpooner85, 2022-02-21:
I get the error Message
Attempt to invoke virtual Method 'void v4.e.connect()' on a null object reference. Every time i will spielen something with the Tag (scan, write, Update)

AbandonedCart, 2022-02-21:
https://github.com/HiddenRamblings/TagMo/wiki#reporting-issues

AbandonedCart, 2022-02-21:
Based on the message alone with no other details, it sounds like the device doesn't support NFC.

ASpooner85, 2022-02-21:
My Device is an OnePlus Nord CE 5G
NFC works fine with my ID card. 

[image]

AbandonedCart, 2022-02-21:
The screenshot has confirmed that the error message was typed correctly.

ASpooner85, 2022-02-21:
Now. After installing the newest Version, the App Crashs without generating a log.

Seems my telephone and his user are both to dumb to use the App.

closed 2022-02-21 by ASpooner85

AbandonedCart, 2022-02-21:
If the app crashes instantly, it writes a log.