HiddenRamblings/TagMo #412: Logcat - TagMo #c57453a

LogcatGitHub, 2022-02-23:
*Sent by [**J0sephc191@hotmail.co.uk**](mailto:J0sephc191@hotmail.co.uk)*

Description:
---

Tried to edit amiibo mii name area but it crashes on my Samsung phones and my old phone s5

Device info:
---
<table>
<tr><td>App version</td><td>2.7.9</td></tr>
<tr><td>App version code</td><td>12</td></tr>
<tr><td>Android build version</td><td>G781BXXS4DVA2</td></tr>
<tr><td>Android release version</td><td>12</td></tr>
<tr><td>Android SDK version</td><td>31</td></tr>
<tr><td>Android build ID</td><td>SP1A.210812.016.G781BXXS4DVA2</td></tr>
<tr><td>Device brand</td><td>samsung</td></tr>
<tr><td>Device manufacturer</td><td>samsung</td></tr>
<tr><td>Device name</td><td>r8q</td></tr>
<tr><td>Device model</td><td>SM-G781B</td></tr>
<tr><td>Device product name</td><td>r8qxeea</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```TagMo - #c57453a
Android S (12)

--------- beginning of main
02-23 06:04:44.125 15478 15478 I ViewRootImpl@e79916e[IssueReporterLauncher$Activity]: handleAppVisibility mAppVisible=true visible=false
02-23 06:04:44.141 15478 15478 I ViewRootImpl@e79916e[IssueReporterLauncher$Activity]: Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)8 dur=3 res=0x5 s={false 0} ch=true fn=23
02-23 06:04:44.141 15478 15478 I ViewRootImpl@e79916e[IssueReporterLauncher$Activity]: stopped(true) old=false
02-23 06:04:44.142 15478 15478 I MSHandlerLifeCycle: isMultiSplitHandlerRequested: windowingMode=1 isFullscreen=true isPopOver=false isHidden=false skipActivityType=false isHandlerType=true this: DecorView@ab03db3[IssueReporterLauncher$Activity]
02-23 06:04:44.142 15478 15478 I MSHandlerLifeCycle: removeMultiSplitHandler: no exist. decor=DecorView@ab03db3[IssueReporterLauncher$Activity]
02-23 06:04:44.150 15478 15478 I MSHandlerLifeCycle: removeMultiSplitHandler: no exist. decor=DecorView@ab03db3[IssueReporterLauncher$Activity]
02-23 06:04:44.152 15478 15478 I ViewRootImpl@e79916e[IssueReporterLauncher$Activity]: dispatchDetachedFromWindow
02-23 06:04:44.157 15478 15478 D InputTransport: Input channel destroyed: 'd7b2c74', fd=83
02-23 06:04:56.580 16086 16086 E ramblings.tagm: Not starting debugger since process cannot load the jdwp agent.
02-23 06:04:56.582 16086 16086 E USNET   : USNET: appName: com.hiddenramblings.tagmo
02-23 06:04:56.583 16086 16086 D ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
02-23 06:04:56.589 16086 16086 D ActivityThread: setConscryptValidator
02-23 06:04:56.589 16086 16086 D ActivityThread: setConscryptValidator - put
--------- beginning of system
02-23 06:04:56.642 16086 16086 D ActivityThread: handleBindApplication()++ app=com.hiddenramblings.tagmo
02-23 06:04:56.643 16086 16086 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10622; state: ENABLED
02-23 06:04:56.680 16086 16086 V GraphicsEnvironment: ANGLE Developer option for 'com.hiddenramblings.tagmo' set to: 'default'
02-23 06:04:56.680 16086 16086 V GraphicsEnvironment: App is not on the allowlist for updatable production driver.
02-23 06:04:56.681 16086 16086 D LoadedApk: LoadedApk::makeApplication() appContext.mOpPackageName=com.hiddenramblings.tagmo appContext.mBasePackageName=com.hiddenramblings.tagmo
02-23 06:04:56.681 16086 16086 D NetworkSecurityConfig: No Network Security Config specified, using platform default
02-23 06:04:56.682 16086 16086 D NetworkSecurityConfig: No Network Security Config specified, using platform default
02-23 06:04:56.683 16086 16086 I MultiDex: VM with version 2.1.0 has multidex support
02-23 06:04:56.683 16086 16086 I MultiDex: Installing application
02-23 06:04:56.683 16086 16086 I MultiDex: VM has multidex support, MultiDex support library is disabled.
02-23 06:04:56.701 16086 16086 D ActivityThread: handleBindApplication() --
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: QUALCOMM build                   : 193b2ee, I593c16c433
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: Build Date                       : 10/07/21
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: OpenGL ES Shader Compiler Version: EV031.32.02.10
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: Local Branch                     : Test-lib-SR05432264
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: Remote Branch                    : 
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: Remote Branch                    : 
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: Reconstruct Branch               : 
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: Build Config                     : S P 10.0.5 AArch64
02-23 06:04:56.718 16086 16121 I AdrenoGLES-0: Driver Path                      : /vendor/lib64/egl/libGLESv2_adreno.so
02-23 06:04:56.731 16086 16121 I AdrenoGLES-0: PFP: 0x016dd093, ME: 0x00000000
02-23 06:04:56.735 16086 16086 I DecorView: [INFO] isPopOver=false, config=true
02-23 06:04:56.735 16086 16086 I DecorView: updateCaptionType >> DecorView@8f64a46[], isFloating=false, isApplication=true, hasWindowControllerCallback=true, hasWindowDecorCaption=false
[... 445 lines cut ...]
02-23 06:05:01.097 16298 16298 W System.err: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1130)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1088)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:686)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:538)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:485)
02-23 06:05:01.097 16298 16298 W System.err: 	at e.e.B(AppCompatDelegateImpl.java:4)
02-23 06:05:01.097 16298 16298 W System.err: 	at e.b.setContentView(AppCompatActivity.java:2)
02-23 06:05:01.097 16298 16298 W System.err: 	at com.hiddenramblings.tagmo.TagDataActivity.onCreate(TagDataActivity.java:2)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.Activity.performCreate(Activity.java:8282)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.Activity.performCreate(Activity.java:8262)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1329)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:4011)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4203)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:103)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2440)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.os.Handler.dispatchMessage(Handler.java:106)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.os.Looper.loopOnce(Looper.java:226)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.os.Looper.loop(Looper.java:313)
02-23 06:05:01.097 16298 16298 W System.err: 	at android.app.ActivityThread.main(ActivityThread.java:8641)
02-23 06:05:01.097 16298 16298 W System.err: 	at java.lang.reflect.Method.invoke(Native Method)
02-23 06:05:01.097 16298 16298 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567)
02-23 06:05:01.097 16298 16298 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1133)
```

BladeTechBlade, 2022-02-23:
Can you make this email private.

AbandonedCart, 2022-02-24:
@BladeTechBlade Reports have the OPTION to include an email, but that doesn't give you permission to harass them.

closed 2022-02-24 by AbandonedCart

BladeTechBlade, 2022-02-24:
> @BladeTechBlade Reports have the OPTION to include an email, but that doesn't give you permission to harass them.

That was my email because someone in my family had my phone and they had no right to do it that is why i asked for it to be removed and that is why i had to make a user for github with the same email so it does not happen again.

AbandonedCart, 2022-02-24:
> Can you make this email private so people can have their private email blank like YouTube. I talked to the person of the email and someone used his phone can you remove the email address or blank it out so none has their email please.

> That was my email because someone in my family had my phone and they had no right to do it that is why i asked for it to be removed and that is why i had to make a user for github with the same email so it does not happen again.

There are some inconsistencies here, but that doesn't really matter. The important things are not to harass others and not to go handing your phone out to people that will sign up for things using your accounts.