HiddenRamblings/TagMo #415: Logcat - TagMo #e077bb4

LogcatGitHub, 2022-02-24:
*Sent by [**thetoddmn@gmail.com**](mailto:thetoddmn@gmail.com)*

Description:
---

I keep trying to open up any of the amiibo edit properties on the newest version and it dies. Not sure what else is going on but when I backdate to the December release it seems to work fine but this new stable is not working

Device info:
---
<table>
<tr><td>App version</td><td>2.7.9</td></tr>
<tr><td>App version code</td><td>12</td></tr>
<tr><td>Android build version</td><td>G965USQS8FTJ6</td></tr>
<tr><td>Android release version</td><td>10</td></tr>
<tr><td>Android SDK version</td><td>29</td></tr>
<tr><td>Android build ID</td><td>QP1A.190711.020.G965USQS8FTJ6</td></tr>
<tr><td>Device brand</td><td>samsung</td></tr>
<tr><td>Device manufacturer</td><td>samsung</td></tr>
<tr><td>Device name</td><td>star2qltesq</td></tr>
<tr><td>Device model</td><td>SM-G965U</td></tr>
<tr><td>Device product name</td><td>star2qltesq</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```TagMo - #e077bb4
Android Q (10)

--------- beginning of main
02-24 09:48:03.415 18279 18535 D OpenGLRenderer: makeCurrent EglSurface : 0x7e20447880 -> 0x0
02-24 09:48:03.415 18279 18535 D OpenGLRenderer: destroyEglSurface : 0x7e20447880
02-24 09:48:03.416 18279 18535 D OpenGLRenderer: ~ReliableSurface : 0x7da93f8400
02-24 09:48:03.417 18279 18279 I ViewRootImpl@28cd805[PopupWindow:2940d4e]: dispatchDetachedFromWindow
02-24 09:48:03.425 18279 18279 D InputTransport: Input channel destroyed: 'b3563c4', fd=64
02-24 09:48:04.683 18279 18535 D OpenGLRenderer: makeCurrent EglSurface : 0x0 -> 0x7db784c780
02-24 09:48:05.033 18279 18279 I ViewRootImpl@cbeb57a[PopupWindow:ce8e30f]: ViewPostIme pointer 0
02-24 09:48:05.087 18279 18279 I ViewRootImpl@cbeb57a[PopupWindow:ce8e30f]: ViewPostIme pointer 1
02-24 09:48:05.087 18279 18279 D AbsListView: onTouchUp() mTouchMode : 0
02-24 09:48:05.188 18279 18279 I ViewRootImpl@cbeb57a[PopupWindow:ce8e30f]: Relayout returned: old=(743,98,1440,1306) new=(743,98,1440,1306) req=(697,1208)0 dur=10 res=0x1 s={true 539947560960} ch=false
02-24 09:48:05.221 18279 18639 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.221 18279 18639 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.222 18279 18621 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.222 18279 18621 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.224 18279 18635 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.224 18279 18635 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.226 18279 18620 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.226 18279 18620 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.262 18279 18535 D OpenGLRenderer: makeCurrent EglSurface : 0x7db784c780 -> 0x7e241eb900
02-24 09:48:05.265 18279 18535 D OpenGLRenderer: makeCurrent EglSurface : 0x7e241eb900 -> 0x7db784c780
02-24 09:48:05.278 18279 18535 D OpenGLRenderer: makeCurrent EglSurface : 0x7db784c780 -> 0x7e241eb900
02-24 09:48:05.279 18279 18279 I ViewRootImpl@66a40f7[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
02-24 09:48:05.279 18279 18279 D InputMethodManager: prepareNavigationBarInfo() DecorView@4977744[BrowserActivity]
02-24 09:48:05.279 18279 18279 D InputMethodManager: getNavigationBarColor() -855310
02-24 09:48:05.281 18279 18279 I ViewRootImpl@cbeb57a[PopupWindow:ce8e30f]: MSG_WINDOW_FOCUS_CHANGED 0 1
02-24 09:48:05.299 18279 18535 D OpenGLRenderer: makeCurrent EglSurface : 0x7e241eb900 -> 0x7db784c780
02-24 09:48:05.355 18279 18620 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.355 18279 18620 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.361 18279 18639 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.361 18279 18639 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.361 18279 18621 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.362 18279 18621 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.380 18279 18635 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.380 18279 18635 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-24 09:48:05.568 18279 18535 D OpenGLRenderer: makeCurrent EglSurface : 0x7db784c780 -> 0x0
02-24 09:48:05.568 18279 18535 D OpenGLRenderer: destroyEglSurface : 0x7db784c780
02-24 09:48:05.577 18279 18535 D OpenGLRenderer: ~ReliableSurface : 0x7e207f1a00
[... 300 lines cut ...]
02-24 09:48:12.034 18279 18279 W System.err: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1084)
02-24 09:48:12.034 18279 18279 W System.err: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1126)
02-24 09:48:12.034 18279 18279 W System.err: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1084)
02-24 09:48:12.034 18279 18279 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:682)
02-24 09:48:12.034 18279 18279 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:534)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:481)
02-24 09:48:12.035 18279 18279 W System.err: 	at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:706)
02-24 09:48:12.035 18279 18279 W System.err: 	at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:195)
02-24 09:48:12.035 18279 18279 W System.err: 	at com.hiddenramblings.tagmo.TagDataActivity.onCreate(TagDataActivity.java:142)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.Activity.performCreate(Activity.java:7963)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.Activity.performCreate(Activity.java:7952)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1307)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3629)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3806)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2267)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.os.Handler.dispatchMessage(Handler.java:107)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.os.Looper.loop(Looper.java:237)
02-24 09:48:12.035 18279 18279 W System.err: 	at android.app.ActivityThread.main(ActivityThread.java:8167)
02-24 09:48:12.035 18279 18279 W System.err: 	at java.lang.reflect.Method.invoke(Native Method)
02-24 09:48:12.035 18279 18279 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
02-24 09:48:12.035 18279 18279 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
```

AbandonedCart, 2022-02-24:
#416
#417

AbandonedCart, 2022-02-25:
Update required.

closed 2022-02-25 by AbandonedCart