HiddenRamblings/TagMo #424: Logcat - TagMo #44bac99

LogcatGitHub, 2022-02-26:
*Submitted by thetoddmn@*****.com*

Description:
---

Update made based on the previous closed issue. Opened app and went into edit properties and had the same crashing behavior.

Device info:
---
<table>
<tr><td>App version</td><td>2.7.9</td></tr>
<tr><td>App version code</td><td>12</td></tr>
<tr><td>Android build version</td><td>G965USQS8FTJ6</td></tr>
<tr><td>Android release version</td><td>10</td></tr>
<tr><td>Android SDK version</td><td>29</td></tr>
<tr><td>Android build ID</td><td>QP1A.190711.020.G965USQS8FTJ6</td></tr>
<tr><td>Device brand</td><td>samsung</td></tr>
<tr><td>Device manufacturer</td><td>samsung</td></tr>
<tr><td>Device name</td><td>star2qltesq</td></tr>
<tr><td>Device model</td><td>SM-G965U</td></tr>
<tr><td>Device product name</td><td>star2qltesq</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```TagMo - #44bac99
Android Q (10)

--------- beginning of main
02-26 11:49:23.439 24904 24904 D InputMethodManager: prepareNavigationBarInfo() DecorView@686fe9d[BrowserActivity]
02-26 11:49:23.439 24904 24904 D InputMethodManager: getNavigationBarColor() -855310
02-26 11:49:23.445 24904 24904 D InputMethodManager: prepareNavigationBarInfo() DecorView@686fe9d[BrowserActivity]
02-26 11:49:23.445 24904 24904 D InputMethodManager: getNavigationBarColor() -855310
02-26 11:49:23.445 24904 24904 V InputMethodManager: Starting input: tba=com.hiddenramblings.tagmo ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
02-26 11:49:23.445 24904 24904 D InputMethodManager: startInputInner - Id : 0
02-26 11:49:23.445 24904 24904 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
02-26 11:49:23.451 24904 24904 I ViewRootImpl@7c74d10[BrowserActivity]: MSG_RESIZED: frame=(0,0,1440,2960) ci=(0,84,0,168) vi=(0,84,0,168) or=1
02-26 11:49:23.453 24904 24904 D InputMethodManager: prepareNavigationBarInfo() DecorView@686fe9d[BrowserActivity]
02-26 11:49:23.453 24904 24904 D InputMethodManager: getNavigationBarColor() -855310
02-26 11:49:23.453 24904 24904 V InputMethodManager: Starting input: tba=com.hiddenramblings.tagmo ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
02-26 11:49:23.453 24904 24904 D InputMethodManager: startInputInner - Id : 0
02-26 11:49:23.844 24904 13381 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-26 11:49:23.844 24904 13381 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-26 11:49:24.349 24904 13319 W System  : A resource failed to call end. 
02-26 11:49:24.869 24904 24904 I ViewRootImpl@7c74d10[BrowserActivity]: ViewPostIme pointer 0
02-26 11:49:24.923 24904 24904 I ViewRootImpl@7c74d10[BrowserActivity]: ViewPostIme pointer 1
02-26 11:49:27.608 24904 24904 I ViewRootImpl@7c74d10[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
02-26 11:49:27.608 24904 24904 D InputMethodManager: prepareNavigationBarInfo() DecorView@686fe9d[BrowserActivity]
02-26 11:49:27.608 24904 24904 D InputMethodManager: getNavigationBarColor() -855310
02-26 11:49:28.839 24904 24904 I ViewRootImpl@7c74d10[BrowserActivity]: stopped(false) old=false
02-26 11:49:28.917 24904 24904 I ViewRootImpl@7c74d10[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
02-26 11:49:28.918 24904 24904 D InputMethodManager: prepareNavigationBarInfo() DecorView@686fe9d[BrowserActivity]
02-26 11:49:28.918 24904 24904 D InputMethodManager: getNavigationBarColor() -855310
02-26 11:49:34.823 14008 14008 E ramblings.tagm: Not starting debugger since process cannot load the jdwp agent.
02-26 11:49:34.829 14008 14008 D ActivityThread: setConscryptValidator
02-26 11:49:34.829 14008 14008 D ActivityThread: setConscryptValidator - put
02-26 11:49:38.633 13410 13410 E ramblings.tagm: Not starting debugger since process cannot load the jdwp agent.
02-26 11:49:38.637 13410 13410 D ActivityThread: setConscryptValidator
02-26 11:49:38.637 13410 13410 D ActivityThread: setConscryptValidator - put
02-26 11:49:38.753 13410 13410 D PhoneWindow: forceLight changed to true [] from com.android.internal.policy.PhoneWindow.updateForceLightNavigationBar:4274 com.android.internal.policy.DecorView.updateColorViews:1547 com.android.internal.policy.PhoneWindow.dispatchWindowAttributesChanged:3252 android.view.Window.setFlags:1153 com.android.internal.policy.PhoneWindow.generateLayout:2474 
02-26 11:49:38.754 13410 13410 I MultiWindowDecorSupport: [INFO] isPopOver = false
02-26 11:49:38.754 13410 13410 I MultiWindowDecorSupport: updateCaptionType >> DecorView@4977744[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
02-26 11:49:38.754 13410 13410 D MultiWindowDecorSupport: setCaptionType = 0, DecorView = DecorView@4977744[]
02-26 11:49:38.831 13410 14353 D NetworkSecurityConfig: No Network Security Config specified, using platform default
02-26 11:49:38.844 13410 14353 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
02-26 11:49:38.844 13410 14353 I System.out: (HTTPLog)-Static: isSBSettingEnabled false
[... 351 lines cut ...]
02-26 11:49:44.212 13410 13410 W System.err: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1084)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1126)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1084)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:682)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:534)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.view.LayoutInflater.inflate(LayoutInflater.java:481)
02-26 11:49:44.212 13410 13410 W System.err: 	at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:706)
02-26 11:49:44.212 13410 13410 W System.err: 	at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:195)
02-26 11:49:44.212 13410 13410 W System.err: 	at com.hiddenramblings.tagmo.TagDataActivity.onCreate(TagDataActivity.java:142)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.Activity.performCreate(Activity.java:7963)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.Activity.performCreate(Activity.java:7952)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1307)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3629)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3806)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2267)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.os.Handler.dispatchMessage(Handler.java:107)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.os.Looper.loop(Looper.java:237)
02-26 11:49:44.212 13410 13410 W System.err: 	at android.app.ActivityThread.main(ActivityThread.java:8167)
02-26 11:49:44.212 13410 13410 W System.err: 	at java.lang.reflect.Method.invoke(Native Method)
02-26 11:49:44.212 13410 13410 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
02-26 11:49:44.212 13410 13410 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
```

GovtFatCat, 2022-02-26:
Just adding I had to uninstall and reinstall the December release.

closed 2022-02-26 by AbandonedCart

GovtFatCat, 2022-02-26:
If I could get the ability to turn off the auto updates that'd help

AbandonedCart, 2022-02-27:
You'd have to update to do that, so it's a bit of a paradox.

commit 3f7d9cd7 (2022-02-26): Remove compat for nine-patch XML drawable

Fixes #424