HiddenRamblings/TagMo #440: [Bug]: Scan, Write and Update buttons entirely nonfunctional

uwunium, 2022-03-04:
### Issue Description

I was able to save backups of a couple of my amiibos, as well as importing some bins. However, as soon as I clicked any of the buttons other than save, the app crashed.

I tried installing previous versions and I had some luck at least getting everything but Update functional, as this version just got stuck on the scanning mifare lite screen. I tried downgrading even more, but every version i tried crashed as soon as I opened the app.

I was also unable to directly send my logcat issue from the form after the crash. 

### Logcat Output

```shell
TagMo - #27ea982
Android KITKAT (4.4.2)

--------- beginning of /dev/log/system
--------- beginning of /dev/log/main
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
I/dalvikvm(19705): Could not find method com.google.android.gms.org.conscrypt.TrustManagerImpl.getDefaultHostnameVerifier, referenced from method com.google.android.gms.org.conscrypt.Conscrypt.getDefaultHostnameVerifier
W/dalvikvm(19705): VFY: unable to resolve static method 24820: Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;.getDefaultHostnameVerifier ()Lcom/google/android/gms/org/conscrypt/ConscryptHostnameVerifier;
D/dalvikvm(19705): VFY: replacing opcode 0x71 at 0x0003
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
W/dalvikvm(19705): VFY: unable to find class referenced in signature (Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;)
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
I/dalvikvm(19705): Could not find method com.google.android.gms.org.conscrypt.TrustManagerImpl.getHostnameVerifier, referenced from method com.google.android.gms.org.conscrypt.Conscrypt.getHostnameVerifier
W/dalvikvm(19705): VFY: unable to resolve virtual method 24822: Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;.getHostnameVerifier ()Lcom/google/android/gms/org/conscrypt/ConscryptHostnameVerifier;
D/dalvikvm(19705): VFY: replacing opcode 0x6e at 0x0004
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
E/dalvikvm(19705): Could not find class 'com.google.android.gms.org.conscrypt.TrustManagerImpl', referenced from method com.google.android.gms.org.conscrypt.Conscrypt.isConscrypt
W/dalvikvm(19705): VFY: unable to resolve instanceof 5876 (Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;) in Lcom/google/android/gms/org/conscrypt/Conscrypt;
D/dalvikvm(19705): VFY: replacing opcode 0x20 at 0x0000
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
I/dalvikvm(19705): Could not find method com.google.android.gms.org.conscrypt.TrustManagerImpl.setDefaultHostnameVerifier, referenced from method com.google.android.gms.org.conscrypt.Conscrypt.setDefaultHostnameVerifier
W/dalvikvm(19705): VFY: unable to resolve static method 24832: Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;.setDefaultHostnameVerifier (Lcom/google/android/gms/org/conscrypt/ConscryptHostnameVerifier;)V
D/dalvikvm(19705): VFY: replacing opcode 0x71 at 0x0003
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
W/dalvikvm(19705): VFY: unable to find class referenced in signature (Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;)
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
I/dalvikvm(19705): Could not find method com.google.android.gms.org.conscrypt.TrustManagerImpl.setHostnameVerifier, referenced from method com.google.android.gms.org.conscrypt.Conscrypt.setHostnameVerifier
W/dalvikvm(19705): VFY: unable to resolve virtual method 24833: Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;.setHostnameVerifier (Lcom/google/android/gms/org/conscrypt/ConscryptHostnameVerifier;)V
D/dalvikvm(19705): VFY: replacing opcode 0x6e at 0x0004
W/dalvikvm(19705): Unable to resolve superclass of Lcom/google/android/gms/org/conscrypt/TrustManagerImpl; (6923)
W/dalvikvm(19705): Link of class 'Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;' failed
E/dalvikvm(19705): Could not find class 'com.google.android.gms.org.conscrypt.TrustManagerImpl', referenced from method com.google.android.gms.org.conscrypt.Conscrypt.toConscrypt
W/dalvikvm(19705): VFY: unable to resolve check-cast 5876 (Lcom/google/android/gms/org/conscrypt/TrustManagerImpl;) in Lcom/google/android/gms/org/conscrypt/Conscrypt;
D/dalvikvm(19705): VFY: replacing opcode 0x1f at 0x002a
D/dalvikvm(19705): GC_CONCURRENT freed 4157K, 30% free 10409K/14732K, paused 3ms+16ms, total 81ms
D/dalvikvm(19705): WAIT_FOR_CONCURRENT_GC blocked 40ms
D/dalvikvm(19705): WAIT_FOR_CONCURRENT_GC blocked 43ms
W/dalvikvm(19705): VFY: unable to resolve static field 184 (EXTERNAL_CONTENT_URI) in Landroid/provider/MediaStore$Downloads;
D/dalvikvm(19705): VFY: replacing opcode 0x62 at 0x0151
W/TagReader(19705): 04C2C688015946031D480FE0F110FFEEA5000F00975C5B54341C6A3A890EEF9FF644C41D1B33883DD63C7551307BE09C9E59F9A022BECB702FC7063348F36C0DAC3548E3C5DED5C3508CDBABF7D8DD9BAB2B8DB701010000000E0002051294133AEB0F98D79E794DCE790576DD9E45E16EEC74BED3117BC94854C6B06DD22907FB17472ECA20A3A36BBF2956C2331BB202A11F28826973F6934187FF4C36B85DC1370FAEE0010EEB69D8C4309AA67D793EFABA8CED1CF0388E36198422DC4A6D83A1A4F86F506E76A451D1FB0D5E964AAB71DFC15090947E6C01933EAF9B2B4CD80BCD025F346304595490FD8130FD22E78D1E4A7FADF87C40A18C2A73F932A3D0738418255ACFABF46E9639D53AC3B67C8DC7A2848EA27D661BF75865ED6F7B99D425F306315CFF02A7DE0FECC41BE829CE351B4437A1E22652BD4BF7AC5D946E8959BA593CF7AB0DFC8F482C64E90C0C421CCAB5DBB7F5A107D3E09C01107E9EED163B175DA32A0CE46D5906114405A07B40FB9EF6A03717ED1A0D7273E8308275D5E4874D8DDC14A26937BC01581A8EB59D156BE95C35DB6F93823F474B7FB893230613483264C7FFCF7B945F0D54EBAD824C00D4B73942D77E00E57367084C73E58F7787D392BFBB5561D0E9F4BB42F88B51EC60E2446FEA884D9DD7702095103D38B6A3ED3A7C4695DEB2C758AE42FD1C5540098150A031A206D0E746892CD47A4046A5FDC201000FBD000000045F000000
I/dalvikvm(19705): Could not find method android.view.MenuItem.setAlphabeticShortcut, referenced from method androidx.core.view.MenuItemCompat.setAlphabeticShortcut
W/dalvikvm(19705): VFY: unable to resolve interface method 2256: Landroid/view/MenuItem;.setAlphabeticShortcut (CI)Landroid/view/MenuItem;
D/dalvikvm(19705): VFY: replacing opcode 0x72 at 0x0010
I/dalvikvm(19705): Could not find method android.view.MenuItem.setContentDescription, referenced from method androidx.core.view.MenuItemCompat.setContentDescription
W/dalvikvm(19705): VFY: unable to resolve interface method 2259: Landroid/view/MenuItem;.setContentDescription (Ljava/lang/CharSequence;)Landroid/view/MenuItem;
D/dalvikvm(19705): VFY: replacing opcode 0x72 at 0x0010
I/dalvikvm(19705): Could not find method android.view.MenuItem.setIconTintList, referenced from method androidx.core.view.MenuItemCompat.setIconTintList
W/dalvikvm(19705): VFY: unable to resolve interface method 2263: Landroid/view/MenuItem;.setIconTintList (Landroid/content/res/ColorStateList;)Landroid/view/MenuItem;
D/dalvikvm(19705): VFY: replacing opcode 0x72 at 0x0010
I/dalvikvm(19705): Could not find method android.view.MenuItem.setIconTintMode, referenced from method androidx.core.view.MenuItemCompat.setIconTintMode
W/dalvikvm(19705): VFY: unable to resolve interface method 2264: Landroid/view/MenuItem;.setIconTintMode (Landroid/graphics/PorterDuff$Mode;)Landroid/view/MenuItem;
D/dalvikvm(19705): VFY: replacing opcode 0x72 at 0x0010
[... 178 lines cut ...]
[... cut ...]
W/dalvikvm(19705): VFY: unable to resolve virtual method 4065: Landroid/widget/ListView;.drawableHotspotChanged (FF)V
D/dalvikvm(19705): VFY: replacing opcode 0x6e at 0x0009
I/dalvikvm(19705): Could not find method android.view.View.drawableHotspotChanged, referenced from method androidx.appcompat.widget.DropDownListView.setPressedItem
W/dalvikvm(19705): VFY: unable to resolve virtual method 2365: Landroid/view/View;.drawableHotspotChanged (FF)V
D/dalvikvm(19705): VFY: replacing opcode 0x6e at 0x0046
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-03-04:
As you may know, depending on how long you've been using TagMo, Android 4.4 support was dropped in about 2017. It was only in the last year or two that it has been partially restored. Somehow the idea of downgrading as a solution has become a thing, though. In this case, that is only going to lead to more problems.

uwunium, 2022-03-04:
Man, I didn’t know anything about this. I’m brand new to tagmo- in fact I bought this phone specifically so I could start using it. 

Before I completely throw in the towel, how many problems would I have if I did try downgrading or something? I don’t want all this to go to waste.

> As you may know, depending on how long you've been using TagMo, Android 4.4 support was dropped in about 2017. It was only in the last year or two that it has been partially restored. Somehow the [...]

AbandonedCart, 2022-03-04:
What I was saying is that downgrading won't help because it didn't start out working and stop. The only direction is forward.

uwunium, 2022-03-04:
> What I was saying is that downgrading won't help because it didn't start out working and stop. The only direction is forward.

Good to know. 

That being said, what is the earliest version of android tagmo is fully functional with?

AbandonedCart, 2022-03-04:
The one just published is as functional as it gets.

closed 2022-03-04 by AbandonedCart