HiddenRamblings/TagMo #446: [Bug]: Auth result was null

uwunium, 2022-03-04:
### Issue Description

Firstly, thank you for the update yesterday that fixed the crashing problem on 4.4! 

as for this error: When I attempt to refresh/update an existing tag, I get “auth result was null.” as an instant error message. Firstly, I don’t believe this to be a hardware problem, as other apps on my phone could read and write data just fine, and I was able to write a blank tag in tagmo itself. In addition to this, the error was instant- the error logs make it look like I moved the tag away, but I did not. 

I noticed the same error when I downgraded to the 2017 version, but instead it said “tag lost.” 

I worked together with a friend to look through the error logs to see what was up, but we couldn’t figure out why the tag registered as moving away when I held it still.

I know tagmo doesn’t support 4.4 anymore, but I thought I’d make you guys aware of the error anyway, just in case it’s something that can be fixed.

Thank you so much! :)

### Logcat Output

```shell
TagMo - #ba9e11a
Android KITKAT (4.4.2)

--------- beginning of /dev/log/system
--------- beginning of /dev/log/main
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/dalvikvm( 2875): GC_FOR_ALLOC freed 1674K, 22% free 17057K/21612K, paused 80ms, total 80ms
I/dalvikvm-heap( 2875): Grow heap (frag case) to 19.741MB for 968272-byte allocation
I/ActivityManager( 2875): Timeline: Activity_idle id: android.os.BinderProxy@41ec29f0 time:595969
W/TagWriter( 2875): Data validation successful!
W/TagWriter( 2875): Password: 79A2ED38
W/TagLostException( 2875): android.nfc.TagLostException: Tag was lost.
W/TagLostException( 2875): 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:48)
W/TagLostException( 2875): 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:151)
W/TagLostException( 2875): 	at android.nfc.tech.NfcA.transceive(NfcA.java:120)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.nfctech.NTAG215.transceive(NTAG215.java:70)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.doAuth(TagWriter.java:251)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.restoreTag(TagWriter.java:190)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:376)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:243)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(NfcActivity.java)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source)
W/TagLostException( 2875): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
W/TagLostException( 2875): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
W/TagLostException( 2875): 	at java.lang.Thread.run(Thread.java:841)
W/Exception( 2875): java.lang.Exception: Auth result was null
W/Exception( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.doAuth(TagWriter.java:253)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.restoreTag(TagWriter.java:190)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:376)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:243)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(NfcActivity.java)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source)
W/Exception( 2875): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
W/Exception( 2875): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
W/Exception( 2875): 	at java.lang.Thread.run(Thread.java:841)
I/ActivityManager( 2875): Timeline: Activity_idle id: android.os.BinderProxy@41d965e0 time:598585
D/dalvikvm( 2875): GC_CONCURRENT freed 2256K, 25% free 16927K/22560K, paused 13ms+12ms, total 130ms
D/dalvikvm( 2875): GC_CONCURRENT freed 1914K, 31% free 15786K/22560K, paused 3ms+5ms, total 102ms
D/dalvikvm( 2875): GC_CONCURRENT freed 1399K, 24% free 15814K/20664K, paused 3ms+6ms, total 91ms
W/ContextImpl( 2875): Failed to ensure directory: /storage/external_SD/Android/data/com.hiddenramblings.tagmo/files
W/DynamiteModule( 2875): Local module descriptor class for com.google.android.gms.providerinstaller.dynamite not found.
I/DynamiteModule( 2875): Considering local module com.google.android.gms.providerinstaller.dynamite:0 and remote module com.google.android.gms.providerinstaller.dynamite:0
W/ProviderInstaller( 2875): Failed to load providerinstaller module: No acceptable module com.google.android.gms.providerinstaller.dynamite found. Local version is 0 and remote version is 0.
W/ContextImpl( 2875): Failed to ensure directory: /storage/external_SD/Android/data/com.hiddenramblings.tagmo/cache
D/dalvikvm( 2875): GC_CONCURRENT freed 1484K, 16% free 17476K/20664K, paused 3ms+9ms, total 101ms
D/dalvikvm( 2875): WAIT_FOR_CONCURRENT_GC blocked 69ms
D/dalvikvm( 2875): WAIT_FOR_CONCURRENT_GC blocked 69ms
D/dalvikvm( 2875): GC_CONCURRENT freed 3933K, 24% free 15811K/20784K, paused 3ms+6ms, total 101ms
I/ActivityManager( 2875): Timeline: Activity_launch_request id:com.hiddenramblings.tagmo time:641609
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/BubblePopupHelper( 2875): isShowingBubblePopup : false
D/dalvikvm( 2875): GC_FOR_ALLOC freed 809K, 24% free 15904K/20784K, paused 75ms, total 75ms
I/dalvikvm-heap( 2875): Grow heap (frag case) to 18.615MB for 968272-byte allocation
I/ActivityManager( 2875): Timeline: Activity_idle id: android.os.BinderProxy@42b4bf40 time:641911
W/TagWriter( 2875): Data validation successful!
W/TagWriter( 2875): Password: FB4EE372
W/TagLostException( 2875): android.nfc.TagLostException: Tag was lost.
W/TagLostException( 2875): 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:48)
W/TagLostException( 2875): 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:151)
W/TagLostException( 2875): 	at android.nfc.tech.NfcA.transceive(NfcA.java:120)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.nfctech.NTAG215.transceive(NTAG215.java:70)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.doAuth(TagWriter.java:251)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.restoreTag(TagWriter.java:190)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:376)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:243)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(NfcActivity.java)
W/TagLostException( 2875): 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source)
W/TagLostException( 2875): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
W/TagLostException( 2875): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
W/TagLostException( 2875): 	at java.lang.Thread.run(Thread.java:841)
W/Exception( 2875): java.lang.Exception: Auth result was null
W/Exception( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.doAuth(TagWriter.java:253)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.nfctech.TagWriter.restoreTag(TagWriter.java:190)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:376)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:243)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(NfcActivity.java)
W/Exception( 2875): 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source)
W/Exception( 2875): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
W/Exception( 2875): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
W/Exception( 2875): 	at java.lang.Thread.run(Thread.java:841)
I/ActivityManager( 2875): Timeline: Activity_idle id: android.os.BinderProxy@41d965e0 time:649081
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-03-04:
Experimental support and not supported are two vastly different things.

AbandonedCart, 2022-03-04:
Standard tags will "shut down" when a failure occurs. The reason it says tag lost is because something you are attempting to write is not supported by the tag. 

As I had mentioned before, we have added support that didn't exist in previous years. This includes getting past the generic "tag lost" error that covered everything and showing the real failure point. In this case, that failure point would be when it attempts to authenticate the tag. The likely cause for that being the tag is not actually compatible with the restore process. TagMo cannot restore to real amiibo and also cannot write data that modifies anything not normally modified by the process of using the amiibo on a console. That is beyond our control.

uwunium, 2022-03-04:
Thanks for the response! I should mention that this error occurs even when I try to update a tag with the exact same bin data. I also had a friend try the refresh feature on their own phone and it worked. She uses the same tags as I have, and is using bins I made.

I'm just using basic NTAG215s and a backup bin I scanned, if that helps any.

Thanks again for the response- you guys are doing great work.

AbandonedCart, 2022-03-04:
That's a tricky one. The failure is coming from the NFC driver, so it will require some investigation.

AbandonedCart, 2022-03-28:
Is this still happening? Unfortunately, one limitation of the Android emulator is that it does not support NFC. Without a physical 4.X Android device, the scan process cannot be tested.

closed 2022-04-10 by AbandonedCart