HiddenRamblings/TagMo #461: Bluup Flask Compatibility

AbandonedCart, 2022-03-27:
This issue is for tracking progress of Bluup Flask compatibility. 

As it stands now, we have lost contact with the device developer and Bluetooth LE (BLE) support is limited to devices with Jelly Bean MR2 (4.3.1) or newer. Development is being conducted in the current test branch.

As there is no hardware to test, progress is hindered by the need for logs. This means that all progress is purely experimental and completely untested when published. Any changes may break device support at any time, as code cannot be verified.

Testing requires logs. Logs are available using the current submission methods provided in TagMo. As this is experimental code, ambiguous error messages and other forms of generic output (ie. Screenshots) do not provide necessary details.

AbandonedCart, 2022-04-01:
https://github.com/HiddenRamblings/TagMo/issues/463
https://github.com/HiddenRamblings/TagMo/issues/464

withgallantry, 2022-04-02:
Hi, I now have some hardware I can provide to develop against. I had some delay due to ongoing chip and components shortages. Let me know where I can send it to at bluup@bluuplabs.com. I'm also reachable at that address if you need anything.

AbandonedCart, 2022-04-02:
Hope you're feeling better. I had a feeling you might not be actively checking GitHub, but didn't want anyone thinking it was just randomly paused for no reason.

AbandonedCart, 2022-04-05:
Doing a little housekeeping to make navigation "pretty" before going back to working on functionality. Just a heads up that most of the recent and upcoming commits for the moment are cosmetic.

AbandonedCart, 2022-04-06:
It seems the permission flow for bluetooth is somewhat convoluted. Documentation claims I need access to fine location for bluetooth. Background location is also acceptable on Android 12. To obtain fine background location, I should request both or the option won't be displayed. Makes sense.

Doing that actually ends up saying I should get foreground permission first and then prevents displaying any prompts and immediately returns denied. Asking for background location and denying it manually works.

AbandonedCart, 2022-04-08:
Has anyone tested connecting recently? It won't actually do anything yet, but it should successfully connect.

Ericzion1999, 2022-04-08:
It says ,,Not a valid flask device,,

AbandonedCart, 2022-04-08:
> Logs are available using the current submission methods provided in TagMo. As this is experimental code, ambiguous error messages and other forms of generic output (ie. Screenshots) do not provide necessary details.

Ericzion1999, 2022-04-08:
So , do you need a Logcat ?

AbandonedCart, 2022-04-10:
The method to identify the device has been changed a bit and now there is a status message when it's found.

Ericzion1999, 2022-04-10:
it finds all my bluetooth devices but not the flask

AbandonedCart, 2022-04-10:
There isn't much I can do with that information.

Ericzion1999, 2022-04-10:
What is the correct Method to find It With tagmo? maybe I'm doing something wrong all the time

AbandonedCart, 2022-04-10:
Scroll over to the "Flask BLE" window. If it is found (based on the device name), it tries to connect. If it was renamed, it should show up in the list and it needs to be clicked. If it's not paired, the add new device option opens pairing options.

It may work or it may not. I have no idea. I don't have one yet to see what happens.

Ericzion1999, 2022-04-10:
this is exactly what i was doing all the time, i connected the flask to my phone but it didn't show up in the tagmo list, only other devices

AbandonedCart, 2022-04-10:
Then it doesn't work.

AbandonedCart, 2022-04-10:
It can confirm that my iPad has BLE services available, but none match the Flask characteristics. Beyond that, I can't test until the hardware arrives.

Miss1sle, 2022-04-15:
I have paid my phone to my flask but the only 3 devices I see in TagMo are my car, my printer and an old Bluetooth speaker. None of which are near me or actively connected to my phone.

AbandonedCart, 2022-04-15:
Everything sounds normal so far. The part I can't test myself is what happens after that point.

AbandonedCart, 2022-04-21:
Added a visual notice for when the app finds the Flask on its own. This should reduce some of the confusion that it isn't being listed.

AbandonedCart, 2022-04-28:
To confirm, there are no popups stating that the device was detected since the latest change to add them?

Ericzion1999, 2022-04-28:
Yes , there is No Popup

AbandonedCart, 2022-05-01:
@Miss1sle Are you getting the same?

Ericzion1999, 2022-05-01:
I don't know if this is a helpful tip, but if you use the web version, there is no point in pairing the flask with the smartphone/PC via Bluetooth beforehand, because then the web version can no longer find the flask

Miss1sle, 2022-05-01:
I now see the flask in the list in tagmo, however when I click it, tagmo tells me it's not a valid device.

AbandonedCart, 2022-05-01:
Connecting without pairing on Android would require the MAC address, which we don't have going into the process.

AbandonedCart, 2022-05-15:
Progress should ramp up in the next week or two. The entire format is changing and will more closely resemble that of the N2 in that selecting the device from a list will be replaced with detecting the device.

AbandonedCart, 2022-05-19:
It seems a bit silly to maintain an issue for testing at this point.

closed 2022-05-19 by AbandonedCart

AbandonedCart, 2022-05-20:
... since connecting works.