HiddenRamblings/TagMo #466: Logcat - TagMo #db4e84f

LogcatGitHub, 2022-04-06:
*Submitted by gedrean@*****.com*

Description:
---

After relaunch when search crashed app

Device info:
---
<table>
<tr><td>App version</td><td>2.8.0</td></tr>
<tr><td>App version code</td><td>13</td></tr>
<tr><td>Android build version</td><td>G950U1UEU8DUG4</td></tr>
<tr><td>Android release version</td><td>9</td></tr>
<tr><td>Android SDK version</td><td>28</td></tr>
<tr><td>Android build ID</td><td>PPR1.180610.011.G950U1UEU8DUG4</td></tr>
<tr><td>Device brand</td><td>samsung</td></tr>
<tr><td>Device manufacturer</td><td>samsung</td></tr>
<tr><td>Device name</td><td>dreamqlteue</td></tr>
<tr><td>Device model</td><td>SM-G950U1</td></tr>
<tr><td>Device product name</td><td>dreamqlteue</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```TagMo - #db4e84f
Android P (9)

--------- beginning of main
04-06 17:57:49.564  1782  1782 D InputMethodManager: prepareNavigationBarInfo() DecorView@cab61a8[BrowserActivity]
04-06 17:57:49.564  1782  1782 D InputMethodManager: getNavigationBarColor() -855310
04-06 17:57:49.564  1782  1782 V InputMethodManager: Starting input: tba=com.hiddenramblings.tagmo ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
04-06 17:57:49.565  1782  1782 D InputMethodManager: startInputInner - Id : 0
04-06 17:57:49.565  1782  1782 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
04-06 17:57:49.569  1782  1794 D InputTransport: Input channel constructed: fd=87
04-06 17:57:49.587  1782  1782 D InputMethodManager: prepareNavigationBarInfo() DecorView@cab61a8[BrowserActivity]
04-06 17:57:49.587  1782  1782 D InputMethodManager: getNavigationBarColor() -855310
04-06 17:57:49.587  1782  1782 V InputMethodManager: Starting input: tba=com.hiddenramblings.tagmo ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
04-06 17:57:49.587  1782  1782 D InputMethodManager: startInputInner - Id : 0
04-06 17:57:49.609  1782  1824 W IOException: java.io.IOException: /storage/F447-D1B1/Amiibo/Amiibo Bins/key_retail.bin size: 160, Expected: 532
04-06 17:57:49.609  1782  1824 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.getTagData(TagReader.java:35)
04-06 17:57:49.609  1782  1824 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readTagFile(TagReader.java:45)
04-06 17:57:49.609  1782  1824 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagUtils.getValidatedFile(TagUtils.java:267)
04-06 17:57:49.609  1782  1824 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:297)
04-06 17:57:49.609  1782  1824 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity.lambda$loadAmiiboFiles$22(BrowserActivity.java:1153)
04-06 17:57:49.609  1782  1824 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity.$r8$lambda$pdKPf2zyqgBD_UPxZpv7k-PGAr0(Unknown Source:0)
04-06 17:57:49.609  1782  1824 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity$$ExternalSyntheticLambda40.run(Unknown Source:6)
04-06 17:57:49.609  1782  1824 W IOException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
04-06 17:57:49.609  1782  1824 W IOException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
04-06 17:57:49.609  1782  1824 W IOException: 	at java.lang.Thread.run(Thread.java:764)
04-06 17:57:50.721  1782  1782 W ramblings.tagm: Accessing hidden method Landroid/view/View;->setTransitionVisibility(I)V (dark greylist, linking)
04-06 17:57:50.733  1782  1782 D skia    : approximate: numPoints=2, lengths[0]=0.000000, totalLength=108.000000, segmentPoints[0]=[0.000000,108.000000]
04-06 17:57:50.733  1782  1782 D skia    : approximate: approximation[0]=0.000000, approximation[1]=0.000000, approximation[2]=108.000000
04-06 17:57:50.733  1782  1782 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[0.000000,0.000000]
04-06 17:57:50.733  1782  1782 D skia    : approximate: approximation[3]=1.000000, approximation[4]=0.000000, approximation[5]=0.000000
04-06 17:57:50.734  1782  1782 D skia    : approximate: numPoints=2, lengths[0]=0.000000, totalLength=108.000000, segmentPoints[0]=[1080.000000,1728.000000]
04-06 17:57:50.734  1782  1782 D skia    : approximate: approximation[0]=0.000000, approximation[1]=1080.000000, approximation[2]=1728.000000
04-06 17:57:50.734  1782  1782 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[1080.000000,1836.000000]
04-06 17:57:50.734  1782  1782 D skia    : approximate: approximation[3]=1.000000, approximation[4]=1080.000000, approximation[5]=1836.000000
04-06 17:57:50.734  1782  1782 D skia    : approximate: numPoints=2, lengths[0]=0.000000, totalLength=108.000000, segmentPoints[0]=[1080.000000,1728.000000]
04-06 17:57:50.734  1782  1782 D skia    : approximate: approximation[0]=0.000000, approximation[1]=1080.000000, approximation[2]=1728.000000
04-06 17:57:50.734  1782  1782 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[1080.000000,1836.000000]
04-06 17:57:50.734  1782  1782 D skia    : approximate: approximation[3]=1.000000, approximation[4]=1080.000000, approximation[5]=1836.000000
04-06 17:57:50.734  1782  1782 D skia    : approximate: numPoints=2, lengths[0]=0.000000, totalLength=108.000000, segmentPoints[0]=[1080.000000,1725.000000]
04-06 17:57:50.734  1782  1782 D skia    : approximate: approximation[0]=0.000000, approximation[1]=1080.000000, approximation[2]=1725.000000
04-06 17:57:50.734  1782  1782 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[1080.000000,1833.000000]
[... 278 lines cut ...]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[0]=0.000000, approximation[1]=1080.000000, approximation[2]=1028.000000
04-06 17:58:06.211  1897  1897 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[1080.000000,1136.000000]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[3]=1.000000, approximation[4]=1080.000000, approximation[5]=1136.000000
04-06 17:58:06.211  1897  1897 D skia    : approximate: numPoints=2, lengths[0]=0.000000, totalLength=108.000000, segmentPoints[0]=[1080.000000,1028.000000]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[0]=0.000000, approximation[1]=1080.000000, approximation[2]=1028.000000
04-06 17:58:06.211  1897  1897 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[1080.000000,1136.000000]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[3]=1.000000, approximation[4]=1080.000000, approximation[5]=1136.000000
04-06 17:58:06.211  1897  1897 D skia    : approximate: numPoints=2, lengths[0]=0.000000, totalLength=108.000000, segmentPoints[0]=[1080.000000,1028.000000]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[0]=0.000000, approximation[1]=1080.000000, approximation[2]=1028.000000
04-06 17:58:06.211  1897  1897 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[1080.000000,1136.000000]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[3]=1.000000, approximation[4]=1080.000000, approximation[5]=1136.000000
04-06 17:58:06.211  1897  1897 D skia    : approximate: numPoints=2, lengths[0]=0.000000, totalLength=108.000000, segmentPoints[0]=[1080.000000,1025.000000]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[0]=0.000000, approximation[1]=1080.000000, approximation[2]=1025.000000
04-06 17:58:06.211  1897  1897 D skia    : approximate: numPoints=2, lengths[1]=108.000000, totalLength=108.000000, segmentPoints[1]=[1080.000000,1133.000000]
04-06 17:58:06.211  1897  1897 D skia    : approximate: approximation[3]=1.000000, approximation[4]=1080.000000, approximation[5]=1133.000000
04-06 17:58:06.240  1897  1897 D ViewRootImpl@cfbe31[PopupWindow:9b7224a]: Relayout returned: old=[480,84][1068,516] new=[480,84][1068,516] result=0x1 surface={valid=true 500119281664} changed=false
[... cut ...]
04-06 17:58:06.298  1897  1897 D ViewRootImpl@cfbe31[PopupWindow:9b7224a]: Relayout returned: old=[480,84][1068,516] new=[480,84][1068,516] result=0x1 surface={valid=true 500119281664} changed=false
04-06 17:58:06.332  1897  1897 D ViewRootImpl@cfbe31[PopupWindow:9b7224a]: Relayout returned: old=[480,84][1068,516] new=[480,84][1068,516] result=0x1 surface={valid=true 500119281664} changed=false
04-06 17:58:07.161  1897  1897 D ViewRootImpl@cfbe31[PopupWindow:9b7224a]: ViewPostIme pointer 0
04-06 17:58:07.177  1897  1897 D ViewRootImpl@cfbe31[PopupWindow:9b7224a]: ViewPostIme pointer 1
04-06 17:58:07.177  1897  1897 D AbsListView: onTouchUp() mTouchMode : 0
```

Gedrean, 2022-04-06:
Hi, OP here (couldn't get it to login right on the app).

Behavior: TagMo crashes on Samsung S8 (non-root) after doing a search
Reproduce: Type in search, ie "Mario"
After about 3 seconds, when search results have displayed, app closes immediately.
Relaunching brings back to search, if search is not cleared, crash repeats.
Cannot get to logcat fast enough to send during that time.

Tried to grab a log after relaunch, hopefully it helps.

AbandonedCart, 2022-04-07:
The log and description originally provided were plenty to get going, but thanks for explaining further. It shouldn't take long to fix thanks to quality reporting.

commit 05cc4900 (2022-04-07): Stop sharing a reference to a dynamic instance

Fixes #466
closed 2022-04-07 by AbandonedCart