HiddenRamblings/TagMo #470: Bulk Write Amiibo's to N2 Elite

cyclikal94, 2022-04-10:
Hi - Is there any way to pre-load the banks for my N2 Elite with the Amiibo's I want, then in one go write them to each bank? Kinda like how the erase all banks goes through each bank?

AbandonedCart, 2022-04-10:
https://tagmo.gitlab.io/n2elite.html?s=writing

AbandonedCart, 2022-04-10:
Loading 10 copies of the same amiibo would load 10 of the same serial, leaving you to go back and change each one to write again. At that point, it saves time to simply generate the serial and write it. Due to the number of users that struggle with basic reading and writing, amiibo data is not cached between operations.

cyclikal94, 2022-04-10:
Ahh thank you I've re-read that a couple times now and finally figured it out! D'oh, I was confused thinking the 'Write X Banks to N2 Elite' actually changed the number of banks the N2 device had - but actually clicking it I can select multiple amiibo's! Thanks for pointing me in the right direction!

cyclikal94, 2022-04-10:
> Loading 10 copies of the same amiibo would load 10 of the same serial, leaving you to go back and change each one to write again. At that point, it saves time to simply generate the serial and write [...]

I think I've figured a way to be able to pre-create the bin's, so I have duplicates in the list with differing serials - maybe! That'll at least save me scanning between each one! Thanks again! :)

AbandonedCart, 2022-04-10:
Making them into separate bins is one option, but can easily get confusing.
 Changing the serial before saving will store a file with the new serial.

AbandonedCart, 2022-04-15:
After the current new features get moving, i'll circle back to improving bulk handling for this. I've kept it open because the current options aren't the most efficient.

AbandonedCart, 2022-06-14:
I think the best option here would be a fill option. That would be an option to select a single bin and make X copies with random serial numbers. That would allow one bin to exist without having to track all of the data, but still allow for making multiple copies with different serial numbers when writing it.

closed 2022-06-15 by AbandonedCart

hutje, 2023-05-24:
But why is it not possible to select multiple Amiibo's and write all Amiibo's in order to each bank?

AbandonedCart, 2023-05-24:
It is...

hutje, 2023-05-24:
> It is...

When press any Amiibo it wants to send only that amiibo to all banks

hutje, 2023-05-24:
I found the problem. When enabling random serial, it only gives 1 Amiibo of course...

random11x, 2023-06-29:
When I have random serial on and I choose "write to banks" the file selector pops up and doesnt let you choose anything. As soon as you go to choose a file it pops up with a dialog asking if you want to write the selected file (even though there are none selected) and then if you say yes it writes blank data to each bank.