HiddenRamblings/TagMo #477: Logcat - TagMo #b2dbcb4

LogcatGitHub, 2022-04-25:
*Submitted by info@*******.de*

Description:
---

Crash

Device info:
---
<table>
<tr><td>App version</td><td>2.8.0</td></tr>
<tr><td>App version code</td><td>13</td></tr>
<tr><td>Android build version</td><td>G780GXXU3CVC2</td></tr>
<tr><td>Android release version</td><td>12</td></tr>
<tr><td>Android SDK version</td><td>31</td></tr>
<tr><td>Android build ID</td><td>SP1A.210812.016.G780GXXU3CVC2</td></tr>
<tr><td>Device brand</td><td>samsung</td></tr>
<tr><td>Device manufacturer</td><td>samsung</td></tr>
<tr><td>Device name</td><td>r8q</td></tr>
<tr><td>Device model</td><td>SM-G780G</td></tr>
<tr><td>Device product name</td><td>r8qeea</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```TagMo - #b2dbcb4
Android S (12)

--------- beginning of main
04-26 01:03:04.518  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: MSG_WINDOW_FOCUS_CHANGED 0 1
04-26 01:03:04.763  9781  9781 D InputTransport: Input channel destroyed: 'ClientS', fd=114
04-26 01:03:04.769  9781  9781 D InsetsSourceConsumer: setRequestedVisible: visible=false, type=19, host=com.hiddenramblings.tagmo/com.heinrichreimersoftware.androidissuereporter.IssueReporterLauncher$Activity, from=android.view.InsetsSourceConsumer.hide:242 android.view.ImeInsetsSourceConsumer.hide:68 android.view.ImeInsetsSourceConsumer.hide:74 android.view.InsetsController.hideDirectly:1473 android.view.InsetsController.controlAnimationUnchecked:1139 android.view.InsetsController.applyAnimation:1456 android.view.InsetsController.applyAnimation:1437 android.view.InsetsController.hide:1006 android.view.ViewRootImpl$ViewRootHandler.handleMessageImpl:6461 android.view.ViewRootImpl$ViewRootHandler.handleMessage:6382 
04-26 01:03:04.773  9781  9781 D ScrollView:  onsize change changed 
04-26 01:03:04.802  9781  9781 D ScrollView:  onsize change changed 
04-26 01:03:05.146  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: handleAppVisibility mAppVisible=true visible=false
04-26 01:03:05.155  9781  9814 D OpenGLRenderer: setSurface called with nullptr
04-26 01:03:05.155  9781  9814 D OpenGLRenderer: setSurface() destroyed EGLSurface
04-26 01:03:05.156  9781  9814 D OpenGLRenderer: destroyEglSurface
04-26 01:03:05.163  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)8 dur=3 res=0x5 s={false 0} ch=true fn=231
04-26 01:03:05.164  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: stopped(true) old=false
04-26 01:03:05.176  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)8 dur=4 res=0x5 s={false 0} ch=false fn=-1
04-26 01:03:53.487  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: handleAppVisibility mAppVisible=false visible=true
04-26 01:03:53.494  9781  9814 D OpenGLRenderer: setSurface called with nullptr
04-26 01:03:53.507  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)4 dur=3 res=0x1 s={false 0} ch=false fn=-1
04-26 01:03:53.507  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: stopped(false) old=true
04-26 01:03:53.508  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: stopped(false) old=false
04-26 01:03:53.509  9781  9781 I DecorView: notifyKeepScreenOnChanged: keepScreenOn=false
04-26 01:03:53.510  9781  9814 D OpenGLRenderer: setSurface called with nullptr
04-26 01:03:53.526  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)0 dur=14 res=0x7 s={true -5476376605069848576} ch=true fn=-1
04-26 01:03:53.526  9781  9814 D OpenGLRenderer: eglCreateWindowSurface
04-26 01:03:53.528  9781  9781 D ScrollView:  onsize change changed 
04-26 01:03:53.528  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: [DP] dp(1) 1 android.view.ViewRootImpl.reportNextDraw:11374 android.view.ViewRootImpl.performTraversals:4167 android.view.ViewRootImpl.doTraversal:2893 
04-26 01:03:53.536  9781  9814 I BufferQueueProducer: [ViewRootImpl@92fd764[IssueReporterLauncher$Activity]#14(BLAST Consumer)14](id:26350000000e,api:1,p:9781,c:9781) queueBuffer: queued for the first time.
04-26 01:03:53.538  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: [DP] pdf(0) 1 android.view.ViewRootImpl.lambda$addFrameCompleteCallbackIfNeeded$3$ViewRootImpl:4969 android.view.ViewRootImpl$$ExternalSyntheticLambda16.run:6 android.os.Handler.handleCallback:938 
04-26 01:03:53.538  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: [DP] rdf()
04-26 01:03:53.538  9781  9781 D ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: reportDrawFinished (fn: -1) 
04-26 01:03:53.603  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: MSG_WINDOW_FOCUS_CHANGED 1 1
04-26 01:03:53.608  9781  9781 D InputMethodManager: startInputInner - Id : 0
04-26 01:03:53.608  9781  9781 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
04-26 01:03:53.720  9781  9781 D InsetsController: show(ime(), fromIme=true)
04-26 01:03:53.722  9781  9781 D InsetsSourceConsumer: setRequestedVisible: visible=true, type=19, host=com.hiddenramblings.tagmo/com.heinrichreimersoftware.androidissuereporter.IssueReporterLauncher$Activity, from=android.view.InsetsSourceConsumer.show:235 android.view.InsetsController.showDirectly:1489 android.view.InsetsController.controlAnimationUnchecked:1137 android.view.InsetsController.applyAnimation:1456 android.view.InsetsController.applyAnimation:1437 android.view.InsetsController.show:976 android.view.ViewRootImpl$ViewRootHandler.handleMessageImpl:6457 android.view.ViewRootImpl$ViewRootHandler.handleMessage:6382 android.os.Handler.dispatchMessage:106 android.os.Looper.loopOnce:226 
04-26 01:03:53.733  9781  9781 D ScrollView:  onsize change changed 
04-26 01:03:54.826  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: The input has been finished in ImeInputStage.
04-26 01:03:54.935  9781  9781 I ViewRootImpl@92fd764[IssueReporterLauncher$Activity]: The input has been finished in ImeInputStage.
04-26 01:03:54.937  9781  9781 D InsetsSourceConsumer: setRequestedVisible: visible=false, type=19, host=com.hiddenramblings.tagmo/com.heinrichreimersoftware.androidissuereporter.IssueReporterLauncher$Activity, from=android.view.InsetsSourceConsumer.hide:242 android.view.ImeInsetsSourceConsumer.hide:68 android.view.ImeInsetsSourceConsumer.hide:74 android.view.InsetsController.hideDirectly:1473 android.view.InsetsController.controlAnimationUnchecked:1139 android.view.InsetsController.applyAnimation:1456 android.view.InsetsController.applyAnimation:1437 android.view.InsetsController.hide:1006 android.view.ViewRootImpl$ViewRootHandler.handleMessageImpl:6461 android.view.ViewRootImpl$ViewRootHandler.handleMessage:6382 
04-26 01:03:54.945  9781  9781 D ScrollView:  onsize change changed 
[... 193 lines cut ...]
04-26 01:04:06.240 16949 16979 D NativeCustomFrequencyManager: [NativeCFMS] BpCustomFrequencyManager::BpCustomFrequencyManager()
04-26 01:04:06.247 16949 16949 I ViewRootImpl@e5cab85[PopupWindow:ff442ce]: setView = android.widget.PopupWindow$PopupDecorView@795bdda TM=true
04-26 01:04:06.252 16949 16962 I ViewRootImpl@e5cab85[PopupWindow:ff442ce]: Resizing android.view.ViewRootImpl@ec17b01: frame=[480,100][1068,676] reportDraw=true forceLayout=false backDropFrame=Rect(0, 0 - 588, 576)
04-26 01:04:06.255 16949 16949 I ViewRootImpl@e5cab85[PopupWindow:ff442ce]: Relayout returned: old=(0,88,1080,2256) new=(480,100,1068,676) req=(588,576)0 dur=4 res=0x7 s={true -5476376604364836864} ch=true fn=-1
04-26 01:04:06.255 16949 16979 D OpenGLRenderer: eglCreateWindowSurface
04-26 01:04:06.259 16949 16949 D AbsListView:  in onLayout changed 
04-26 01:04:06.260 16949 16949 I ViewRootImpl@e5cab85[PopupWindow:ff442ce]: [DP] dp(1) 0 android.view.ViewRootImpl.reportNextDraw:11374 android.view.ViewRootImpl.performTraversals:4167 android.view.ViewRootImpl.doTraversal:2893 
[... cut ...] false fn=33
04-26 01:05:24.891 16949 16949 I ViewRootImpl@e5cab85[PopupWindow:ff442ce]: ViewPostIme pointer 0
04-26 01:05:24.918 16949 16949 I ViewRootImpl@e5cab85[PopupWindow:ff442ce]: ViewPostIme pointer 1
04-26 01:05:24.920 16949 16949 D AbsListView: onTouchUp() mTouchMode : 0
04-26 01:05:25.002 16949 16949 I ViewRootImpl@e5cab85[PopupWindow:ff442ce]: Relayout returned: old=(480,100,1068,676) new=(480,100,1068,676) req=(588,576)0 dur=10 res=0x1 s={true -5476376604364836864} ch=false fn=41
```

Maximilian1021, 2022-04-25:
The app is crashing when you click on the Sync AmiiboAPI button.

AbandonedCart, 2022-04-25:
Duplicate of https://github.com/HiddenRamblings/TagMo/issues/476

closed 2022-04-26 by AbandonedCart