HiddenRamblings/TagMo #481: [Bug]: Purple codes

L3thalSequ3l, 2022-05-14:
### Issue Description

Purple code on certain amiibos. Unable to do anything with regards to those amiibos

### Logcat Output

```shell
TagMo - #fcd6b02
Android S (12)

--------- beginning of main
05-14 06:46:01.039 32640 32651 D InputTransport: Input channel destroyed: 'ClientS', fd=171
05-14 06:46:01.040 32640 32664 D OpenGLRenderer: setSurface called with nullptr
05-14 06:46:01.576 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 0
05-14 06:46:01.648 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 1
05-14 06:46:01.689 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: setView = android.widget.PopupWindow$PopupDecorView@409e0d7 TM=true
05-14 06:46:01.697 32640 32655 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: Resizing android.view.ViewRootImpl@f78a230: frame=[541,2101][1276,3001] reportDraw=true forceLayout=false backDropFrame=Rect(0, 0 - 735, 900)
05-14 06:46:01.697 32640 32640 D InsetsSourceConsumer: ensureControlAlpha: for ITYPE_NAVIGATION_BAR on PopupWindow:5f13ffb
05-14 06:46:01.698 32640 32640 D InsetsSourceConsumer: ensureControlAlpha: for ITYPE_STATUS_BAR on PopupWindow:5f13ffb
05-14 06:46:01.698 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: Relayout returned: old=(0,100,1440,3032) new=(541,2101,1276,3001) req=(735,900)0 dur=5 res=0x7 s={true -5476376650937052256} ch=true fn=-1
05-14 06:46:01.698 32640 32664 D OpenGLRenderer: eglCreateWindowSurface
05-14 06:46:01.704 32640 32640 D AbsListView:  in onLayout changed 
05-14 06:46:01.704 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: [DP] dp(1) 0 android.view.ViewRootImpl.reportNextDraw:11387 android.view.ViewRootImpl.performTraversals:4185 android.view.ViewRootImpl.doTraversal:2911 
05-14 06:46:01.705 32640 32640 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Creating frameDrawingCallback nextDrawUseBlastSync=false reportNextDraw=true hasBlurUpdates=false
05-14 06:46:01.705 32640 32640 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Creating frameCompleteCallback
05-14 06:46:01.706 32640 32691 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Received frameDrawingCallback frameNum=1. Creating transactionCompleteCallback=false
05-14 06:46:01.708 32640 32664 I BufferQueueProducer: [ViewRootImpl@58e818[PopupWindow:5f13ffb]#144(BLAST Consumer)144](id:7f8000000090,api:1,p:32640,c:32640) queueBuffer: queued for the first time.
05-14 06:46:01.709 32640 32664 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Received frameCompleteCallback  lastAcquiredFrameNum=1 lastAttemptedDrawFrameNum=1
05-14 06:46:01.711 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: [DP] pdf(0) 0 android.view.ViewRootImpl.lambda$addFrameCompleteCallbackIfNeeded$3$ViewRootImpl:4987 android.view.ViewRootImpl$$ExternalSyntheticLambda16.run:6 android.os.Handler.handleCallback:938 
05-14 06:46:01.711 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: [DP] rdf()
05-14 06:46:01.711 32640 32640 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: reportDrawFinished (fn: -1) 
05-14 06:46:01.713 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: handleResized, msg = 5 frames=ClientWindowFrames{frame=[541,2101][1276,3001] display=[-10000,-10000][10000,10000] backdrop=[0,0][735,900]} forceNextWindowRelayout=false displayId=0 frameChanged=false backdropFrameChanged=true configChanged=false displayChanged=false mNextDrawUseBlastSync=false mergedConfiguration={mGlobalConfig={1.0 310mcc280mnc [en_US] ldltr sw384dp w384dp h781dp 600dpi nrml long port night finger -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1440, 3088) mAppBounds=Rect(0, 100 - 1440, 3032) mMaxBounds=Rect(0, 0 - 1440, 3088) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_0 mPopOver=off mFreeformTaskPinningState=unpinned} s.3267 fontWeightAdjustment=0 ff=0 bf=0 bts=0 desktop/d dm/n dc/d ?dcui ?dcaf enb/d themeSeq=0} mOverrideConfig={1.0 310mcc280mnc [en_US] ldltr sw384dp w384dp h781dp 600dpi nrml long port night finger -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1440, 3088) mAppBounds=Rect(0, 100 - 1440, 3032) mMaxBounds=Rect(0, 0 - 1440, 3088) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=standard mAlwaysOnTop=undefined mRotation=ROTATION_0 mPopOver=off mFreeformTaskPinningState=unpinned} s.6 fontWeightAdjustment=0 ff=0 bf=0 bts=0 desktop/d dm/n dc/d ?dcui ?dcaf enb/d themeSeq=0}}
05-14 06:46:01.714 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: [DP] dp(1) 0 android.view.ViewRootImpl.reportNextDraw:11387 android.view.ViewRootImpl.handleResized:2222 android.view.ViewRootImpl.access$1200:294 
05-14 06:46:01.714 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
05-14 06:46:01.718 32640 32640 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Creating frameDrawingCallback nextDrawUseBlastSync=false reportNextDraw=true hasBlurUpdates=false
05-14 06:46:01.718 32640 32640 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Creating frameCompleteCallback
05-14 06:46:01.719 32640 32692 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Received frameDrawingCallback frameNum=3. Creating transactionCompleteCallback=false
05-14 06:46:01.720 32640 32664 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: Received frameCompleteCallback  lastAcquiredFrameNum=3 lastAttemptedDrawFrameNum=3
05-14 06:46:01.727 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: [DP] pdf(0) 0 android.view.ViewRootImpl.lambda$addFrameCompleteCallbackIfNeeded$3$ViewRootImpl:4987 android.view.ViewRootImpl$$ExternalSyntheticLambda16.run:6 android.os.Handler.handleCallback:938 
05-14 06:46:01.728 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: [DP] rdf()
05-14 06:46:01.728 32640 32640 D ViewRootImpl@58e818[PopupWindow:5f13ffb]: reportDrawFinished (fn: -1) 
05-14 06:46:01.737 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: MSG_WINDOW_FOCUS_CHANGED 1 1
05-14 06:46:02.757 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: ViewPostIme pointer 0
05-14 06:46:02.794 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: Relayout returned: old=(541,2101,1276,3001) new=(541,2101,1276,3001) req=(735,900)0 dur=17 res=0x1 s={true -5476376650937052256} ch=false fn=32
05-14 06:46:02.813 32640 32640 D InsetsSourceConsumer: ensureControlAlpha: for ITYPE_NAVIGATION_BAR on com.hiddenramblings.tagmo/com.hiddenramblings.tagmo.BrowserActivity
05-14 06:46:02.814 32640 32640 D InsetsSourceConsumer: ensureControlAlpha: for ITYPE_STATUS_BAR on com.hiddenramblings.tagmo/com.hiddenramblings.tagmo.BrowserActivity
05-14 06:46:02.816 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: MSG_WINDOW_FOCUS_CHANGED 0 1
05-14 06:46:02.820 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
05-14 06:46:02.822 32640 32640 D InputMethodManager: startInputInner - Id : 0
05-14 06:46:02.822 32640 32640 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
05-14 06:46:02.848 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: ViewPostIme pointer 1
05-14 06:46:03.105 32640 32664 D OpenGLRenderer: setSurface called with nullptr
05-14 06:46:03.105 32640 32664 D OpenGLRenderer: setSurface() destroyed EGLSurface
05-14 06:46:03.105 32640 32664 D OpenGLRenderer: destroyEglSurface
05-14 06:46:03.107 32640 32640 I ViewRootImpl@58e818[PopupWindow:5f13ffb]: dispatchDetachedFromWindow
05-14 06:46:03.112 32640 32640 D InputTransport: Input channel destroyed: 'aa6f2aa', fd=99
05-14 06:46:03.386 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 0
05-14 06:46:03.474 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 1
05-14 06:46:03.941 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 0
05-14 06:46:04.008 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 1
05-14 06:46:04.485 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 0
05-14 06:46:04.566 32640 32640 I ViewRootImpl@3d69475[BrowserActivity]: ViewPostIme pointer 1
05-14 06:46:04.610 32640 32640 I ViewRootImpl@a675854[PopupWindow:2069ba7]: setView = android.widget.PopupWindow$PopupDecorView@7f74b43 TM=true
05-14 06:46:04.620 32640 32655 I ViewRootImpl@a675854[PopupWindow:2069ba7]: Resizing android.view.ViewRootImpl@3d8923e: frame=[541,2101][1276,3001] reportDraw=true forceLayout=false backDropFrame=Rect(0, 0 - 735, 900)
[... cut ...]
05-14 06:46:17.822 32640 32640 I ViewRootImpl@59ffd8[PopupWindow:ecdebb]: ViewPostIme pointer 0
05-14 06:46:17.985 32640 32640 I ViewRootImpl@59ffd8[PopupWindow:ecdebb]: ViewPostIme pointer 1
05-14 06:46:17.985 32640 32640 D AbsListView: onTouchUp() mTouchMode : 2
05-14 06:46:18.004 32640 32640 I ViewRootImpl@59ffd8[PopupWindow:ecdebb]: Relayout returned: old=(690,115,1425,835) new=(690,115,1425,835) req=(735,720)0 dur=14 res=0x1 s={true -5476376650937052256} ch=false fn=40
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-05-14:
Those are unreleased / spoofed amiibo.

closed 2022-05-14 by AbandonedCart

L3thalSequ3l, 2022-05-15:
Gotcha, thank you!