HiddenRamblings/TagMo #486: Cannot scan N2 Elite

CAYCE-VII, 2022-06-03:


CAYCE-VII, 2022-06-03:
[image]

CAYCE-VII, 2022-06-03:
TagMo - #4d669b2
Android S_V2 (12)

--------- beginning of main
06-02 17:31:39.634 15314 15314 D InputMethodManager: showSoftInput() view=com.google.android.material.textfield.TextInputEditText{2cd90bd VFED.VCL. .F.P..ID 0,0-1272,231 #7f09004c app:id/air_inputDescription aid=1073741825} flags=0 reason=SHOW_SOFT_INPUT
06-02 17:31:39.645 15314 15314 I AssistStructure: Flattened final assist data: 5100 bytes, containing 1 windows, 27 views
06-02 17:31:39.690 15314 15314 D InsetsController: show(ime(), fromIme=true)
06-02 17:31:53.489 15314 15339 D OpenGLRenderer: endAllActiveAnimators on 0xb400007e215d1e60 (RippleDrawable) with handle 0xb400007d31515b90
06-02 17:32:51.603 15314 20198 W IOException: java.io.IOException: /storage/emulated/0/Documents/key_retail.bin size: 160, Expected: 532
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.getTagData(TagReader.java:35)
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readTagFile(TagReader.java:45)
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagUtils.getValidatedFile(TagUtils.java:267)
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:298)
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:295)
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity.lambda$loadAmiiboFiles$24(BrowserActivity.java:1315)
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity.$r8$lambda$FVwueBdhBCAV0uk4D1bcJ0ktJBc(Unknown Source:0)
06-02 17:32:51.603 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity$$ExternalSyntheticLambda40.run(Unknown Source:6)
06-02 17:32:51.603 15314 20198 W IOException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
06-02 17:32:51.603 15314 20198 W IOException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
06-02 17:32:51.603 15314 20198 W IOException: 	at java.lang.Thread.run(Thread.java:920)
06-02 17:32:52.311 15314 20198 W IOException: java.io.IOException: /storage/emulated/0/.$Trash$/.$Content$/.4106865493382056216_0/Amiibo Bin Dump/key_retail.bin size: 160, Expected: 532
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.getTagData(TagReader.java:35)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readTagFile(TagReader.java:45)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagUtils.getValidatedFile(TagUtils.java:267)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:298)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:295)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:295)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:295)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:295)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity.lambda$loadAmiiboFiles$24(BrowserActivity.java:1315)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity.$r8$lambda$FVwueBdhBCAV0uk4D1bcJ0ktJBc(Unknown Source:0)
06-02 17:32:52.311 15314 20198 W IOException: 	at com.hiddenramblings.tagmo.BrowserActivity$$ExternalSyntheticLambda40.run(Unknown Source:6)
06-02 17:32:52.311 15314 20198 W IOException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
06-02 17:32:52.311 15314 20198 W IOException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
06-02 17:32:52.311 15314 20198 W IOException: 	at java.lang.Thread.run(Thread.java:920)
06-02 17:32:53.762 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:32:53.763 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:32:53.763 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:32:53.763 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:32:55.486 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
06-02 17:32:55.487 15314 15314 E ramblings.tagm: Invalid ID 0x00000000.
[... cut ...]
06-02 17:33:41.889 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:33:52.411 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:33:52.411 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:33:52.411 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()
06-02 17:33:52.411 15314 15314 W TextView: onProvideContentCaptureStructure(): calling assumeLayout()

AbandonedCart, 2022-06-03:
Did you enable N2 in settings?

CAYCE-VII, 2022-06-03:
Yes, it's shown as enabled in the video link I posted. I have also cleared appdata, and re-granted permissions and re-enabled N2 in settings, restarted my device, locked and unlocked the N2 in the N2 app itself.

I looked at these closed issues as well:

https://github.com/HiddenRamblings/TagMo/issues/409
https://github.com/HiddenRamblings/TagMo/issues/404 - seems the closest
https://github.com/HiddenRamblings/TagMo/issues/368
https://github.com/HiddenRamblings/TagMo/issues/359
https://github.com/HiddenRamblings/TagMo/issues/252

CAYCE-VII, 2022-06-03:
Hmm, after some fiddling around, that error happens when you scan the N2 while on an empty bank. Luckily I still had a Bank that had data on it, after setting that as the active bank in the N2 app itself, hopping over to Tagmo and scanning again, it pulls up the file picker

CAYCE-VII, 2022-06-03:
No worries, had my N2 had no banks with data on it I wouldn't have been able to get around it. Is it notated or documented anywhere that the N2 can't scan on an empty bank and I just missed it? Also you can close this issue.

closed 2022-06-03 by AbandonedCart

AbandonedCart, 2022-06-03:
There is the initial inconvenience of confirming it's an N2 when the active bank is empty, but only when entering the bank browser. It will skip verifying again until the window is closed.

commit 998483ed (2022-06-03): Support handling a blank active tag for N2

Fixes #486