HiddenRamblings/TagMo #507: [Bug]:

EndahCookieh, 2022-07-06:
### Issue Description

Attempt to invoke virtual method 'voidcom.hiddinramblings.tagmo.nfctech.NTAG215.connect()' on a null object reference.

Every time i scan or try to write into a tag this error pops up , could I have a fix?

Thank you

### Logcat Output

```shell
Attempt to invoke virtual method 'voidcom.hiddinramblings.tagmo.nfctech.NTAG215.connect()' on a null object reference.

Every time i scan or try to write into a tag this error pops up , could I have a fix?

Thank you
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-07-06:
The phone probably doesn't support NFC. It's hard to say based on just an error message.

AbandonedCart, 2022-07-06:
Make sure to update.

EndahCookieh, 2022-07-10:
Update, The issue was with the tags, I bought new ones and it seemed to work , just from a different provider

closed 2022-07-10 by AbandonedCart