HiddenRamblings/TagMo #512: [Bug]: Cant write any tags with correct ntag215

Kobra299, 2022-07-19:
### Issue Description

[image]

every time I try to write to the ntag it shows this error no matter what I try I have used the NXP Tag Info to make sure it a true ntag215 which it says it is see the txt in this any help is welcome

[04-81-2B-01-78-02-03_2022-07-19 21-35-54_taginfo_scan.txt](https://github.com/HiddenRamblings/TagMo/files/9144605/04-81-2B-01-78-02-03_2022-07-19.21-35-54_taginfo_scan.txt)

### Logcat Output

```shell
#db97a5c
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-07-19:
Please add the log provided by the app or submit a new issue from within the app. It's in the top right menu as "Report Problem"

Kobra299, 2022-07-19:
Submitted by mewthree19@*****.com
Description:

Issue with writing to ntag
Device info:
App version	2.9.8
App version code	298
Android build version	00WW_5_22D
Android release version	9
Android SDK version	28
Android build ID	00WW_5_22D
Device brand	Nokia
Device manufacturer	HMD Global
Device name	NE1
Device model	TA-1020
Device product name	TA-1020_00WW
Device hardware name	mt6735
ABIs	[arm64-v8a, armeabi-v7a, armeabi]
ABIs (32bit)	[armeabi-v7a, armeabi]
ABIs (64bit)	[arm64-v8a]
Logcat:

TagMo - #db97a5c
Android P (9)

--------- beginning of crash
07-19 14:19:42.950  8676  8676 E AndroidRuntime: FATAL EXCEPTION: main
07-19 14:19:42.950  8676  8676 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 8676
07-19 14:19:42.950  8676  8676 E AndroidRuntime: java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=685555169, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.hiddenramblings.tagmo/com.hiddenramblings.tagmo.browser.BrowserActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.startScan(java.util.List, android.bluetooth.le.ScanSettings, android.bluetooth.le.ScanCallback)' on a null object reference
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.ActivityThread.deliverResults(ActivityThread.java:4391)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.ActivityThread.handleSendResult(ActivityThread.java:4433)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:49)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:68)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1836)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:193)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6702)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.startScan(java.util.List, android.bluetooth.le.ScanSettings, android.bluetooth.le.ScanCallback)' on a null object reference
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment.scanBluetoothServices(FlaskSlotFragment.java:615)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment.selectBluetoothDevice(FlaskSlotFragment.java:647)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment.onAdapterEnabled(FlaskSlotFragment.java:917)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.hiddenramblings.tagmo.eightbit.bluetooth.BluetoothEnabler.lambda$new$4(BluetoothEnabler.java:149)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.hiddenramblings.tagmo.eightbit.bluetooth.BluetoothEnabler.$r8$lambda$p3XBAUlYKh6ZD19Wt3ObgwuOZ5w(Unknown Source:0)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at com.hiddenramblings.tagmo.eightbit.bluetooth.BluetoothEnabler$$ExternalSyntheticLambda4.onActivityResult(Unknown Source:8)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:392)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:351)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at androidx.activity.ComponentActivity.onRequestPermissionsResult(ComponentActivity.java:667)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at androidx.fragment.app.FragmentActivity.onRequestPermissionsResult(FragmentActivity.java:612)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:7608)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.Activity.dispatchActivityResult(Activity.java:7458)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	at android.app.ActivityThread.deliverResults(ActivityThread.java:4384)
07-19 14:19:42.950  8676  8676 E AndroidRuntime: 	... 11 more
07-19 15:42:38.369 10012 10012 E AndroidRuntime: FATAL EXCEPTION: main
07-19 15:42:38.369 10012 10012 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 10012
07-19 15:42:38.369 10012 10012 E AndroidRuntime: java.lang.ArrayIndexOutOfBoundsException: length=835; index=835
07-19 15:42:38.369 10012 10012 E AndroidRuntime: 	at java.util.ArrayList.clear(ArrayList.java:569)
[... cut ...]
07-19 21:48:06.103 16234 16240 I ramblings.tagm: Background concurrent copying GC freed 54772(1949KB) AllocSpace objects, 2(9MB) LOS objects, 21% free, 21MB/27MB, paused 206us total 239.077ms
07-19 21:48:07.645 16234 16234 D ViewRootImpl[PopupWindow:4a995ff]: hardware acceleration = true , fakeHwAccelerated = false, sRendererDisabled = false, forceHwAccelerated = false, sSystemRendererDisabled = false
07-19 21:48:07.697 16234 16268 D mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000

@Kobra299

AbandonedCart, 2022-07-20:
Looks like it may be related to Android 9 compatibility. I'll need to investigate.

AbandonedCart, 2022-07-20:
https://github.com/HiddenRamblings/TagMo/commit/3b8c53b80dc583c05eb3b7fefaf0ce9a26a1fdcb addresses all the issues in the log, but it may not address the tag issue.

closed 2022-07-22 by AbandonedCart