HiddenRamblings/TagMo #521: Logcat - TagMo #6fa3a08

LogcatGitHub, 2022-07-26:
*Submitted by pado9757@*****.com*

Description:
---

same problem as a few updates ago, shortly after tagmo has connected to the flask, the app crashes , unless I already have at least 2 amiibo on it beforehand

Device info:
---
<table>
<tr><td>App version</td><td>3.0.3</td></tr>
<tr><td>App version code</td><td>303</td></tr>
<tr><td>Android build version</td><td>V12.5.8.0.RJZEUXM</td></tr>
<tr><td>Android release version</td><td>11</td></tr>
<tr><td>Android SDK version</td><td>30</td></tr>
<tr><td>Android build ID</td><td>RKQ1.200826.002 test-keys</td></tr>
<tr><td>Device brand</td><td>Redmi</td></tr>
<tr><td>Device manufacturer</td><td>Xiaomi</td></tr>
<tr><td>Device name</td><td>joyeuse</td></tr>
<tr><td>Device model</td><td>Redmi Note 9 Pro</td></tr>
<tr><td>Device product name</td><td>joyeuse_eea</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```
TagMo - #6fa3a08
Android R (11)
Installed by release

--------- beginning of main
07-27 01:36:19.792  5994  6014 W FlaskGattService: FlaskTX onCharacteristicWrite 0
07-27 01:36:19.813  5994  6014 W FlaskGattService: FlaskRX  - Amiibo Fe|ic7wi|0
07-27 01:36:19.814  5994  6014 W FlaskGattService: FlaskRX ")
07-27 01:36:19.924  5994  6014 W FlaskGattService: FlaskRX ={ freeBytes: 62868 
07-27 01:36:19.935  5994  6014 W FlaskGattService: FlaskRX }
07-27 01:36:19.935  5994  6014 W FlaskGattService: >
07-27 01:36:20.062  5994  6014 W FlaskGattService: FlaskTX onCharacteristicWrite 0
07-27 01:36:20.093  5994  6014 W FlaskGattService: FlaskRX tag.getList()
07-27 01:36:20.093  5994  6014 W FlaskGattService: =[ 
07-27 01:36:20.105  5994  6014 W FlaskGattService: FlaskRX   "Malzeno|15pptu|0"
07-27 01:36:20.116  5994  6014 W FlaskGattService: FlaskRX , 
07-27 01:36:20.116  5994  6014 W FlaskGattService:   "DJ KK|2qcqq|0
07-27 01:36:20.123  5994  6014 W FlaskGattService: FlaskTX onCharacteristicWrite 0
07-27 01:36:20.126  5994  6014 W FlaskGattService: FlaskRX "
07-27 01:36:20.126  5994  6014 W FlaskGattService:  ]
07-27 01:36:20.126  5994  6014 W FlaskGattService: >
07-27 01:36:20.151  5994  6014 W FlaskGattService: FlaskTX onCharacteristicWrite 0
07-27 01:36:20.159  5994  6014 W FlaskGattService: FlaskRX tag.remove("Stitches
07-27 01:36:20.171  5994  6014 W FlaskGattService: FlaskRX  - A
07-27 01:36:20.183  5994  6014 W FlaskGattService: FlaskRX miibo Fe|ic7wi|0
07-27 01:36:20.184  5994  6014 W FlaskGattService: FlaskTX onCharacteristicWrite 0
07-27 01:36:20.215  5994  6016 W FlaskGattService: FlaskRX ")
07-27 01:36:20.287  5994  6016 W FlaskGattService: FlaskTX onCharacteristicWrite 0
07-27 01:36:20.319  5994  6016 W FlaskGattService: FlaskRX ={ freeBytes: 62868 
07-27 01:36:20.329  5994  6016 W FlaskGattService: FlaskRX }
07-27 01:36:20.329  5994  6016 W FlaskGattService: >tag.getList()
07-27 01:36:20.329  5994  6016 W FlaskGattService: =
07-27 01:36:20.341  5994  6016 W FlaskGattService: FlaskRX [ 
07-27 01:36:20.341  5994  6016 W FlaskGattService:   "Malzeno|15ppt
07-27 01:36:20.342  5994  6016 W FlaskGattService: FlaskTX onCharacteristicWrite 0
07-27 01:36:20.353  5994  6014 W FlaskGattService: FlaskRX u|0", 
07-27 01:36:20.353  5994  6014 W FlaskGattService:   "DJ KK|2qc
07-27 01:36:20.364  5994  6014 W FlaskGattService: FlaskRX qq|0"
07-27 01:36:20.364  5994  6014 W FlaskGattService:  ]
07-27 01:36:20.364  5994  6014 W FlaskGattService: >
[... 297 lines cut ...]
07-27 01:36:49.925  6813  6847 W IOException: 	at java.lang.Thread.run(Thread.java:923)
07-27 01:36:49.956  6813  6858 W IOException: java.io.IOException: /storage/emulated/0/Download/key_retail.bin size: 160, Expected: 532
07-27 01:36:49.956  6813  6858 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.getTagData(TagReader.java:35)
07-27 01:36:49.956  6813  6858 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readTagFile(TagReader.java:45)
07-27 01:36:49.956  6813  6858 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagUtils.getValidatedFile(TagUtils.java:327)
07-27 01:36:49.956  6813  6858 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:298)
07-27 01:36:49.956  6813  6858 W IOException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.lambda$loadAmiiboFiles$38(BrowserActivity.java:1671)
07-27 01:36:49.956  6813  6858 W IOException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.$r8$lambda$evbC1JPDNjEkXH9X6qlHi4IkMlA(Unknown Source:0)
07-27 01:36:49.956  6813  6858 W IOException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity$$ExternalSyntheticLambda73.run(Unknown Source:6)
07-27 01:36:49.956  6813  6858 W IOException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
07-27 01:36:49.956  6813  6858 W IOException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
07-27 01:36:49.956  6813  6858 W IOException: 	at java.lang.Thread.run(Thread.java:923)
07-27 01:36:50.032  6813  6872 W IOException: java.io.IOException: /storage/emulated/0/Download/key_retail.bin size: 160, Expected: 532
07-27 01:36:50.032  6813  6872 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.getTagData(TagReader.java:35)
07-27 01:36:50.032  6813  6872 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readTagFile(TagReader.java:45)
07-27 01:36:50.032  6813  6872 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagUtils.getValidatedFile(TagUtils.java:327)
07-27 01:36:50.032  6813  6872 W IOException: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:298)
07-27 01:36:50.032  6813  6872 W IOException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.lambda$loadAmiiboFiles$38(BrowserActivity.java:1671)
07-27 01:36:50.032  6813  6872 W IOException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.$r8$lambda$evbC1JPDNjEkXH9X6qlHi4IkMlA(Unknown Source:0)
07-27 01:36:50.032  6813  6872 W IOException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity$$ExternalSyntheticLambda73.run(Unknown Source:6)
07-27 01:36:50.032  6813  6872 W IOException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
07-27 01:36:50.032  6813  6872 W IOException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
07-27 01:36:50.032  6813  6872 W IOException: 	at java.lang.Thread.run(Thread.java:923)
07-27 01:36:50.143  6813  6813 I TetheringManager: registerTetheringEventCallback:com.hiddenramblings.tagmo
```

AbandonedCart, 2022-07-26:
Google Play is a bit slow to publish updates and will delay the process further when submitting new versions before the previous ones are finished review.

Pado975, 2022-07-26:
I use the GitHub Version

AbandonedCart, 2022-07-26:
Then you should see an update soon that will resolve it.

Pado975, 2022-07-27:
It works now , thank you 
But i have a new Problem:
tagmo gives me the info that there is a new update although I have the latest version and when I click on update I installed an older version (6fa3a08)
[image]

AbandonedCart, 2022-07-27:
https://github.com/HiddenRamblings/TagMo/actions/runs/2743250225

AbandonedCart, 2022-07-27:
Once the next build starts, it will fix itself. It should happen in about half an hour.

closed 2022-07-27 by AbandonedCart