HiddenRamblings/TagMo #523: Update prompt

serratedhalo, 2022-07-27:
I'm on github version 2.7.8 which appears to be the most recent stable version.  Each time I open the app I get a prompt "Do you want to update this app?"  I tried toggling "stable update checks" checked and unchecked in the settings menu but they both result in the same prompt when the app is opened.

AbandonedCart, 2022-07-27:
2.7.8 is like a year old.

serratedhalo, 2022-07-27:
Ah on the releases tab it looks like the most current stable version, but I'm guessing that's because the app store version just updates in the top "master" slot?

AbandonedCart, 2022-07-27:
Google Play requires each new release have a separate incremental version number, so GitHub uses the hash of the commit for versioning.

serratedhalo, 2022-07-27:
Makes sense, thanks for the great app.

closed 2022-07-27 by serratedhalo