HiddenRamblings/TagMo #525: Logcat - TagMo #d5e701b

LogcatGitHub, 2022-08-01:
*Submitted by briansieu@*****.com*

Description:
---

app crashes when attempting to disable automatic nfc scan.

- pixel 6 pro user

Device info:
---
<table>
<tr><td>App version</td><td>3.0.0</td></tr>
<tr><td>App version code</td><td>300</td></tr>
<tr><td>Android build version</td><td>8671607</td></tr>
<tr><td>Android release version</td><td>12</td></tr>
<tr><td>Android SDK version</td><td>32</td></tr>
<tr><td>Android build ID</td><td>SQ3A.220705.003</td></tr>
<tr><td>Device brand</td><td>google</td></tr>
<tr><td>Device manufacturer</td><td>Google</td></tr>
<tr><td>Device name</td><td>raven</td></tr>
<tr><td>Device model</td><td>Pixel 6 Pro</td></tr>
<tr><td>Device product name</td><td>raven</td></tr>
<tr><td>Device hardware name</td><td>raven</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```
TagMo - #d5e701b
Android S_V2 (12)
Installed by publish

--------- beginning of main
08-01 17:16:12.419 18474 18474 W .tagmo.eightbi: JIT profile information will not be recorded: profile file does not exist.
08-01 17:16:12.419 18474 18474 W .tagmo.eightbi: JIT profile information will not be recorded: profile file does not exist.
08-01 17:16:12.419 18474 18474 W .tagmo.eightbi: JIT profile information will not be recorded: profile file does not exist.
08-01 17:16:12.419 18474 18474 W .tagmo.eightbi: JIT profile information will not be recorded: profile file does not exist.
08-01 17:16:12.423 18474 18474 D NetworkSecurityConfig: No Network Security Config specified, using platform default
08-01 17:16:12.423 18474 18474 D NetworkSecurityConfig: No Network Security Config specified, using platform default
08-01 17:16:12.432 18474 18493 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/lib/arm64'
08-01 17:16:12.432 18474 18493 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/base.apk!/lib/arm64-v8a'
08-01 17:16:12.432 18474 18493 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/split_config.arm64_v8a.apk!/lib/arm64-v8a'
08-01 17:16:12.432 18474 18493 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/split_config.en.apk!/lib/arm64-v8a'
08-01 17:16:12.432 18474 18493 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/split_config.ja.apk!/lib/arm64-v8a'
08-01 17:16:12.432 18474 18493 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/split_config.xxxhdpi.apk!/lib/arm64-v8a'
08-01 17:16:12.434 18474 18493 D mali_cmarp_predictor: checking cmar_predictor for com.hiddenramblings.tagmo.eightbit
08-01 17:16:12.456 18474 18474 D CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10413; state: ENABLED
08-01 17:16:12.477 18474 18474 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService()
08-01 17:16:12.478 18474 18474 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService() returned 0x0
08-01 17:16:12.478 18474 18474 D RenderScript HIDL Adaptation: Using Fallback Path.
08-01 17:16:12.482 18474 18474 D .tagmo.eightbi: Successfully queried cache dir: /data/user_de/0/com.hiddenramblings.tagmo.eightbit/code_cache
08-01 17:16:12.482 18474 18474 D RenderScript: Setting cache dir: /data/user_de/0/com.hiddenramblings.tagmo.eightbit/code_cache
08-01 17:16:12.488 18474 18517 W AmiiboDocument: Primary doc=amiibo, mime=vnd.android.document/directory
08-01 17:16:12.498 18474 18474 I PlayCore: UID: [10413]  PID: [18474] AppUpdateService : requestUpdateInfo(com.hiddenramblings.tagmo.eightbit)
08-01 17:16:12.498 18474 18526 W AmiiboDocument: Primary doc=amiibo, mime=vnd.android.document/directory
08-01 17:16:12.499 18474 18529 I PlayCore: UID: [10413]  PID: [18474] AppUpdateService : Initiate binding to the service.
08-01 17:16:12.515 18474 18474 I PlayCore: UID: [10413]  PID: [18474] AppUpdateService : ServiceConnectionImpl.onServiceConnected(ComponentInfo{com.android.vending/com.google.android.finsky.installservice.DevTriggeredUpdateService})
08-01 17:16:12.516 18474 18529 I PlayCore: UID: [10413]  PID: [18474] AppUpdateService : linkToDeath
08-01 17:16:12.531 18474 18490 I PlayCore: UID: [10413]  PID: [18474] OnRequestInstallCallback : onRequestInfo
08-01 17:16:12.531 18474 18529 I PlayCore: UID: [10413]  PID: [18474] AppUpdateService : Unbind from service.
08-01 17:16:12.547 18474 18541 W AmiiboDocument: Primary doc=amiibo, mime=vnd.android.document/directory
08-01 17:16:12.566 18474 18474 W View    : requestLayout() improperly called by android.widget.FrameLayout{7b42e74 V.E..V... ......ID 0,0-1440,2639 #2} during layout: running second layout pass
08-01 17:16:12.566 18474 18474 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{f974f9d VFED.V... ......ID 0,1358-1440,2635 #7f09010e app:id/foomiibo_list} during layout: running second layout pass
08-01 17:16:12.566 18474 18474 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{e1cec12 VFED.V... ......ID 0,0-1440,1260 #7f09005f app:id/amiibos_list} during layout: running second layout pass
08-01 17:16:14.431 18474 18484 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai
08-01 17:16:14.431 18474 18484 I BpBinder: onLastStrongRef automatically unlinking death recipients: <uncached descriptor>
[... 259 lines cut ...]
08-01 17:16:28.697 18681 18699 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/split_config.en.apk!/lib/arm64-v8a'
08-01 17:16:28.697 18681 18699 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/split_config.ja.apk!/lib/arm64-v8a'
08-01 17:16:28.697 18681 18699 D vulkan  : searching for layers in '/data/app/~~7TR2R6wh5zEgCMrZ5wmMBA==/com.hiddenramblings.tagmo.eightbit-ic3xrmnXMxSXhZ79LDlgYw==/split_config.xxxhdpi.apk!/lib/arm64-v8a'
08-01 17:16:28.699 18681 18699 D mali_cmarp_predictor: checking cmar_predictor for com.hiddenramblings.tagmo.eightbit
08-01 17:16:28.717 18681 18681 D CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10413; state: ENABLED
08-01 17:16:28.739 18681 18681 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService()
08-01 17:16:28.740 18681 18681 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService() returned 0x0
08-01 17:16:28.740 18681 18681 D RenderScript HIDL Adaptation: Using Fallback Path.
08-01 17:16:28.744 18681 18681 D .tagmo.eightbi: Successfully queried cache dir: /data/user_de/0/com.hiddenramblings.tagmo.eightbit/code_cache
08-01 17:16:28.744 18681 18681 D RenderScript: Setting cache dir: /data/user_de/0/com.hiddenramblings.tagmo.eightbit/code_cache
08-01 17:16:28.754 18681 18722 W AmiiboDocument: Primary doc=amiibo, mime=vnd.android.document/directory
08-01 17:16:28.762 18681 18729 W AmiiboDocument: Primary doc=amiibo, mime=vnd.android.document/directory
08-01 17:16:28.762 18681 18681 I PlayCore: UID: [10413]  PID: [18681] AppUpdateService : requestUpdateInfo(com.hiddenramblings.tagmo.eightbit)
08-01 17:16:28.763 18681 18731 I PlayCore: UID: [10413]  PID: [18681] AppUpdateService : Initiate binding to the service.
08-01 17:16:28.781 18681 18681 I PlayCore: UID: [10413]  PID: [18681] AppUpdateService : ServiceConnectionImpl.onServiceConnected(ComponentInfo{com.android.vending/com.google.android.finsky.installservice.DevTriggeredUpdateService})
08-01 17:16:28.781 18681 18731 I PlayCore: UID: [10413]  PID: [18681] AppUpdateService : linkToDeath
08-01 17:16:28.796 18681 18696 I PlayCore: UID: [10413]  PID: [18681] OnRequestInstallCallback : onRequestInfo
08-01 17:16:28.796 18681 18731 I PlayCore: UID: [10413]  PID: [18681] AppUpdateService : Unbind from service.
[... cut ...]
08-01 17:16:28.840 18681 18681 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{e1cec12 VFED.V... ......ID 0,0-1440,1260 #7f09005f app:id/amiibos_list} during layout: running second layout pass
08-01 17:16:30.738 18681 18691 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai
08-01 17:16:30.739 18681 18691 I BpBinder: onLastStrongRef automatically unlinking death recipients: <uncached descriptor>
```

AbandonedCart, 2022-08-01:
Currently blocked by https://github.com/HiddenRamblings/TagMo/issues/522

AbandonedCart, 2022-08-05:
I would recommend considering a conversion to the GitHub version. It requires removing the one from Google Play first, but obviously won't suffer from the obnoxious delays.

closed 2022-08-06 by AbandonedCart