HiddenRamblings/TagMo #528: [Bug]: Crashing immediately upon opening

PlazmaDolphin, 2022-08-04:
### Issue Description

I'm not sure if this is an issue on my part or with the app, but I tried opening it for the first time on a Samsung S4 mini. It flashed a few times, then crashed. I downloaded the sdk to an SD card with very limited space (64 MB), and installed it right after formatting the phone's memory.

### Logcat Output

```shell
N/A
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-08-04:
How to determine the cause of a crash is almost always from the log, which was (ironically) marked not applicable (or N/A).

AbandonedCart, 2022-08-04:
Crashes that cause the application to close, also called exceptions, are passed through a section of the log labeled 'AndroidRuntimeException'. They will list the type of exception, such as a 'NullPointerException' (variable being read is null). More importantly, they will list the specific file and even the line number that caused the error. As you might have guessed, this means we not only know what is wrong, but exactly where to find it. Instead of a tedious process attempting to recreate the exact scenario to reproduce the issue and then trying to eliminate unlikely causes until possible fixes can be attempted, the entire task is reduced to mere minutes. This is why logs are marked required.

PlazmaDolphin, 2022-08-04:
Oooh I didn't know what logcat meant, so sorry. Where can I find the error file, I'm not familiar with Android file system.

AbandonedCart, 2022-08-05:
Since it's an instant crash, the feature to retrieve the log probably won't be available. Was it downloaded from Google Play or GitHub?

PlazmaDolphin, 2022-08-05:
I put the sdk from github on a microSD using my computer, then put the microSD in the phone and installed it. I figured out how to logcat in the meantime, and got about 2000 lines of debug from the 5 seconds it ran, which I put into this pastebin: https://pastebin.com/ZtTZape5

AbandonedCart, 2022-08-05:
Thanks. In the future, the logcat should be available in the clipboard even if the app crashes instantly, which is what happens when using the menu option now. I thought that had already been added, but apparently it got lost somewhere in other changes.

closed 2022-08-05 by AbandonedCart