HiddenRamblings/TagMo #529: [Bug]: Loss os amiibo, app crash.

k3ir44, 2022-08-04:
### Issue Description

Amiibos in card have been lost and app keeps crashing.

### Logcat Output

```shell
TagMo - #d5e701b
Android Q (10)
Installed by publish

--------- beginning of crash
08-04 15:01:24.106  2041 11711 E AndroidRuntime: FATAL EXCEPTION: pool-12-thread-1
08-04 15:01:24.106  2041 11711 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 2041
08-04 15:01:24.106  2041 11711 E AndroidRuntime: java.lang.RuntimeException: Can't create handler inside thread Thread[pool-12-thread-1,5,main] that has not called Looper.prepare()
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:207)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:119)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at android.app.Dialog.<init>(Dialog.java:162)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at android.app.Dialog.<init>(Dialog.java:247)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at androidx.appcompat.app.AppCompatDialog.<init>(AppCompatDialog.java:57)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog.<init>(AlertDialog.java:98)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog$Builder.create(AlertDialog.java:983)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog$Builder.show(AlertDialog.java:1008)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:588)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source:4)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
08-04 15:01:24.106  2041 11711 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:919)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: FATAL EXCEPTION: pool-12-thread-1
08-04 15:01:33.545  6091 12972 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 6091
08-04 15:01:33.545  6091 12972 E AndroidRuntime: java.lang.RuntimeException: Can't create handler inside thread Thread[pool-12-thread-1,5,main] that has not called Looper.prepare()
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:207)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:119)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at android.app.Dialog.<init>(Dialog.java:162)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at android.app.Dialog.<init>(Dialog.java:247)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at androidx.appcompat.app.AppCompatDialog.<init>(AppCompatDialog.java:57)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog.<init>(AlertDialog.java:98)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog$Builder.create(AlertDialog.java:983)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog$Builder.show(AlertDialog.java:1008)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:588)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source:4)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
08-04 15:01:33.545  6091 12972 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:919)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: FATAL EXCEPTION: pool-14-thread-1
08-04 15:01:50.631 12061 15157 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 12061
08-04 15:01:50.631 12061 15157 E AndroidRuntime: java.lang.RuntimeException: Can't create handler inside thread Thread[pool-14-thread-1,5,main] that has not called Looper.prepare()
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:207)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:119)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at android.app.Dialog.<init>(Dialog.java:162)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at android.app.Dialog.<init>(Dialog.java:247)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at androidx.appcompat.app.AppCompatDialog.<init>(AppCompatDialog.java:57)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog.<init>(AlertDialog.java:98)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog$Builder.create(AlertDialog.java:983)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at androidx.appcompat.app.AlertDialog$Builder.show(AlertDialog.java:1008)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:588)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source:4)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
08-04 15:01:50.631 12061 15157 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:919)
08-04 15:02:11.847 13131 18477 E AndroidRuntime: FATAL EXCEPTION: pool-14-thread-1
08-04 15:02:11.847 13131 18477 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 13131
08-04 15:02:11.847 13131 18477 E AndroidRuntime: java.lang.RuntimeException: Can't create handler inside thread Thread[pool-14-thread-1,5,main] that has not called Looper.prepare()
08-04 15:02:11.847 13131 18477 E AndroidRuntime: 	at android.os.Handler.<init>(Handler.java:207)
[... 209 lines cut ...]
08-04 15:06:51.782 19289 19289 D InputMethodManager: startInputInner - Id : 0
08-04 15:06:51.782 19289 19289 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
08-04 15:06:52.607 19289 19289 I ViewRootImpl@a49835a[BrowserActivity]: ViewPostIme pointer 0
08-04 15:06:52.646 19289 19289 I ViewRootImpl@a49835a[BrowserActivity]: ViewPostIme pointer 1
08-04 15:06:52.702 19289 19289 I ViewRootImpl@b2f375d[PopupWindow:ae35e46]: setView = android.widget.PopupWindow$PopupDecorView@4e718d2 TM=true MM=false
08-04 15:06:52.720 19289 19289 I ViewRootImpl@b2f375d[PopupWindow:ae35e46]: Relayout returned: old=(0,72,1080,2076) new=(480,84,1068,660) req=(588,576)0 dur=7 res=0x7 s={true 486709239808} ch=true
08-04 15:06:52.730 19289 19289 D AbsListView:  in onLayout changed 
08-04 15:06:52.741 19289 19289 I ViewRootImpl@b2f375d[PopupWindow:ae35e46]: MSG_WINDOW_FOCUS_CHANGED 1 1
08-04 15:06:52.746 19289 19289 I ViewRootImpl@b2f375d[PopupWindow:ae35e46]: MSG_RESIZED: frame=(480,84,1068,660) ci=(0,0,0,0) vi=(0,0,0,0) or=1
08-04 15:06:52.747 19289 19289 I ViewRootImpl@a49835a[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
08-04 15:06:52.747 19289 19289 D InputMethodManager: prepareNavigationBarInfo() DecorView@a446fce[BrowserActivity]
08-04 15:06:52.747 19289 19289 D InputMethodManager: getNavigationBarColor() -855310
[... cut ...]
08-04 15:06:53.725 19289 19289 I ViewRootImpl@a49835a[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
08-04 15:06:53.725 19289 19289 D InputMethodManager: prepareNavigationBarInfo() DecorView@a446fce[BrowserActivity]
08-04 15:06:53.725 19289 19289 D InputMethodManager: getNavigationBarColor() -855310
08-04 15:06:53.731 19289 19289 I ViewRootImpl@b2f375d[PopupWindow:ae35e46]: MSG_WINDOW_FOCUS_CHANGED 0 1
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-08-04:
Blocked by https://github.com/HiddenRamblings/TagMo/issues/522

AbandonedCart, 2022-08-05:
I would recommend considering a conversion to the GitHub version. It requires removing the one from Google Play first, but obviously won't suffer from the obnoxious delays.

closed 2022-08-06 by AbandonedCart