HiddenRamblings/TagMo #531: [Bug]: App still DOA for me

PlazmaDolphin, 2022-08-07:
### Issue Description

Same circumstances as bug #528, only this time I was able to logcat from clipboard, so we know that works now. The log may be a bit truncated on some parts, though.

### Logcat Output

```shell
android.view.InflateException: Binary XML file line #50: Error inflating class <unknown>
	at android.view.LayoutInflater.createView(LayoutInflater.java:626)
	at com.android.internal.policy.impl.PhoneLayoutInflater.onCreateView(PhoneLayoutInflater.java:56)
	at android.view.LayoutInflater.onCreateView(LayoutInflater.java:675)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:700)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:761)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:769)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:769)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:498)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:398)
	at com.hiddenramblings.tagmo.browser.BrowserFragment.onCreateView(BrowserFragment.java:100)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:3104)
	at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:524)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:261)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1890)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1808)
	at androidx.fragment.app.FragmentManager.execSingleAction(FragmentManager.java:1720)
	at androidx.fragment.app.BackStackRecord.commitNow(BackStackRecord.java:317)
	at androidx.viewpager2.adapter.FragmentStateAdapter.placeFragmentInViewHolder(FragmentStateAdapter.java:341)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:276)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:67)
	at androidx.recyclerview.widget.RecyclerView.dispatchChildAttached(RecyclerView.java:7867)
	at androidx.recyclerview.widget.RecyclerView$5.addView(RecyclerView.java:893)
	at androidx.recyclerview.widget.ChildHelper.addView(ChildHelper.java:107)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addViewInt(RecyclerView.java:8902)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:8860)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:8848)
	at androidx.recyclerview.widget.LinearLayoutManager.layoutChunk(LinearLayoutManager.java:1645)
	at androidx.recyclerview.widget.LinearLayoutManager.fill(LinearLayoutManager.java:1591)
	at androidx.recyclerview.widget.LinearLayoutManager.onLayoutChildren(LinearLayoutManager.java:668)
	at androidx.recyclerview.widget.RecyclerView.dispatchLayoutStep2(RecyclerView.java:4309)
	at androidx.recyclerview.widget.RecyclerView.dispatchLayout(RecyclerView.java:4012)
	at androidx.recyclerview.widget.RecyclerView.onLayout(RecyclerView.java:4578)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at androidx.viewpager2.widget.ViewPager2.onLayout(ViewPager2.java:527)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.layoutChild(CoordinatorLayout.java:1213)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayoutChild(CoordinatorLayout.java:899)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayout(CoordinatorLayout.java:919)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1677)
	at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1531)
	at android.widget.LinearLayout.onLayout(LinearLayout.java:1440)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at androidx.drawerlayout.widget.DrawerLayout.onLayout(DrawerLayout.java:1263)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:453)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:388)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at androidx.appcompat.widget.ActionBarOverlayLayout.onLayout(ActionBarOverlayLayout.java:536)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:453)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:388)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1677)
	at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1531)
	at android.widget.LinearLayout.onLayout(LinearLayout.java:1440)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:453)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:388)
	at android.view.View.layout(View.java:15762)
	at android.view.ViewGroup.layout(ViewGroup.java:4880)
	at android.view.ViewRootImpl.performLayout(ViewRootImpl.java:2356)
	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2069)
	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1254)
	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6624)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:812)
	at android.view.Choreographer.doCallbacks(Choreographer.java:612)
	at android.view.Choreographer.doFrame(Choreographer.java:582)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:798)
	at android.os.Handler.handleCallback(Handler.java:733)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:146)
	at android.app.ActivityThread.main(ActivityThread.java:5602)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:515)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1283)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1099)
	at dalvik.system.NativeStart.main(Native Method)
Caused by: java.lang.reflect.InvocationTargetException
	at java.lang.reflect.Constructor.constructNative(Native Method)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at android.view.LayoutInflater.createView(LayoutInflater.java:600)
	... 86 more
Caused by: android.content.res.Resources$NotFoundException: File res/T7.xml from drawable resource ID #0x7f08007f
	at android.content.res.Resources.loadDrawable(Resources.java:3430)
	at android.content.res.TypedArray.getDrawable(TypedArray.java:602)
	at android.view.View.<init>(View.java:3708)
	at android.view.ViewGroup.<init>(ViewGroup.java:480)
	at android.widget.LinearLayout.<init>(LinearLayout.java:176)
	at android.widget.LinearLayout.<init>(LinearLayout.java:172)
	... 89 more
Caused by: java.lang.UnsupportedOperationException: Can't convert to color: type=0x2
	at android.content.res.TypedArray.getColor(TypedArray.java:327)
	at android.graphics.drawable.GradientDrawable.inflate(GradientDrawable.java:970)
	at android.graphics.drawable.Drawable.createFromXmlInner(Drawable.java:990)
	at android.graphics.drawable.Drawable.createFromXml(Drawable.java:930)
	at android.content.res.Resources.loadDrawable(Resources.java:3426)
	... 94 more
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-08-07:
What Android version is it? It looks like the issue is a view that isn't supported. Could be a library not checking API restrictions.

PlazmaDolphin, 2022-08-07:
It probably has something to do with it, cause my phone is *OLD*. I'm not sure how old version 4.4.2 is, but that's what it's on and it tells me it's up to date. Hardware version is L520.08, if it makes a difference.

AbandonedCart, 2022-08-07:
Looks like the tabs and stuff are pulled from the material library instead of compat. This will take a minute.

AbandonedCart, 2022-08-07:
So it seems tab layouts date back just one version after that one and Google's supposed "compatibility" libraries don't actually support it, only remove the error that it doesn't exist.

All is not lost, though. By removing the tabs for that version (keeping all the functionality minus the neat little bar for quick navigation), it should work. It will be just the slightest bit less convenient and a touch less pretty, but it should work.

closed 2022-08-08 by AbandonedCart

AbandonedCart, 2022-08-08:
The commit closes the issue, but is it working now? I wanted to make sure before moving forward.

PlazmaDolphin, 2022-08-10:
It does work now! I wrote an Isabelle onto one of my tags (although I had to turn off NTAG215 validation) and it works perfectly on smash as far as I can tell.