HiddenRamblings/TagMo #532: About amiibo.json

qazrfv1234, 2022-08-08:
Hello,
If available, which part may attached with translation behind the original strings and keep filter working like before? 

ex:
1."amiibo_series" ： "Super Smash Bros任天堂明星大亂鬥",
2."amiibos"："name": "Mario瑪利歐"
3."characters"："Luigi路易吉"
4."types" ： "Card卡片"

AbandonedCart, 2022-08-08:
That's processed from the actual amiibo API project.

closed 2022-08-08 by qazrfv1234

AbandonedCart, 2022-08-08:
If AmiiboAPI ever does add translation support, it is possible to have TagMo pull the appropriate file based on the device settings. I also know the project is one developer with a LOT going on and adding that sort of support would be a HUGE task. It's not something I would ever ask or suggest asking because of the amount of work required.