HiddenRamblings/TagMo #537: Donation banner

Miss1sle, 2022-08-12:
The donation banner appears to be loading ever other load or so. Is this intentional?

It pops up right over the folder navigation leading to UI navigation errors.

I understand wanting to monitize your work but the placement and frequency seems aggressive.

AbandonedCart, 2022-08-12:
It's scheduled to run every fifth run as a self-dismissed snackbar.

Miss1sle, 2022-08-12:
Can we have it somewhere other than over the naivation elements?

AbandonedCart, 2022-08-12:
It appears at the bottom. Navigation is positioned at the top.

Miss1sle, 2022-08-12:
If I want to change folders I need access to the up arrow along the bottom.

AbandonedCart, 2022-08-12:
The duration of the notice is 3.5 seconds.

AbandonedCart, 2022-08-12:
Ironically, I believe changing folders frequently may be triggering a reset that isn't ignored, causing the banner to appear more frequently. Since folder selection is often a "set it and forget it" option, it may have slipped through without being disregarded (similar to refreshing the list for certain settings changes).

Miss1sle, 2022-08-12:
Idk maybe I'm using the app weird. But all my files are split by game into folders.

AbandonedCart, 2022-08-12:
Sorting by game is missing and really should be an option. Game series is the actual game origin of the character, so Min Min is ARMS when ideally it's SSBU. Not sure if a parameter exists for that, but part of the donation campaign was mirroring AmiiboAPI and adding stuff like that. It's free to host at the moment, but I'm not sure how much bandwidth I get before that starts being a monthly fee.

I'll look into why the banner is getting displayed too often and see about anchoring the bottom bar so it won't get covered. That seems fair enough. The idea was subtle without being invisible.

Miss1sle, 2022-08-12:
Mirroring the api seems unnecessary, the amiibo api is the standard place for amiibo data. If the data needs to be more granular resources should be given to that project rather than splitting and adding cost and division.

Just my opinion.

AbandonedCart, 2022-08-13:
The official API only accepts final release data. This means either those amiibo show up as unsupported or updating API data has to be done with a full TagMo build.

As for what is accepted, those contributions can be checked using https://github.com/N3evin/AmiiboAPI/commits?author=AbandonedCart 

I fully understand the desire for a completely free premium experience that considers every platform and user equally. It's also a lot to ask of one person.

Miss1sle, 2022-08-13:
I understand your perspective, but TagMo is (was) an app that backed up released Amiibo for people.  So I think it's perfectly understandable that it doesn't support unreleased Amiibo.  What happens if donations aren't enough and you get slapped with a giant server bill and the server goes down? Will it fail over to the regular amiibo API?  It seems like a risky prospect which could greatly devalue the app if anything happens to the mirror.

I obviously appreciate the work that's been done over the last year, adding Flask support is extremely impressive, and the N2 support has been great.

> It's not about profit, but rather not going into debt trying to keep the project afloat.

I find this more than a little worrisome.  TagMo is a standalone app which shouldn't have any reoccurring running costs.  No one should be going into debt working on free software, nor be worrying that they got enough donations that month to pay server costs.

I've donated to the project before, and I'd do it again to remove the donation notification, displaying for 3.5 seconds or not, i hit it constantly because the folder system is how I navigate my files.

I understand not everyone is willing to pay for apps, and you've said you want to keep the software as a free project so maybe that's off the table as an option.  But in terms of usability it's taken a dive for me, I now have to dodge the donation bar and the foomibo banner.  There's already a donation link every time you hit the menu icon it feels like there was already a subtle indication that you accept donations.

I understand I'm not everyone, my experience isn't everyone's, but I still feel like I need to make my voice heard.  Thanks for reading and all your work.

AbandonedCart, 2022-08-13:
I don't predict being slapped with a huge server bill. Heroku will shut off the service if bandwidth runs out and then offer the option to purchase more allotment. It's not like Google that will simply start charging per request and leave figuring out how much the overages are when the month ends.

I think you are factoring out one major component, though. This is not a job. That means every hour I spend trying to justify a passing notice that donations accepted is another hour not working, not spent with family, not asleep. Sure, I could just as easily leave it for when I have a few minutes. N2, Flask, the guides, and all the other things would still get done over the course of a year or two. Bugs would get fixed in a matter of months. TagMo would not be leading the community in functionality without being a paid app or having any scan limitations (see nearly any iOS clone).

TagMo has received $35 in donations for 2021 and $6 for 2022. The assumption was that perhaps the icon in settings was a little too hidden, since there has never been a lack of issues and requests. Again, it's not about trying to profit like my iOS counterparts. It's literally about having the (possibly misguided) belief that all the work is actually valued.

The "foomiibo banner", as you called it, is actually an entire second set of available data. It provides a legal alternative to what is likely pirated files in many cases. What you call a dive is one of the most valuable features to thousands of others. Placing it in a separate page made it unstable and / or prone to abuse. The next best thing was allowing it to be dragged down to nothing more than a second bar.

Saying you appreciate what is done while berating the fact that I tried to find a minimally intrusive way of saying I need some help keeping it going is a bit of a contradiction. I get you don't like it, but I can live with that.

AbandonedCart, 2022-08-17:
As promised, the notification trigger was tested and adjusted. In addition, the original 10 run offset was restored because it was not possible to cleanly move where it appeared (as demonstrated in commit history). An option to disable the Foomiibo was also added. While it is not uncommon for there to be that one person who opposes donations, this does appear to have been the work of someone with malicious intent. This is the only reasonable explanation to opt into early testing with no desire to provide feedback, but then submit multiple issues after the feature went live and then post a rant about donations only to delete the account when it did not get the desired result.

For archival purposes, "ghost" refers to the account previously registered as Miss1sle

closed 2022-08-17 by AbandonedCart