HiddenRamblings/TagMo #540: Logcat - TagMo #e9cf532a

LogcatGitHub, 2022-08-15:
*Submitted by mikeyt940@*****.com*

Description:
---

Turning off automatic tag scanning freezes the app, have to close out completely before app becomes responsive again. Also, it auto scans any tags even if the option is turned off.

Device info:
---
<table>
<tr><td>App version</td><td>3.0.4</td></tr>
<tr><td>App version code</td><td>304</td></tr>
<tr><td>Android build version</td><td>G973USQU6IVE7</td></tr>
<tr><td>Android release version</td><td>12</td></tr>
<tr><td>Android SDK version</td><td>31</td></tr>
<tr><td>Android build ID</td><td>SP1A.210812.016.G973USQU6IVE7</td></tr>
<tr><td>Device brand</td><td>samsung</td></tr>
<tr><td>Device manufacturer</td><td>samsung</td></tr>
<tr><td>Device name</td><td>beyond1q</td></tr>
<tr><td>Device model</td><td>SM-G973U</td></tr>
<tr><td>Device product name</td><td>beyond1qltesq</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```
TagMo - #e9cf532a
Android S (12)
Installed by publish

--------- beginning of main
08-15 14:54:47.508 26095 26142 W AmiiboDocument: Primary doc=Tagmo, mime=vnd.android.document/directory
08-15 14:54:47.518 26095 26095 I PlayCore: UID: [10542]  PID: [26095] AppUpdateService : requestUpdateInfo(com.hiddenramblings.tagmo.eightbit)
08-15 14:54:47.519 26095 26150 I PlayCore: UID: [10542]  PID: [26095] AppUpdateService : Initiate binding to the service.
08-15 14:54:47.520 26095 26147 W AmiiboDocument: Primary doc=Tagmo, mime=vnd.android.document/directory
08-15 14:54:47.545 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: setView = com.android.internal.policy.DecorView@3280e78 TM=true
08-15 14:54:47.553 26095 26095 I PlayCore: UID: [10542]  PID: [26095] AppUpdateService : ServiceConnectionImpl.onServiceConnected(ComponentInfo{com.android.vending/com.google.android.finsky.installservice.DevTriggeredUpdateService})
08-15 14:54:47.554 26095 26150 I PlayCore: UID: [10542]  PID: [26095] AppUpdateService : linkToDeath
08-15 14:54:47.563 26095 26121 I PlayCore: UID: [10542]  PID: [26095] OnRequestInstallCallback : onRequestInfo
08-15 14:54:47.563 26095 26150 I PlayCore: UID: [10542]  PID: [26095] AppUpdateService : Unbind from service.
08-15 14:54:47.590 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: Relayout returned: old=(0,0,1440,3040) new=(0,0,1440,3040) req=(1440,3040)0 dur=9 res=0x7 s={true -5476376658497377568} ch=true fn=-1
08-15 14:54:47.595 26095 26129 D OpenGLRenderer: eglCreateWindowSurface
08-15 14:54:47.602 26095 26165 W AmiiboDocument: Primary doc=Tagmo, mime=vnd.android.document/directory
08-15 14:54:47.630 26095 26095 W View    : requestLayout() improperly called by android.widget.FrameLayout{1df42f7 V.E..V... ......ID 0,0-1440,2442 #2} during layout: running second layout pass
08-15 14:54:47.630 26095 26095 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{480f564 VFED.V... ......ID 0,247-1440,2438 #7f09010f app:id/foomiibo_list} during layout: running second layout pass
08-15 14:54:47.630 26095 26095 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{50523cd VFED.V... ......ID 0,0-1440,149 #7f09005f app:id/amiibos_list} during layout: running second layout pass
08-15 14:54:47.631 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: [DP] dp(1) 1 android.view.ViewRootImpl.reportNextDraw:11374 android.view.ViewRootImpl.performTraversals:4167 android.view.ViewRootImpl.doTraversal:2893 
08-15 14:54:47.649 26095 26129 I BufferQueueProducer: [ViewRootImpl@7a7043e[BrowserActivity]#0(BLAST Consumer)0](id:65ef00000000,api:1,p:26095,c:26095) queueBuffer: queued for the first time.
08-15 14:54:47.652 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: [DP] pdf(0) 1 android.view.ViewRootImpl.lambda$addFrameCompleteCallbackIfNeeded$3$ViewRootImpl:4969 android.view.ViewRootImpl$$ExternalSyntheticLambda16.run:6 android.os.Handler.handleCallback:938 
08-15 14:54:47.652 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: [DP] rdf()
08-15 14:54:47.652 26095 26095 D ViewRootImpl@7a7043e[BrowserActivity]: reportDrawFinished (fn: -1) 
08-15 14:54:47.737 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
08-15 14:54:47.738 26095 26095 D InputMethodManager: startInputInner - Id : 0
08-15 14:54:47.738 26095 26095 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
08-15 14:54:47.829 26095 26095 D InputMethodManager: startInputInner - Id : 0
08-15 14:54:48.597 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: ViewPostIme pointer 0
08-15 14:54:48.631 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: ViewPostIme pointer 1
08-15 14:54:49.432 26095 26110 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai
08-15 14:54:49.432 26095 26110 I BpBinder: onLastStrongRef automatically unlinking death recipients: <uncached descriptor>
08-15 14:54:49.756 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: ViewPostIme pointer 0
08-15 14:54:49.808 26095 26095 I ViewRootImpl@7a7043e[BrowserActivity]: ViewPostIme pointer 1
08-15 14:54:49.825 26095 26095 D AndroidRuntime: Shutting down VM
--------- beginning of crash
08-15 14:54:49.825 26095 26095 E AndroidRuntime: FATAL EXCEPTION: main
08-15 14:54:49.825 26095 26095 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 26095
08-15 14:54:49.825 26095 26095 E AndroidRuntime: java.lang.IllegalArgumentException: Component class com.hiddenramblings.tagmo.eightbit.NFCIntentFilter does not exist in com.hiddenramblings.tagmo.eightbit
[... 173 lines cut ...]
08-15 14:55:14.539 26260 26260 D InputTransport: Input channel destroyed: 'fe4011c', fd=114
08-15 14:55:14.814 26260 26260 I ViewRootImpl@b261fec[BrowserActivity]: ViewPostIme pointer 0
08-15 14:55:14.910 26260 26260 I ViewRootImpl@b261fec[BrowserActivity]: ViewPostIme pointer 1
08-15 14:55:14.962 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: setView = android.widget.PopupWindow$PopupDecorView@bae81b8 TM=true
08-15 14:55:14.970 26260 26277 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: Resizing android.view.ViewRootImpl@ee7faf6: frame=[740,164][1426,836] reportDraw=true forceLayout=false backDropFrame=Rect(0, 0 - 686, 672)
08-15 14:55:14.972 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: Relayout returned: old=(0,150,1440,2872) new=(740,164,1426,836) req=(686,672)0 dur=5 res=0x7 s={true -5476376658497351408} ch=true fn=-1
08-15 14:55:14.972 26260 26295 D OpenGLRenderer: eglCreateWindowSurface
08-15 14:55:14.976 26260 26260 D AbsListView:  in onLayout changed 
08-15 14:55:14.977 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: [DP] dp(1) 0 android.view.ViewRootImpl.reportNextDraw:11374 android.view.ViewRootImpl.performTraversals:4167 android.view.ViewRootImpl.doTraversal:2893 
08-15 14:55:15.026 26260 26295 I BufferQueueProducer: [ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]#3(BLAST Consumer)3](id:669400000003,api:1,p:26260,c:26260) queueBuffer: queued for the first time.
08-15 14:55:15.049 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: [DP] pdf(0) 0 android.view.ViewRootImpl.lambda$addFrameCompleteCallbackIfNeeded$3$ViewRootImpl:4969 android.view.ViewRootImpl$$ExternalSyntheticLambda16.run:6 android.os.Handler.handleCallback:938 
08-15 14:55:15.049 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: [DP] rdf()
08-15 14:55:15.049 26260 26260 D ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: reportDrawFinished (fn: -1) 
08-15 14:55:15.071 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: handleResized, msg = 5 frames=ClientWindowFrames{frame=[740,164][1426,836] display=[-10000,-10000][10000,10000] backdrop=[0,0][686,672]} forceNextWindowRelayout=false displayId=0 frameChanged=false backdropFrameChanged=true configChanged=false displayChanged=false mNextDrawUseBlastSync=false mergedConfiguration={mGlobalConfig={1.0 310mcc410mnc [en_US] ldltr sw411dp w411dp h777dp 560dpi nrml long port night finger -keyb/v/h -nav/h winConfig={ mBounds=Rect(0, 0 - 1440, 3040) mAppBounds=Rect(0, 149 - 1440, 2872) mMaxBounds=Rect(0, 0 - 1440, 3040) mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivity [... cut ...]
08-15 14:55:15.139 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: MSG_WINDOW_FOCUS_CHANGED 1 1
08-15 14:55:16.224 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: ViewPostIme pointer 0
08-15 14:55:16.306 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: ViewPostIme pointer 1
08-15 14:55:16.306 26260 26260 D AbsListView: onTouchUp() mTouchMode : 0
08-15 14:55:16.386 26260 26260 I ViewRootImpl@75cbc1b[PopupWindow:1e1e4cc]: Relayout returned: old=(740,164,1426,836) new=(740,164,1426,836) req=(686,672)0 dur=9 res=0x1 s={true -5476376658497351408} ch=false fn=15
```

AbandonedCart, 2022-08-15:
Probably another one of Google's ridiculous changes on what permissions are required for a task and whether or not they can be granted without first making it tedious.

AbandonedCart, 2022-08-15:
Looks like the issue was the required package name conversion for Google Play. It will be fixed in the next update.

AbandonedCart, 2022-08-15:
Fixed in the code, pending review and publishing on Google.

closed 2022-08-15 by AbandonedCart