HiddenRamblings/TagMo #548: Puck.js support

DanTheMan827, 2022-08-24:
The puck.js with a [custom script](https://dantheman827.github.io/ntag215-puck.js/) can be made to simulate an NTAG215 tag.

I've implemented Bluetooth support into the script so that it can be controlled and modified over a Bluetooth connection, much like the flask can.

I can provide all the documentation necessary, but I'm not familiar with the TagMo code to implement support for it myself.

## Bluetooth Protocol

The Bluetooth protocol involves sending commands to one characteristic, while reading the data from another. The command and return characteristics.

A full WebBluetooth implementation of everything in TypeScript can be found in [puck.ts](https://github.com/DanTheMan827/ntag215-puck.js/blob/master/puck-ntag215-manager/src/puck.ts)

**BLE Service:** `78290001-d52e-473f-a9f4-f03da7c67dd1`  
**Command Characteristic:** `78290002-d52e-473f-a9f4-f03da7c67dd1`  
**Return Characteristic:** `78290003-d52e-473f-a9f4-f03da7c67dd1`  
**Name Characteristic:** `78290004-d52e-473f-a9f4-f03da7c67dd1`

Writing to the name characteristic will change the bluetooth name the puck.js identifies as.

## Bluetooth flow

### Basic Information
- The first step is to connect to the device, then request the slot information.
- Next you would loop and request the slot information with the slot specified for the total amount of slots.

### Reading
To read the contents of a slot, you would need to call the read command three times.

Below is an example to read the entire 572 bytes of slot 1:
```javascript
0x02, 0x00, 0x00, 0x3F

// Read the data and store to a buffer at index 0

0x02, 0x00, 0x3F, 0x3F

// Read the data and store to a buffer at index 252

0x02, 0x00, 0x7E, 0x11

// Read the data and store to a buffer at index 504
```

### Writing
The tag as stored in the puck is 572 bytes, to write a tag you should write at least 540 bytes, with 572 being recommended.

To write the tag to a slot, you need to iterate over the tag buffer in 16 byte chunks while sending the write command with each chunk of data.

To write the first 16 bytes to slot 1, you would send the command `0x03, 0x00, 0x00, Data...`

To write the next 16 bytes to slot 1, you would send the command `0x03, 0x00, 0x04, Data...`

And continue on until all of the data has been written.

After writing, you should send the command `0x04, <Current Slot>` followed by `0xFF` if the current slot is the one you just wrote to.

### Changing the active slot
You can change the active slot by sending the command `0xFF, Slot`, this will cause the puck to flash if it is within the first 7 slots.

## Commands

**0x01 - Slot Information \<Slot>**

If a slot is specified, a subset of the tag will be returned that can be reconstructed and decrypted to get the amiibo information (like nickname and character).

These 80 bytes are sliced from the full tag with the following code.
```javascript
var output = Uint8Array(80);
output.set(tags[slot].buffer.slice(0, 8), 0);
output.set(tags[slot].buffer.slice(16, 24), 8);
output.set(tags[slot].buffer.slice(32, 52), 20);
output.set(tags[slot].buffer.slice(84, 92), 40);
output.set(tags[slot].buffer.slice(96, 128), 48);
```

Example: `0x01, Slot, <80 bytes>`

If a slot is not specified, it will return `0x01, Current Slot, Slot Count`

**0x02 - Read \<Slot> \<StartPage> \<PageCount> \<Data...>**
This is used to read the raw tag data from the puck memory, one page is 4 bytes, and the most you can request at a time is 63 pages.

Data returned is `0x02, Slot, Start Page, Page Count, Data...`

**0x03 - Write \<Slot> \<StartPage>, \<Data...>**
The start page can be anything between 0-142, the maximum amount of data that can be sent at once is 16 bytes, this does not need to be divisible by 4.

Data returned is `0x03, Slot, Start Page, Data...`

**0x04 - Save \<Slot>**
If SAVE_TO_FLASH is enabled in the script, this function will write the data in that slot to flash storage.  You probably want to run this after writing a tag.

Data returned is `0x04, Slot`

**0xFD - Move \<Source Slot> \<Destination Slot>**
This moves the slot from the source to the desination index.

Data returned is `0xFD, Source Slot, Destination Slot`

**0xFE - Enable UART**
This method enables UART so that the Espruino IDE can connect again, you need to disconnect after calling this.

**0xFF - Restart NFC \<Slot>**
This restarts the NFC with the slot specified, or the current slot if none was provided.

This is required if the current slot has been overwritten.

Data returned is `0xFF, Slot`

AbandonedCart, 2022-08-24:
After the train wreck that happened attempting to implement Flask support ahead of the actual device arriving, there is no chance of that happening again. It will have to wait until I can have one delivered, but there also doesn't appear to be a huge demand on either project. The Flask (and the N2) had a rather large demand for support. This is still fundamentally in the testing stages. Omniibo even goes so far as to state that the firmware is unstable. I will need to think it over.

AbandonedCart, 2022-08-25:
I think the best way to decide this would be to leave it to the community.  I would still need to get the hardware and install the software, but wouldn't be able to start doing that until the end of the month anyway. That should be enough time to get an idea of whether it is even something that will be used.

AbandonedCart, 2022-08-29:
This doesn't appear to be gaining any traction, but we can circle back to it if that changes in the future. For now, the focus has been to clean up TagMo and get it stable. Perhaps after that is complete, the necessity will be less of a factor.

AbandonedCart, 2022-11-04:
[image]

I'm not sure I understand why you would try to sabotage donations that would make your own request possible. It is just as beneficial to officially drop any plans to support this. You just freed up $30+ in funding. Thank you.

DanTheMan827, 2022-11-04:
You've had a problem with me for a long time, and I have no idea why.

If you want to continue working on the project, continue... but if you don't want to because it isn't profitable no one is forcing you to.

It seems like you're becoming less interested in this as a hobby overall if you don't want to continue developing it, but if that's the case just say so.

AbandonedCart, 2022-11-04:
If you want to be an asshole, then expect to be treated like one. I won't block you and I will still review your issues and requests in an unbiased manner. If you break the rules, you will be treated the same as anyone else. That is the part you need to accept. You do not have a golden ticket here.

DanTheMan827, 2022-11-04:
I never expected special treatment, nor did I ask for it, and if I offended you on reddit I apologize (as I have before), but as far as I was aware, you never made a post, and this is what I see when I click into the one that was linked
[image]

I figured it was someone spamming tagmo's donation links that was removed for some reason by the mods.

DanTheMan827, 2022-11-04:
In regards to the UI of tagmo, and bluetooth devices, you could have a standard UI for connecting to bluetooth devices.

Have a common "connect to bluetooth device" button or whatever, have it scan for all the different bluetooth devices supported, and show them in a combined list.

Then have an "interface" or whatever java calls it that the different implementations have to conform to.

Have an implementation for flask, this, n2, and whatever might pop up

Just a suggestion

AbandonedCart, 2022-11-04:
I know the whole donation campaign has some up in arms. My total earnings for the month from **all** of my projects is roughly $250. That covers internet, games, hardware, prescriptions (medicine), and anything else I need that isn't food and rent. I have a choice to make. Either I change jobs and can't spare the time or I have to find a way for this to cover the time it takes up. 

A template was added and maintained alongside the Flask code until such time as I could get hardware to test it. After two failed attempts to support hardware before I had it in hand, a third seemed like the definition of insanity.

I'll be ordering the device soon. I have to wait on stuff to clear so I can start reinvesting it back into the project.

DanTheMan827, 2022-11-04:
If it helps, the puck is the same hardware as the flask, and also runs the same espruino firmware, so you could use scripts designed for the puck.js with https://espruino.com/ide

The latest code requires some modifications to the firmware to improve NFC compatibility, but there are older commits that ran purely on the base epsruino firmware.

Again, I'm sorry, I get how hard it can be when you're just barely getting by, and I was a bit of an ass about it...

AbandonedCart, 2022-11-04:
I make it a point not to tamper with hardware that was donated to prevent becoming a variable in debugging. The hardware should arrive in a week, so it's not a huge setback.

DanTheMan827, 2022-11-04:
Fair enough, and if you don't want to add support for this I understand.

The hope for the bluetooth API was partially that any new projects might end up using a compatible implementation of it similarly to how some other devices emulate the N2, that's why it's fairly minimal and didn't use the nordic uart interface, and the one fork of my script does continue to use the same bluetooth API, but with a different implementation of the emulation code.

AbandonedCart, 2022-11-07:
The hardware ships out today and is estimated to arrive in 5 to 10 days. It should be early enough to avoid most of the early Christmas shopping delays.

AbandonedCart, 2022-11-10:
Just to prevent any confusion, the reason @withgallantry is (retroactively) on the copyright notices for the Puck (and Flask) files is code authorship. It should have been added when the licensing was converted, but was overlooked. It doesn't change much for usage and all that, but gives proper credit for a significant amount of code I had no part in writing that was brought over from the Flask.

AbandonedCart, 2022-11-14:
Support contacted me today and updated the estimated arrival to a minimum of 10 days, so it appears I did hit some of the shipping delays.

AbandonedCart, 2022-11-18:
> 0x01 - Slot Information <Slot>
> ...
> If a slot is not specified, it will return 0x01, Current Slot, Slot Count

I don't get a return on this one, which is the main holdup.

DanTheMan827, 2022-11-18:
Some commands may take longer to process and return data.

The way I go about it is to write a value and keep trying to read the return characteristic until it starts with the command I sent to the command characteristic.

puck should be sending notifications if you subscribe to the characteristic though, although I don't use it like that myself

AbandonedCart, 2022-11-18:
I read through that code, but how you are doing it is perfectly fine for JS, but doesn't fit into the way Android handles Bluetooth. If I try to sit on a loop of read attempts, that will end in a giant ANR crash. Not to mention, it is a bit silly when the GATT interface has a method that fires when data changes. What's odd is that it's not getting data from that step, but does continue on without it. Your method is to call each slot and keep anything not invalid. I was trying to filter out the invalid ahead of time, but it seems that won't be possible.

DanTheMan827, 2022-11-18:
I can check if there any sort of error with gatt subscriptions

it might be possible that I'm missing something in the espruino api

AbandonedCart, 2022-11-18:
I have no idea what's going on yet. It runs one command and then stops. It doesn't seem to matter how or how fast I run the command, it always stops responding after one.

DanTheMan827, 2022-11-18:
does the android gatt interface have a write with response method?

AbandonedCart, 2022-11-18:
A broken one.