HiddenRamblings/TagMo #551: [Bug]: App Crashes on Looking for Amiibo notification

DraconicNEO, 2022-08-31:
### Issue Description

After updating the app to the latest version it comes up with a prompt on the top of the screen saying "searching for Amiibo" and the app freezes every time.

If I remove storage permission the app doesn't crash, also it generated an Empty folder at "/storage/emulated/0/download/Tagmo/" so I think it might be getting stuck downloading something alrh I'm not sure.

### Logcat Output

```shell
N/A
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-08-31:
Log?

DraconicNEO, 2022-08-31:
> Logs are required to report a crash because they record the specific error and explanation.
> 
> https://github.com/HiddenRamblings/TagMo/wiki#reporting-issues

Ok, I added a logcat from the app.

DraconicNEO, 2022-08-31:
I just tested it and it didn't fix the issue, here's another logcat, I cleared the data since the last one.

```

TagMo - #2e42214
Android O_MR1 (8.1.0)
Installed by release

--------- beginning of main
08-31 01:15:57.041 12194 12205 I zygote64: Background concurrent copying GC freed 55502(2MB) AllocSpace objects, 96(8MB) LOS objects, 50% free, 23MB/46MB, paused 687us total 536.662ms
08-31 01:16:17.263 12530 12541 I zygote64: Background concurrent copying GC freed 2282(1141KB) AllocSpace objects, 2(40KB) LOS objects, 53% free, 1334KB/2MB, paused 149us total 156.777ms
08-31 01:16:17.949 12530 12530 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService()
08-31 01:16:17.955 12530 12530 D vndksupport: Loading /vendor/lib64/hw/android.hardware.renderscript@1.0-impl.so from current namespace instead of sphal namespace.
08-31 01:16:17.988 12530 12530 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService() returned 0x75efe85460
08-31 01:16:17.988 12530 12530 D RenderScript HIDL Adaptation: HIDL successfully loaded.
08-31 01:16:18.039 12530 12554 D vndksupport: Loading libllvm-qcom.so from current namespace instead of sphal namespace.
08-31 01:16:18.082 12530 12530 D         : Successfully queried cache dir: /data/user_de/0/com.hiddenramblings.tagmo.eightbit/code_cache
08-31 01:16:18.082 12530 12530 D RenderScript: Setting cache dir: /data/user_de/0/com.hiddenramblings.tagmo.eightbit/code_cache
08-31 01:16:18.159 12530 12566 D NetworkSecurityConfig: No Network Security Config specified, using platform default
08-31 01:16:18.170 12530 12566 I DpmTcmClient: RegisterTcmMonitor from: com.android.okhttp.TcmIdleTimerMonitor
08-31 01:16:18.173 12530 12566 I DpmTcmClient: RegisterTcmMonitor from: com.android.okhttp.TcmIdleTimerMonitor
08-31 01:16:18.223 12530 12575 D OpenGLRenderer: HWUI GL Pipeline
08-31 01:16:18.417 12530 12575 I Adreno  : QUALCOMM build                   : 04a4b71, Iaf0ce12d04
08-31 01:16:18.417 12530 12575 I Adreno  : Build Date                       : 08/28/19
08-31 01:16:18.417 12530 12575 I Adreno  : OpenGL ES Shader Compiler Version: EV031.22.00.01_06_07
08-31 01:16:18.417 12530 12575 I Adreno  : Local Branch                     : 
08-31 01:16:18.417 12530 12575 I Adreno  : Remote Branch                    : 
08-31 01:16:18.417 12530 12575 I Adreno  : Remote Branch                    : 
08-31 01:16:18.417 12530 12575 I Adreno  : Reconstruct Branch               : 
08-31 01:16:18.418 12530 12541 I zygote64: Background concurrent copying GC freed 9537(543KB) AllocSpace objects, 1(320KB) LOS objects, 49% free, 4MB/8MB, paused 112us total 177.930ms
08-31 01:16:18.425 12530 12575 D vndksupport: Loading /vendor/lib64/hw/gralloc.sdm660.so from current namespace instead of sphal namespace.
08-31 01:16:18.500 12530 12575 I Adreno  : PFP: 0x005ff087, ME: 0x005ff063
08-31 01:16:18.547 12530 12575 I zygote64: android::hardware::configstore::V1_0::ISurfaceFlingerConfigs::hasWideColorDisplay retrieved: 0
08-31 01:16:18.552 12530 12575 I OpenGLRenderer: Initialized EGL, version 1.4
08-31 01:16:18.552 12530 12575 D OpenGLRenderer: Swap behavior 2
08-31 01:16:18.640 12530 12541 I zygote64: Background concurrent copying GC freed 73216(2MB) AllocSpace objects, 4(424KB) LOS objects, 49% free, 10MB/20MB, paused 89us total 131.236ms
08-31 01:16:18.843 12530 12530 W View    : requestLayout() improperly called by android.widget.FrameLayout{e89bced V.E..V... ......ID 0,0-1080,1608 #2} during layout: running second layout pass
08-31 01:16:18.843 12530 12530 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{483d722 VFED.V... ......ID 0,96-1080,1605 #7f09010f app:id/foomiibo_list} during layout: running second layout pass
08-31 01:16:18.843 12530 12530 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{dfb25b3 VFED.V... ......ID 0,0-1080,12 #7f09005f app:id/amiibos_list} during layout: running second layout pass
08-31 01:16:18.900 12530 12541 I zygote64: Background concurrent copying GC freed 20328(833KB) AllocSpace objects, 2(9MB) LOS objects, 49% free, 14MB/29MB, paused 173us total 203.608ms
08-31 01:16:18.971 12530 12530 I Choreographer: Skipped 40 frames!  The application may be doing too much work on its main thread.
08-31 01:16:19.022 12530 12575 D vndksupport: Loading /vendor/lib64/hw/android.hardware.graphics.mapper@2.0-impl.so from current namespace instead of sphal namespace.
08-31 01:16:19.048 12530 12575 D vndksupport: Loading /vendor/lib64/hw/gralloc.sdm660.so from current namespace instead of sphal namespace.
08-31 01:16:20.004 12530 12541 I zygote64: Background concurrent copying GC freed 463725(16MB) AllocSpace objects, 12(7MB) LOS objects, 49% free, 6MB/12MB, paused 173us total 252.939ms
 ```

AbandonedCart, 2022-08-31:
If there are a lot of files on your phone, have you tried sliding up the bottom bar and setting a folder? It could be that it is simply trying to look through too much data.

DraconicNEO, 2022-08-31:
> If there are a lot of files on your phone, have you tried sliding up the bottom bar and setting a folder? It could be that it is simply trying to look through too much data.

I haven't tried that yet but I don't know if it would work since it just freezes instantly the moment I open it unless I block storage permission, at which case I can't set a folder.

AbandonedCart, 2022-08-31:
Swipe up the bottom bar and click "Update Document Storage" but it should have also had you pick it on the first run when denying All Files permission.

DraconicNEO, 2022-08-31:
> Swipe up the bottom bar and click "Update Document Storage" but it should have also had you pick it on the first run when denying All Files permission.

I don't have that option available, the only options I see are "Browse Physical Storage" and "Enable All Files Manager". Neither of which do anything (I'm guessing because I didn't Grant file permissions but the moment that I do the app freezes).

[image]

AbandonedCart, 2022-08-31:
According to your screenshot, you currently have file permission enabled.

DraconicNEO, 2022-08-31:
> According to your screenshot, you currently have file permission enabled.

Really? that's weird, first thing I did when opening up the app was hit the deny button.

DraconicNEO, 2022-08-31:
Here's what happens if I enable it:
[image]

And the crash screen follows after around 25-30 seconds
[image]

AbandonedCart, 2022-08-31:
That's the log you need.

PZH2000s, 2022-08-31:
I also have the same problem. The old version will not appear and will display pictures. However, the latest version will not display the picture. After the permission is granted, the response is not responded after a few seconds.

AbandonedCart, 2022-08-31:
@PZH2000s Log?

PZH2000s, 2022-08-31:
But I have tried many versions, only one version 2.7.9 can be used, and then only versions after 3.0 will crash and cannot - be clicked, with no pictures.

PZH2000s, 2022-08-31:
> ou

TagMo - #d69fde4
Android R (11)
Installed by release

--------- beginning of main
08-31 20:51:22.965 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Welcome Amiibo, doc=[AC] W27 - Claude.bin, mime=application/octet-stream
08-31 20:51:22.965 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Welcome Amiibo, doc=[AC] W18 - Jacob.bin, mime=application/octet-stream
08-31 20:51:22.965 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Welcome Amiibo, doc=[AC] W13 - Carrie.bin, mime=application/octet-stream
08-31 20:51:22.976 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Special Edition, doc=[AC] CP2 - K.K Slider.bin, mime=application/octet-stream
08-31 20:51:22.976 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Special Edition, doc=[AC] CP1 - Isabelle.bin, mime=application/octet-stream
08-31 20:51:23.029 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 123 - Fuchsia.bin, mime=application/octet-stream
08-31 20:51:23.029 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 172 - Agnes.bin, mime=application/octet-stream
08-31 20:51:23.029 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 185 - Chops.bin, mime=application/octet-stream
08-31 20:51:23.029 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 196 - Freya.bin, mime=application/octet-stream
08-31 20:51:23.029 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 113 - Isabelle.bin, mime=application/octet-stream
08-31 20:51:23.029 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 150 - Coco.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 162 - Mathilda.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 131 - Pate.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 153 - Alfonso.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 169 - Bud.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 193 - Keaton.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 102 - Reese.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 122 - Lucha.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 135 - Phoebe.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 194 - Gladys.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 130 - Prince.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 141 - Nana.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc=[AC] 188 - Ankha.bin, mime=application/octet-stream
08-31 20:51:23.030 11230 11405 W AmiiboDocument: Child parent=primary:Amiibo/Amiibo data 2021.11.08/!Animal Crossing Amiibo/Amiibo Cards_/!Series 2, doc

AbandonedCart, 2022-08-31:
You don't have any crashes in the log, though. Let's see if we can filter out that AmiiboDocument garbage

PZH2000s, 2022-09-01:
> 

But I turned off "Search subfolders" and the unresponsive problem went away. But the problem of not displaying pictures still exists.

AbandonedCart, 2022-09-01:
Sounds like you set your folder to one with a bunch of other stuff in it. Unfortunately, I have no control over how fast your device reads files. The option exists for that reason.

PZH2000s, 2022-09-01:
> 听起来您将文件夹设置为一个包含一堆其他内容的文件夹。不幸的是，我无法控制您的设备读取文件的速度。出于这个原因，该选项存在。

This is all amiibo files, nothing else. I find it very strange that older versions of the same folder can be read without crashing and displaying pictures.

PZH2000s, 2022-09-01:
> 似乎你的收藏夹将是一个普通的藏品，无法用其他类似内容的文件夹。

Now clicking on the grab log app will not do anything

AbandonedCart, 2022-09-01:
https://github.com/HiddenRamblings/TagMo/issues/552

PZH2000s, 2022-09-01:
> #552
> 
> Are you trying to run TagMo on a Ti-86? It sounds like a large part of these issues is expecting everything to load instantly, but your phone is slow.

So uninstall all Tagmo and install only one latest version of Tagmo to use it normally?
My processor is Qualcomm Snapdragon 885 plus

AbandonedCart, 2022-09-01:
I honestly have no idea. It seems like this is going in circles.

DraconicNEO, 2022-09-01:
> I honestly have no idea. It seems like this is going in circles.

It certainly does, by the way I tried the new version and it has the exact same problem.

Here's the logcat output:

```

TagMo - #205d33e
Android O_MR1 (8.1.0)
Installed by release

--------- beginning of main
09-01 10:57:15.488 27181 27181 W View    : requestLayout() improperly called by android.widget.FrameLayout{22d389b V.E..V... ......ID 0,0-1080,1608 #2} during layout: running second layout pass
09-01 10:57:15.488 27181 27181 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{7628438 VFED.V... ......ID 0,0-1080,12 #7f09005f app:id/amiibos_list} during layout: running second layout pass
09-01 10:57:15.488 27181 27181 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{48b8c11 VFED.V... ......ID 0,96-1080,1605 #7f09010f app:id/foomiibo_list} during layout: running second layout pass
09-01 10:57:15.614 27181 27181 W StaticLayout: maxLineHeight should not be -1.  maxLines:3 lineCount:3
09-01 10:57:15.617 27181 27181 W StaticLayout: maxLineHeight should not be -1.  maxLines:3 lineCount:3
09-01 10:58:41.549 27462 27462 W View    : requestLayout() improperly called by android.widget.FrameLayout{8c75aec V.E..V... ......ID 0,0-1080,1608 #2} during layout: running second layout pass
09-01 10:58:41.550 27462 27462 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{91dceb5 VFED.V... ......ID 0,96-1080,1605 #7f09010f app:id/foomiibo_list} during layout: running second layout pass
09-01 10:58:41.550 27462 27462 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{baa594a VFED.V... ......ID 0,0-1080,12 #7f09005f app:id/amiibos_list} during layout: running second layout pass
 
```

I'm no expert in reading these logcats but to me it seems like the problem occurs whenever the app starts searching for amiibo so maybe adding the ability to disable scanning for amiibo could help with this issue.

AbandonedCart, 2022-09-01:
I'm not sure what the purpose of the app would be if you remove the core function. There is also nothing in the log to say that scanning for files is actually causing the issue. It points to loading the Flask interface, but also something that should be handled.

DraconicNEO, 2022-09-01:
>I'm not sure what the purpose of the app would be if you remove the core function.

Not remove it entirely just have it only start scanning after the option is turned on, maybe even from a prompt upon starting the app. Idk it might not help or do anything at all for the issue.

> There is also nothing in the log to say that scanning for files is actually causing the issue. It points to loading the Flask interface, but also something that should be handled.

Is that something storage related? I just know that it stops working when it accesses storage. Blocking the storage permission from Android settings makes it not crash anymore.

I don't really even have files on this device at all , just the few screenshots I took, the tagmo apks , and the keys for amiibo.

PZH2000s, 2022-09-01:
> > 如果您删除核心功能，我不确定应用程序的目的是什么。
> 
> 不完全删除它只是让它在选项打开后才开始扫描，甚至可能在启动应用程序时出现提示。Idk 它可能对这个问题没有帮助或做任何事情。
> 
> > 日志中也没有任何内容表明扫描文件实际上是导致问题的原因。它指向加载 Flask 接口，也是应该处理的东西。
> 
> 这和存储有关吗？我只知道它在访问存储时停止工作。从 Android 设置中阻止存储权限使其不再崩溃。
> 
> 我什至在这个设备上根本没有文件，只有我拍的几张截图、tagmo apks 和 amiibo 的密钥。

Can your app load images?
There is a "Search subfolders" to turn it off and it won't crash.

AbandonedCart, 2022-09-01:
> Can your app load images? There is a "Search subfolders" to turn it off and it won't crash.

Your issue appears to be with the API, not the files. They are two different issues.

DraconicNEO, 2022-09-01:
> > > 如果您删除核心功能，我不确定应用程序的目的是什么。
> > 
> > 
> > 不完全删除它只是让它在选项打开后才开始扫描，甚至可能在启动应用程序时出现提示。Idk 它可能对这个问题没有帮助或做任何事情。
> > > 日志中也没有任何内容表明扫描文件实际上是导致问题的原因。它指向加载 Flask 接口，也是应该处理的东西。
> > 
> > 
> > 这和存储有关吗？我只知道它在访问存储时停止工作。从 Android 设置中阻止存储权限使其不再崩溃。
> > 我什至在这个设备上根本没有文件，只有我拍的几张截图、tagmo apks 和 amiibo 的密钥。
> 
> Can your app load images?

It can.

>There is a "Search subfolders" to turn it off and it won't crash.

I can't find any such option in the app.

AbandonedCart, 2022-09-01:
What doesn't make sense is that I run TagMo on three live devices and two emulators. None of them seem to have either of these issues. That is combined with the 10,000 active users who installed the Google Play version because that one logs ANR (Application Not Responding) and Exception (crashes) automatically. I would think something related to storage would have made a much bigger "splash" but the only other issue has been that anyone who happened to get the version with the heroku server couldn't load the API data without updating to a newer build.

AbandonedCart, 2022-09-01:
I just rewrote the entire way files are located to look for bins first before starting to traverse directories and to skip any step it can possibly determine won't be necessary. It's pretty much as fast as it can possibly run on that side. On phones released in the last 2 years, the entire load time should not exceed 2 seconds.

AbandonedCart, 2022-09-01:
Fixed by https://github.com/HiddenRamblings/TagMo/commit/1829debb4bac81c491c25da2f6476d6ea1a71137

closed 2022-09-01 by AbandonedCart