HiddenRamblings/TagMo #556: [Bug]: "Tag already an amiibo™" Prevents tags from repairing mismatched dynamic locks.

MegaBaconSticks, 2022-09-02:
### Issue Description

In the past, a user would be able to repair an NFC Tag if the Nintendo Console could not recognize the NTAG215 Tag as an amiibo. **This problem would be solved by simply rewriting the tag. This is how I have fixed my cards in the past.**

However, there have been updates that prevent tags from getting a rewrite over the Exception "Tag already an amiibo™". This prevents users from repairing problems such as Dynamic Lock Mismatches.

Please change this or remove the Exception so Tags can get repairs.

### Logcat Output

```shell
TagMo - #7af707e
Android TIRAMISU (13)
5.34 GB RAM
Installed by release

--------- beginning of main
09-01 22:39:07.184   939   939 W OnBackInvokedCallback: OnBackInvokedCallback is not enabled for the application.
09-01 22:39:07.184   939   939 W OnBackInvokedCallback: Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
09-01 22:39:13.668   939   989 W Parcel  : Expecting binder but got null!
09-01 22:39:13.829   939   939 W OnBackInvokedCallback: OnBackInvokedCallback is not enabled for the application.
09-01 22:39:13.829   939   939 W OnBackInvokedCallback: Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
09-01 22:39:15.260   939  5833 W IOException: java.io.IOException: java.io.IOException: Tag already an amiibo™
09-01 22:39:15.260   939  5833 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:114)
09-01 22:39:15.260   939  5833 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:379)
09-01 22:39:15.260   939  5833 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
09-01 22:39:15.260   939  5833 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
09-01 22:39:15.260   939  5833 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source:4)
09-01 22:39:15.260   939  5833 W IOException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
09-01 22:39:15.260   939  5833 W IOException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
09-01 22:39:15.260   939  5833 W IOException: 	at java.lang.Thread.run(Thread.java:1012)
09-01 22:39:15.260   939  5833 W IOException: Caused by: java.io.IOException: Tag already an amiibo™
09-01 22:39:15.260   939  5833 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.validateBlankTag(TagReader.java:26)
09-01 22:39:15.260   939  5833 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:112)
09-01 22:39:15.260   939  5833 W IOException: 	... 7 more
09-01 22:39:19.321   939  5838 W IOException: java.io.IOException: java.io.IOException: Tag already an amiibo™
09-01 22:39:19.321   939  5838 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:114)
09-01 22:39:19.321   939  5838 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:379)
09-01 22:39:19.321   939  5838 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
09-01 22:39:19.321   939  5838 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
09-01 22:39:19.321   939  5838 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda13.run(Unknown Source:4)
09-01 22:39:19.321   939  5838 W IOException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
09-01 22:39:19.321   939  5838 W IOException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
09-01 22:39:19.321   939  5838 W IOException: 	at java.lang.Thread.run(Thread.java:1012)
09-01 22:39:19.321   939  5838 W IOException: Caused by: java.io.IOException: Tag already an amiibo™
09-01 22:39:19.321   939  5838 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.validateBlankTag(TagReader.java:26)
09-01 22:39:19.321   939  5838 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:112)
09-01 22:39:19.321   939  5838 W IOException: 	... 7 more
09-01 22:39:55.590   939   939 W OnBackInvokedCallback: OnBackInvokedCallback is not enabled for the application.
09-01 22:39:55.590   939   939 W OnBackInvokedCallback: Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
09-01 22:39:57.883   939   989 W Parcel  : Expecting binder but got null!
09-01 22:39:58.041   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:00.564   939   939 E ImeBackDispatcher: Ime callback not found. Ignoring unregisterReceivedCallback. callbackId: 230162211
09-01 22:40:07.500   939   939 W BottomSheetBehavior: Cannot set state: 5
09-01 22:40:07.637   939   939 W RecyclerView: No adapter attached; skipping layout
09-01 22:40:08.094   939  5952 W cr_media: Requires BLUETOOTH permission
09-01 22:40:16.354   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:27.347   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:30.893   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:33.196   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:35.047   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:36.165   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:37.361   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:39.717   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:41.092   939   989 W Parcel  : Expecting binder but got null!
09-01 22:40:42.590   939   989 W Parcel  : Expecting binder but got null!
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-09-02:
The sections of a tag that cannot be modified *have not changed*. The error messages have been clarified and more prevention for writing faulty tags have been added. You are still required to use "Update" to modify an existing tag and "Write" will still throw an error that the tag was already written.

MegaBaconSticks, 2022-09-02:
My bad then.

I just wonder how do I prevent my cards from ending up faulty and unusable.

AbandonedCart, 2022-09-02:
I'm not entirely sure. It could be the tags. It could also be something as random as the NFC radio being a bit unstable and not performing a clean write. There isn't really an effective test that the app can perform beyond making sure the data fits the profile of a valid tag.

I did double check and compared the current validation to the one from 2020 (2.7.0) just to be sure.

https://github.com/HiddenRamblings/TagMo/blob/9a79211786ff594ffbed1dbe3460d3ced12e9c77/app/src/main/java/com/hiddenramblings/tagmo/nfctech/TagReader.java#L21-L28

https://github.com/HiddenRamblings/TagMo/blob/c45c9fbcb284cef8189e4af5a89bbc139d47162d/app/src/main/java/com/hiddenramblings/tagmo/TagWriter.java#L47-L55

MegaBaconSticks, 2022-09-02:
I ended up having to use a new Tag and write the same amiibo and it worked. Honestly it could've been Tags or the Unstable NFC radio signal. It feels like my Pixel 4 gets worse for each passing day, lol.

MegaBaconSticks, 2022-09-02:
Couldn't have said it any better myself. It doesn't help that phones only have NFC support for the soul sake and purpose of Wireless Pay or Zero Touch Payment.

Apologies if I have wasted any time and caused any misunderstandings with the initial post. Should I mark this as complete?

AbandonedCart, 2022-09-02:
No worries. It never hurts to make sure the updates are actually improvements.

Thanks. I will take another run through the verification methods just to be sure there isn't something else that might improve things, but you'll see any of that in an app update.

closed 2022-09-02 by AbandonedCart