HiddenRamblings/TagMo #557: [Bug]: Tagmo doesn't read tags with Samsung Note 10

kinibee, 2022-09-05:
### Issue Description

When I try to write to tags it can't seems to find the tags at all. my NFC works fine at cash registers but doesn't seem to work with the app. The bin files and keys are all installed properly and the NTAG15 tags are confirmed to work on a older version of Tagmo on an older phone I had.

### Logcat Output

```shell
TagMo - #5f22ed52
Android S (12)
7.28 GB RAM
Installed by publish

--------- beginning of main
09-05 15:07:04.519 31457 31457 W View    : requestLayout() improperly called by android.widget.FrameLayout{7bd9a6b V.E..V... ......ID 0,0-1080,1806 #7} during layout: running second layout pass
09-05 15:07:04.519 31457 31457 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{aada40c VFED.V... ......ID 0,96-1080,1803 #7f09010f app:id/foomiibo_list} during layout: running second layout pass
09-05 15:07:04.519 31457 31457 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{f991812 VFED.V... ......ID 0,0-1080,12 #7f09005f app:id/amiibos_list} during layout: running second layout pass
09-05 15:07:04.648 31457 31473 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai
09-05 15:07:21.534 31457 31474 W System  : A resource failed to call end. 
09-05 15:07:41.095 31457 31457 W BottomSheetBehavior: Cannot set state: 5
09-05 15:07:41.251 31457 31457 W RecyclerView: No adapter attached; skipping layout
09-05 15:16:34.232 31457 31457 W View    : requestLayout() improperly called by androidx.appcompat.widget.AppCompatTextView{5c1ec0 V.ED..... ......ID 216,44-393,124} during layout: running second layout pass
09-05 15:16:34.232 31457 31457 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{aada40c VFED.V... ......ID 0,96-1080,1803 #7f09010f app:id/foomiibo_list} during layout: running second layout pass
09-05 15:16:34.232 31457 31457 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{f991812 VFED.V... ......ID 0,0-1080,12 #7f09005f app:id/amiibos_list} during layout: running second layout pass
09-05 15:16:34.327 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{c141d10 V.E..V... ........ 0,0-1044,228} during second layout pass: posting in next frame
09-05 15:16:34.327 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{14fd11a V.E..V... ........ 0,71-792,112 #7f09011d app:id/group1} during second layout pass: posting in next frame
09-05 15:16:34.327 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{809e4e6 V.E..V... ........ 0,112-792,153 #7f09011e app:id/group2} during second layout pass: posting in next frame
09-05 15:16:34.327 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{b35b1b3 V.E..V... ........ 0,0-1044,228} during second layout pass: posting in next frame
09-05 15:16:34.327 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{790c8a5 V.E..V... ........ 0,71-792,112 #7f09011d app:id/group1} during second layout pass: posting in next frame
09-05 15:16:34.327 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{46a4a21 V.E..V... ........ 0,112-792,153 #7f09011e app:id/group2} during second layout pass: posting in next frame
09-05 15:16:34.328 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{770ca82 V.E..V... ........ 0,0-1044,228} during second layout pass: posting in next frame
09-05 15:16:34.328 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{15e42fc V.E..V... ........ 0,71-792,112 #7f09011d app:id/group1} during second layout pass: posting in next frame
09-05 15:16:34.328 31457 31457 W View    : requestLayout() improperly called by android.widget.LinearLayout{aca92e8 V.E..V... ........ 0,112-792,153 #7f09011e app:id/group2} during second layout pass: posting in next frame
```

### Limitations and Requirements

- [X] I have read and understood the limitations and requirements

AbandonedCart, 2022-09-05:
And you're sure it's not a case or simply touching the tags to the wrong spot on the phone? It's not a failed read, so it's also not something I can really diagnose through the app.

AbandonedCart, 2022-09-06:
Just a heads up - Issues that don't have a clear cause related to TagMo are typically closed if they are abandoned. Abandoned issues are those where more information is required and no response is received after 24 - 48 hours, depending on the severity of the issue.

kinibee, 2022-09-06:
I'm definitely holding it over the NFC part of the phone so I'm not sure what is wrong. Possibly the rest of the NFC card pack are duds I guess.

AbandonedCart, 2022-09-06:
Have you tried another app, such as NFC Tools, to see if the tags can be identified by the phone?

AbandonedCart, 2022-09-06:
Typically it is meant for much older phones, but disabling the NTAG215 validation in settings may work. I don't have a lot of confidence it will make a difference, but I also don't want to completely rule out the possibility it is something in TagMo without knowing for sure what exactly is causing it.

AbandonedCart, 2022-09-07:
If it ends up not being the tags and disabling the validation option doesn't help, drop me a line and I will look for other possible solutions.

closed 2022-09-07 by AbandonedCart