HiddenRamblings/TagMo #559: Support for NFC writing through Joy-Con Bluetooth

TheMemeSniper, 2022-09-08:
Allow NFC tags to be written through the right Joy-Con with Bluetooth. This would allow users without NFC support to use TagMo or for users on Switchroot Android, they can write NFC tags through the Switch.

AbandonedCart, 2022-09-08:
I'm going to take that as a request, even though it reads a lot like a demand. The concept is already being tested, but may not be possible because of use awkward connection process.

AbandonedCart, 2022-09-21:
After some heavy research, it appears it has been done to some extent. The problem being that Android sort of lagged behind on the necessary drivers and such to support it. Connecting an HID (Human Interface Device) through Bluetooth is common on Windows, Mac, etc. On Android, they more or less assumed you would only ever need to do it at the system level. The catch is that to connect a Joy-Con as an actual Joy-Con and not a generic controller, there is a whole process of handshakes and verification. Android connects by simply saying "oh, a controller" and connecting it like any other.

AbandonedCart, 2022-09-21:
TL;DR it is going to be very hacky on most devices and support will be more or less depending on if and when the phone manufacturer added the drivers.

TheMemeSniper, 2022-09-25:
alright thank you

AbandonedCart, 2022-11-12:
Haven't successfully connected to a controller for anything more than standard input yet. I'm not sure what is missing, but something is definitely not happening that should.

german77, 2023-06-16:
I invested quite a bit of time trying to accomplish the same thing. If the issue is configuring the controller I know how to activate the joycon NFC and being able to read or write data from tags.

AbandonedCart, 2023-06-16:
This has been on the back burner for a while, mostly because the only real use for it would be the 1 in 10,000 devices that don't natively support NFC. Allegedly, it connects. I'm just not sure if the issue is that it isn't actually connecting or it's the mismatched Bluetooth implementation in the library.

AbandonedCart, 2023-06-16:
That makes sense. For NFC alone, there's only really a handful of devices that don't already have it and most of them are already struggling to run the app at all. In the last 3 years, there have only been 3 requests for it and something else has always come up.

AbandonedCart, 2023-06-19:
With the amount of overhead required for this feature and the fact that it would primarily be used by devices with RAM in the megabytes, it just doesn't make sense to implement it.

closed 2023-06-19 by AbandonedCart