HiddenRamblings/TagMo #561: [Request] custom amiiboapi url config

ca1e, 2022-09-09:
since #552 , can we use a custom config for amiiboapi url? it can also use the amiiboapi.com as default when user do nothing

AbandonedCart, 2022-09-09:
It's not a configuration option. TagMo has it's own mirror of the API that falls back to the default.

AbandonedCart, 2022-09-09:
I linked the custom API to the Foomiibo (which is the reason it exists in the first place). Disabling Foomiibo disables the custom API. You need to refresh in settings to see the changes, though.

AbandonedCart, 2022-09-11:
I'm not entirely sure if you were asking to use your own supplied address or if that is what happens because I read the post before seeing the title. 

To better explain, the official database only allows for officially documented (released) records. To allow for pre-release and spoof files, we have our own mirror. The heroku mess was because literally the same week that the mirror was created, they announced that free plans would be discontinued. The host moved to render, but the code was also added to always fall back to the default when the mirror doesn't respond.

Since then, even the images fall back alongside the host (albeit they are both on GitHub and it has no real purpose) to avoid any issues where an image is called and doesn't exist.

As I mentioned above, disabling the Foomiibo bar will disable the mirror, but the catch is that any unofficial files will then appear "broken" in the browser.

ca1e, 2022-09-11:
yes I was asking to use my own supplied address, cause for now the render mirror's database is old and amiiboapi.com still woks

AbandonedCart, 2022-09-11:
None of them refresh automatically and I have a feeling that adding a third custom value may lead to a lot of false reports about issues related to someone entering an invalid address or other mirrors getting too far out of sync. The current mirror is synced with the main at least daily.

closed 2022-09-11 by AbandonedCart

AbandonedCart, 2022-09-11:
I don't believe that adding a third option of a custom URL is actually necessary. It sounds like the issue is that you booted up after AmiiboAPI added Steve and Alex and they didn't load, at least that is the assumption based on timing.

The mirror is updated daily (or more), but sometimes it doesn't catch every change instantly. The entire API doesn't get thrown out when an update is a little late for a new bin file. The reason no API updates were coming through was that the URL for the time dropped the "/api/" during the addition of the fallback. This applied to both addresses, so the issue would have been in the default URL and likely any third option.

I actually found the issue by running a sync in the app and then using the "Report a Problem" feature to print the log. That is the best option when an issue is encountered. In all fairness, mirroring the API is all automated. The only thing I modify are JSON values and images. It's pretty hard to mess that up and render emails me instantly if there is an issue.