HiddenRamblings/TagMo #562: F-Droid support

Soundtoxin, 2022-09-11:
Are there any plans to add the app to F-Droid? Either through the main repo or your own repo. I can get 99% of my apps from F-Droid, but this one isn't on there, and TagMo seems to be GPL 3.0 so that part shouldn't be an issue either. I can get the app off Aurora Store for now, and you guys post APKs to release here as well which is nice, but F-Droid would be my preference if it were an option.

AbandonedCart, 2022-09-11:
There are no plans to host on any other stores. Honestly, the addition of Google Play was more for show than anything. A lot of false speculation was being spread that it was blocked, banned, or rejected. Actually, downloading through Aurora Store is probably the worst option because of the real reasons.

If they source the binary from Google Play, you get all of the downgrades with no real purpose. Google Play requires prominent disclosures for obvious tasks, such as enabling location permission to enable bluetooth and then not using location. Simply by the permission being required, you get stuck being notified that the app is requesting permission that isn't being used, but also (by Google's requirements) that it would possibly collect your location data if it were being used. Google Play also requires that only file managers use file permissions, leaving the files to be handled by their slow and broken file management protocol (that is still slow after the performance workarounds).

Honestly, though, the app is self updating. One of the biggest things that is always checked, tested, and considered is that the app can always update itself. Even during the package name change, the app installed the new package and cleaned up the old one so there was no break in compatibility for the updates. Even those using the (since removed) experimental branch can still update.

That said, the only real time any storefront is required would be for the initial install or because Google Play only allows updating through Google Play. Even that, though, can be replaced with a GitHub version seamlessly.

Soundtoxin, 2022-09-11:
Thank you for the thorough explanation. I was unaware of the auto-updater. You might still be able to host your own f-droid repo without meeting the same criteria for the main one, but I won't push on that.

closed 2022-09-11 by Soundtoxin