HiddenRamblings/TagMo #567: Possible Huawei Android 10 incompatibility

AbandonedCart, 2022-09-16:
Seem to be getting a lot of beta feedback from Google about issues from Huawei users on Android 10, but none have actually submitted logs and the reviews are somewhat vague. The general idea is that apparently it freezes. For now, I am posting this issue in case someone with a Huawei on Android 10 can provide more information.

ZIK999, 2022-09-19:
I'm on Asus ROG5 (Android 12), app freezes every time after 5-10 seconds after start. Can't use it at all.
Sent logs through google form.

AbandonedCart, 2022-09-19:
Doesn't appear to be a hardware issue and there is no evidence to suggest it is ongoing, so this is being closed in favor of individual reports.

closed 2022-09-19 by AbandonedCart