HiddenRamblings/TagMo #569: [Bug]: White screen

Buzbee, 2022-09-17:
### Issue Description

After installing versions 0f0182d or 9497fd0 I get a white screen after startup on Android 4.1.2. 7b60375 works fine still.

### Logcat Output

```shell
no logcat
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2022-09-17:
Let's start with what device and Android version. Also make sure there isn't a log being saved to the clipboard using the newest one. A year of changes is a lot to cover.

AbandonedCart, 2022-09-17:
0485312937beb2eb6fd06b40c7f2c055dd94e8ae should fix the issue, but I would prefer confirmation before closing it automatically.

Buzbee, 2022-09-17:
[logcat_09-17-2022_10-46-06.txt](https://github.com/HiddenRamblings/TagMo/files/9591955/logcat_09-17-2022_10-46-06.txt)

AbandonedCart, 2022-09-17:
Is that from the one that was just posted? Still not entirely sure what Android version we are talking about.

Buzbee, 2022-09-17:
it was from android 4.1.2 and 9497fd0

Buzbee, 2022-09-17:
still has same issue with  ed5c052 .

[logcat_09-17-2022_12-06-19.txt](https://github.com/HiddenRamblings/TagMo/files/9592106/logcat_09-17-2022_12-06-19.txt)

AbandonedCart, 2022-09-17:
Unfortunately, it appears to be struggling with the existence of Bluetooth.

Buzbee, 2022-09-17:
still failing with latest

[logcat_09-17-2022_13-05-50.txt](https://github.com/HiddenRamblings/TagMo/files/9592225/logcat_09-17-2022_13-05-50.txt)

AbandonedCart, 2022-09-18:
I ran it through an emulator and found that apparently Glide and BlurView fight for control of RenderScript when using compatibility mode. For now, BlurView is disabled. It may not be possible to restore it, but it is not a huge loss. It was a visual effect to make certain windows stand out a bit more.

Buzbee, 2022-09-18:
This fixes the startup up issues but Blurview seems to be used for menu transitions and refreshes. 

1. When I press on guide it freezes and locks up during a menu transition.
2. When I uncheck Hide Foomiiibo panel in settings (or go to n2elite panel and back), the browser never refreshes to show my amiibos again. I have to exit and start program again.
3. Searching for amiibo is always at the top of the screen.

Unrelated but Load amiibo database file is not working in this version. I have latest file from amiiboapi and this file loaded successfully in 2.87. Faiks to parse in this version.

Buzbee, 2022-09-18:
Latest fix still has same issues.

AbandonedCart, 2022-09-18:
You may want to consider updating to a new device. The whole premise to adding legacy support into TagMo was "until it becomes impossible to maintain" and it sounds like we are reaching that point. As much as remaining as compatible as possible was the goal of finding a way to support it, it's also a platform nearly 10 years old being supported for the sake of people picking up $20 devices on eBay. Supporting the Flask (which appears to be part of the problem for pre-Lollipop devices) is much higher priority.

Buzbee, 2022-09-18:
I do not mind sticking with an older version. Are there archived versions newer than 2.87 without blurview?  The last 3 super smash bros and last 3 monster hunter rise amiibos are showing up as unknown with the latest json in 2.87 (and they are in the json). So I wanted to try a newer version to see if it would fix this. I do not use any of the new features like flask in the newer versions.

AbandonedCart, 2022-09-18:
The blur view is disabled on that Android version. The issue with 4.1 has and always will be Google Play services  won't allow unpatched access. The SSL had a huge security hole. There is some compatibility stuff that can be done, but it's finicky. The BlurView was causing the white screen, but that's it. The rest is SSL. Anything that's https (AmiiboAPI, GitHub, etc) will reject the system SSL because it's dangerous.

AbandonedCart, 2022-09-18:
The log excerpts you posted were a bit of a false positive because they only showed the error for the Bluetooth failing, but I was able to verify that is actually being handled as expected in the emulator. It never crashes, just complains it can't run and moves on.

Buzbee, 2022-09-18:
9d182eac allows tab refreshing between n2 elite and flask and browser. 1 2 & 3 above otherwise still exist. sometimes pressing on guide it would minimize app instead of crashing it. now it freezes app while saying https website not responding.

Buzbee, 2022-09-18:
223f1d4 still freezes or minimizes app when going to guides. When I choose flask I get the option displayed below of choosing browser (only chrome option works for displaying "connect flask". I do not get these options for guides.

These 2 problems still exist also.
1. When I uncheck Hide Foomiiibo panel in settings the browser never refreshes to show my amiibos again. I have to exit and start program again.
2. Searching for amiibo is always at the top of the screen.
[image]
[image]

AbandonedCart, 2022-09-18:
They haven't been addressed yet. The changes being posted are cleanup for the conditions.

AbandonedCart, 2022-09-19:
You aren't going to get browser options for Guides. That is an integrated WebView. Flask, on the other hand, can support Web Bluetooth through Chrome. It is a fallback for when conventional Bluetooth isn't accessible. On your device, it is likely neither will truly work.

The only thing that can really be addressed would be the searching bar not being dismissed. Anything to do with web is a result of [CVE-2014-022](4http://web.nvd.nist.gov/view/vuln/detail?vulnId=CVE-2014-0224)

AbandonedCart, 2022-09-19:
The file issues should be worked out. Storage permissions didn't exist until Android M, so asking for them was causing the system to think you were using SAF (a MUCH later version of file management)

Buzbee, 2022-09-20:
8a8b34c freezes app now after searching for amiibo for amiibo Sometimes it freezes before the searching for amiibo screen goes away and sometimes after. The browser never populates with any amiibo.

AbandonedCart, 2022-09-20:
That message means the Google Play Services patch is being rejected or isn't available. That's not something that can be fixed with a workaround. That is the workaround. Without it, Android itself blocks access to the internet.