HiddenRamblings/TagMo #574: App will not start correctly

JackTheCuddlyLunatic, 2022-09-27:
Whenever I attempt to launch the application, it will stay on either a gray or black screen, and will not do anything, as if it had crashed. Whenever I try to back out of the application, there is significant delay before it sends me to the homescreen of my device. I have tried all three versions, the two on GitHub, and the one on the play store, including uninstalling and reinstalling several times, but nothing has worked. If it helps, I'm using a phone with Android 12.

AbandonedCart, 2022-09-27:
There are a handful of reports about something similar. All of them appear to be Android 12 and all say roughly the same thing. Unfortunately, that is all the info we have about it.

AbandonedCart, 2022-09-27:
Assuming the library works as advertised, the freeze should now crash the app and (in doing so) copy a log to the clipboard that will give some idea why it is happening. So, the process for testing the upcoming build is:

Install
Launch
Wait for crash
Paste log

AbandonedCart, 2022-09-27:
Yes, the previous version did not crash on a lock. That is why the new version was necessary.

AbandonedCart, 2022-09-28:
I posted the build with the debugging in a zip file. The newest build will also work, but in case anything changes.

AbandonedCart, 2022-09-28:
Does the "Report Error" option in the top menu not work when it's first opened? Logs are not discarded between runs.

AbandonedCart, 2022-09-29:
I wonder if it's trying to add settings when settings exists and hits just the right timing to overlap.

AbandonedCart, 2022-09-29:
Everything showing in the logs should be addressed as of https://github.com/HiddenRamblings/TagMo/commit/d46f0d6753c72012c0f0f53866d8d2204f36358c, but I am not sure if this also resolves the original report by @JackTheCuddlyLunatic. Settings does load regardless of opening it, so it is entirely possible. It also wasn't expected for settings itself to cause an ANR, since it's more or less just an overlay.

AbandonedCart, 2022-09-30:
Haven't had any new reports of the lockup from Android 12 since the update went through on play, so I'll assume this was an orphaned report. If the OP still has the issue, ill open it back up.

closed 2022-09-30 by AbandonedCart