HiddenRamblings/TagMo #576: [Bug]: Failed to decrypt tag data

AbandonedCart, 2022-09-29:
### Issue Description

Failed to decrypt tag data

### Logcat Output

```shell
09-28 21:43:30.295 16571 17902 W Exception: java.lang.Exception: Failed to decrypt tag data
09-28 21:43:30.295 16571 17902 W Exception: 	at com.hiddenramblings.tagmo.amiibo.KeyManager.decrypt(KeyManager.java:130)
09-28 21:43:30.295 16571 17902 W Exception: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:93)
09-28 21:43:30.295 16571 17902 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:379)
09-28 21:43:30.295 16571 17902 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
09-28 21:43:30.295 16571 17902 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
09-28 21:43:30.295 16571 17902 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda16.run(Unknown Source:4)
09-28 21:43:30.295 16571 17902 W Exception: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
09-28 21:43:30.295 16571 17902 W Exception: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
09-28 21:43:30.295 16571 17902 W Exception: 	at java.lang.Thread.run(Thread.java:1012)
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2022-09-29:
Fixed by 1bb41af7250d33bf189c3e22cd1d349c8e2a76b1

closed 2022-09-30 by AbandonedCart

AbandonedCart, 2022-10-03:
[image]

Google Play review is still pending, but there is no way to actually notify anyone before they send feedback.

AbandonedCart, 2022-10-03:
The update finally hit some time early this morning.

DarkSkull91, 2022-10-08:
I continue to get this error. What I need to do?

AbandonedCart, 2022-10-08:
Update

DarkSkull91, 2022-10-08:
is the version on the Google Play store good?

AbandonedCart, 2022-10-08:
3.2.0 or newer was after the issue was fixed.

DarkSkull91, 2022-10-08:
I have installe the 3.2.2 version

AbandonedCart, 2022-10-08:
That is the newest Google Play version, but keep in mind the Play builds are notoriously slower.

DarkSkull91, 2022-10-08:
I have installed the 3.2.2 version but the problem is also there and the version on this repo can't install on my phone.

AbandonedCart, 2022-10-08:
https://github.com/HiddenRamblings/TagMo/wiki#reporting-issues