HiddenRamblings/TagMo #578: Logcat - TagMo #420566f4

LogcatGitHub, 2022-09-29:
*Submitted by hawkingscrest@*****.com*

Description:
---

It won't let me write amiibos now after this last update 

Device info:
---
<table>
<tr><td>App version</td><td>3.1.6</td></tr>
<tr><td>App version code</td><td>316</td></tr>
<tr><td>Android build version</td><td>G998USQS5CVHB</td></tr>
<tr><td>Android release version</td><td>12</td></tr>
<tr><td>Android SDK version</td><td>31</td></tr>
<tr><td>Android build ID</td><td>SP1A.210812.016.G998USQS5CVHB</td></tr>
<tr><td>Device brand</td><td>samsung</td></tr>
<tr><td>Device manufacturer</td><td>samsung</td></tr>
<tr><td>Device name</td><td>p3q</td></tr>
<tr><td>Device model</td><td>SM-G998U</td></tr>
<tr><td>Device product name</td><td>p3qsqw</td></tr>
<tr><td>Device hardware name</td><td>qcom</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```
TagMo - #420566f4
Android S (12)
10.07 GB RAM
Installed by publish

--------- beginning of crash
09-29 07:53:28.729 13681 13883 E AndroidRuntime: FATAL EXCEPTION: pool-8-thread-1
09-29 07:53:28.729 13681 13883 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 13681
09-29 07:53:28.729 13681 13883 E AndroidRuntime: java.lang.SecurityException: Permission Denial: opening provider com.android.externalstorage.ExternalStorageProvider from ProcessRecord{7fa69f0 13681:com.hiddenramblings.tagmo.eightbit/u0a1035} (pid=13681, uid=11035) requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2438)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.os.Parcel.createException(Parcel.java:2422)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.os.Parcel.readException(Parcel.java:2405)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.os.Parcel.readException(Parcel.java:2347)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.app.IActivityManager$Stub$Proxy.getContentProvider(IActivityManager.java:6848)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.app.ActivityThread.acquireProvider(ActivityThread.java:7849)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.app.ContextImpl$ApplicationContentResolver.acquireUnstableProvider(ContextImpl.java:3527)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.content.ContentResolver.acquireUnstableProvider(ContentResolver.java:2533)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.content.ContentResolver.query(ContentResolver.java:1210)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.content.ContentResolver.query(ContentResolver.java:1158)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at android.content.ContentResolver.query(ContentResolver.java:1114)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.hiddenramblings.tagmo.amiibo.AmiiboDocument.listFiles(AmiiboDocument.java:81)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.hiddenramblings.tagmo.amiibo.AmiiboDocument.listFiles(AmiiboDocument.java:66)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiiboDocuments(AmiiboManager.java:346)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.lambda$loadAmiiboDocuments$44(BrowserActivity.java:1833)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.$r8$lambda$sDbEPDnt6h8qsTHQiZ64Zij5itY(Unknown Source:0)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity$$ExternalSyntheticLambda74.run(Unknown Source:6)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:1012)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: Caused by: android.os.RemoteException: Remote stack trace:
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.checkAssociationAndPermissionLocked(ContentProviderHelper.java:831)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.getContentProviderImpl(ContentProviderHelper.java:459)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.getContentProviderImpl(ContentProviderHelper.java:200)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.getContentProvider(ContentProviderHelper.java:174)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 	at com.android.server.am.ActivityManagerService.getContentProvider(ActivityManagerService.java:7638)
09-29 07:53:28.729 13681 13883 E AndroidRuntime: 
09-29 07:53:49.504 13681 14466 E AndroidRuntime: FATAL EXCEPTION: pool-12-thread-1
09-29 07:53:49.504 13681 14466 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 13681
09-29 07:53:49.504 13681 14466 E AndroidRuntime: java.lang.SecurityException: Permission Denial: opening provider com.android.externalstorage.ExternalStorageProvider from ProcessRecord{7fa69f0 13681:com.hiddenramblings.tagmo.eightbit/u0a1035} (pid=13681, uid=11035) requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
09-29 07:53:49.504 13681 14466 E AndroidRuntime: 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2438)
[... 277 lines cut ...]
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.os.Parcel.readException(Parcel.java:2347)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.app.IActivityManager$Stub$Proxy.getContentProvider(IActivityManager.java:6848)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.app.ActivityThread.acquireProvider(ActivityThread.java:7849)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.app.ContextImpl$ApplicationContentResolver.acquireUnstableProvider(ContextImpl.java:3527)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.content.ContentResolver.acquireUnstableProvider(ContentResolver.java:2533)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.content.ContentResolver.query(ContentResolver.java:1210)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.content.ContentResolver.query(ContentResolver.java:1158)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at android.content.ContentResolver.query(ContentResolver.java:1114)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.hiddenramblings.tagmo.amiibo.AmiiboDocument.listFiles(AmiiboDocument.java:81)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.hiddenramblings.tagmo.amiibo.AmiiboDocument.listFiles(AmiiboDocument.java:66)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiiboDocuments(AmiiboManager.java:346)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.lambda$loadAmiiboDocuments$44(BrowserActivity.java:1833)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.$r8$lambda$sDbEPDnt6h8qsTHQiZ64Zij5itY(Unknown Source:0)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.hiddenramblings.tagmo.browser.BrowserActivity$$ExternalSyntheticLambda74.run(Unknown Source:6)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:1012)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: Caused by: android.os.RemoteException: Remote stack trace:
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.checkAssociationAndPermissionLocked(ContentProviderHelper.java:831)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.getContentProviderImpl(ContentProviderHelper.java:459)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.getContentProviderImpl(ContentProviderHelper.java:200)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.android.server.am.ContentProviderHelper.getContentProvider(ContentProviderHelper.java:174)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 	at com.android.server.am.ActivityManagerService.getContentProvider(ActivityManagerService.java:7638)
09-29 16:20:05.155 27069 27323 E AndroidRuntime: 
```

AbandonedCart, 2022-09-29:
It looks like all permissions for files were denied. Did you select a folder?

AbandonedCart, 2022-09-30:
Fixed by https://github.com/HiddenRamblings/TagMo/commit/903fd25a9775a807ef099e492d73f25f0d052894, duplicate of https://github.com/HiddenRamblings/TagMo/issues/576

closed 2022-09-30 by AbandonedCart