HiddenRamblings/TagMo #579: [Feature Request] Add support for Pokémon Rumble U (Topaz 512) writing

M1CR0H4CK3R, 2022-09-29:
I am aware that Pokémon Rumble U figures are vastly different to Amiibo, but that should be part of the challenge. Plus, according to sources online, there are no passwords used in them, although the bin files are still encrypted. The challenge would be figuring out the encryption. If this feature is already planned, please let me know. Thanks.

AbandonedCart, 2022-09-29:
Figuring out the encryption for a completely different format with pending requests for Skylanders, Lego Dimensions, and reading from the Joy-Con means this is not something that will be considered any time soon.

AbandonedCart, 2022-09-29:
It also appears they are using the same format as payment cards, which creates a bit of a problem with filtering them out. The current format is NTAG215 and then the other requests are for NTAG213, but all of them fall under mifare ultralight. The Topaz are mifare classic, which are what this app currently has alarms to prevent using because they are commonly mistaken for compatible. Adding a format that uses those cards is a pretty big invitation for a mess.

closed 2022-11-19 by AbandonedCart

Squall-Leonhart, 2024-09-03:
The topaz implementation can probably be worked out refering to this driver https://github.com/mdeverdelhan/Topaz512-driver

AbandonedCart, 2024-09-03:
Alright. If you happen to work one out, I will reopen this to test it.