HiddenRamblings/TagMo #587: [Feature]: Need help with Splatoon

AbandonedCart, 2022-10-07:
### Summarize Feature

So the whole TOS no sharing files thing means I don't want your bin files, but instead have added a new feature. What I do need is those with Splatoon amiibo that have game data to capture the Hex code view of any one(s) they have handy. The best results would be saving this file, then making a single change using the game and saving it again with the new data.

* Click an item in the browser
* Click the three dots on the right
* Click "View Hex Code"
* Click the disk icon in the top right

In `/sdcard/Download/TagMo/HexCode` should be a new image file. This is what I need.

By comparing this data to the data of a blank and data from a previous instance, it should be possible to start figuring out the regions to modify game data.

### Usage Confirmation

- [X] I confirm this is a request for a new feature, not an issue

Brickleberry19, 2022-10-07:
[splatoon_HexCode_With_saves_from_splatoon_2_.zip](https://github.com/HiddenRamblings/TagMo/files/9734326/splatoon_HexCode_With_saves_from_splatoon_2_.zip)

AbandonedCart, 2022-10-07:
Awesome, thanks!

AbandonedCart, 2022-10-07:
@Brickleberry19 If you have a chance, there is a new build of TagMo coming up that fixes capturing the whole hex code. I was worried about the width and forgot to check it captured the full length (sorry about that). I only need one figure, though. If my suspicion is correct, I think that can make it possible to write "registration" data for splatoon 2. Thanks again and no worries if you can't.

Brickleberry19, 2022-10-07:
[image]

Brickleberry19, 2022-10-07:
[image]

Brickleberry19, 2022-10-07:
let me know if you need anything else

AbandonedCart, 2022-10-07:
That should be plenty. Thank you!

AbandonedCart, 2022-10-07:
Everything up to 01B0 is identical in all the files, which (according to the names) should be all the different characters. That means whatever data the game writes up to that point is generic and can be generated. The last 4 bits are the part that apply to that character. I have no idea yet what the generic data means specifically, but it is enough to start making fake data that should transfer to Splatoon 3 as a "used" character.

AbandonedCart, 2022-10-07:
https://ocr.space/ ->
```
String generic = "01 01 00 00" +
            "00 00 00 00 42 C0 7B 0A 36 FA 8F C1 48 F5 09 60" +
            "C3 8E C5 52 EE E2 9A CE 27 71 5C 65 40 5B C3 F1" +
            "45 BF C6 3E 72 68 A8 9E DB 41 F4 25 8D 3A 4B 75" +
            "FC AA 74 19 24 EF 4F 3A 73 EC C1 CF C5 84 82 DB" +
            "73 8A 5F 63 A8 14 54 4C F1 91 B5 05 E4 23 B7 50" +
            "93 F5 85 6F 9A A5 C1 01 EB 51 33 1B 8B 54 37 1B" +
            "A8 7A 7C 44 6C A6 96 85 5C B3 E9 8B E4 DO FE 6E" +
            "AF 60 56 DA 2C 66 20 38 88 EO CF C5 3E A8 6E FA" +
            "2D A0 08 38 F4 DF 62 06 BO 6E BA FA 5C AE 84 68" +
            "BD 91 70 B1 7F 34 9E 8A B9 94 20 21 FF D4 08 B6" +
            "9E F3 63 B2 BO AE DD A2 5D 09 CC 9B 9B D1 D5 3E" +
            "88 BO 6D C2 AF 9D 66 7E 20 24 77 3B 88 7C AF 7F" +
            "88 9C B6 3D 7E CD BC 22 41 79 5B BA 9A D5 70 18";
```

AbandonedCart, 2022-10-08:
Deleted the old content with partial output (my fault). Now if someone else with a separate copy of Splatoon 2 and their own set of amiibo posts, it should be possible to confirm what data is actually universal. For now, it's set up to inject the chunk of app data that matched. All of it is strictly inside the game region, so it doesn't identify the player or console. I just don't know if any of it is specific progress in the game.

AbandonedCart, 2022-10-11:
And we have confirmation!

closed 2022-10-11 by AbandonedCart