HiddenRamblings/TagMo #588: Error properly connecting to Flask

CoatsyJnr, 2022-10-09:
I can connect and see what amiibo are on the flask, but I get the following error, along with Connecting to Bluup Flask stuck at the bottom.

[image]

TagMo - #46d88def
Android TIRAMISU (13)
11.25 GB RAM
Installed by release

java.lang.NumberFormatException: For input string: "h4qd"
	at java.lang.Long.parseLong(Long.java:736)
	at java.lang.Long.parseLong(Long.java:861)
	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment.getAmiiboByTail(FlaskSlotFragment.java:604)
	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment.-$$Nest$mgetAmiiboByTail(Unknown Source:0)
	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment$1$1.lambda$onFlaskListRetrieved$2(FlaskSlotFragment.java:170)
	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment$1$1.$r8$lambda$EJMZdF6UMQp2kF7Lls2VFEvr2Pw(Unknown Source:0)
	at com.hiddenramblings.tagmo.browser.FlaskSlotFragment$1$1$$ExternalSyntheticLambda5.run(Unknown Source:4)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
	at java.lang.Thread.run(Thread.java:1012)

AbandonedCart, 2022-10-09:
Extension of https://github.com/HiddenRamblings/TagMo/issues/571

closed 2022-10-09 by AbandonedCart

AbandonedCart, 2022-10-09:
Keep in mind that it will require an update to see the fix.