HiddenRamblings/TagMo #591: app runs with 20 fps when using other display options than icon tile grid

schicho, 2022-10-16:
### Issue Description

The app offers several different layout styles.

However, when using any other layout style than "icon tile grid" the whole app runs with what appears to be 20 FPS or worse. It's especially noticable when scrolling or opening the sidebar.

I can attach a video later if needed and in case of not being reproducable.

### Logcat Output

```shell
- since this is not crash related log won't be of any use, I believe.
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2022-10-17:
I can't really diagnose the issue because you felt a log wasn't necessary.

schicho, 2022-10-17:
TagMo 3.2.6 (GitHub Release) #cd9cbe4
Android R (11) - 7.34 GB RAM

--------- beginning of main
10-17 09:35:04.778  1654  1654 W linker  : Warning: "/data/app/~~4PpRzJymBAiNRVTT5RZCdw==/com.google.android.trichromelibrary_519513634-TxjdHedJ5Y7yHQ6X9BDzuQ==/base.apk!/lib/arm64-v8a/libmonochrome_64.so" unused DT entry: unknown processor-specific (type 0x70000001 arg 0x0) (ignoring)
10-17 09:35:05.245  1654  2351 E Parcel  : Reading a NULL string not supported here.
10-17 09:35:05.254  1654  2351 E Parcel  : Reading a NULL string not supported here.
10-17 09:35:06.043  1654  2373 E Parcel  : Reading a NULL string not supported here.
10-17 09:35:33.685  1654  1654 W BottomSheetBehavior: Cannot set state: 5
10-17 09:35:33.804  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:33.805  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:33.833  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:33.840  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:33.866  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:33.866  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:33.958  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:33.959  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.057  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.058  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.159  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.272  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.650  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.753  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.853  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.854  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.954  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:34.955  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.067  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.067  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.167  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.167  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.283  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.383  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.473  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.474  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.573  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.670  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.814  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.815  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.914  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:35.995  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.105  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.106  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.222  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.308  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.429  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.429  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.496  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.497  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:36.609  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.203  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.351  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.351  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.423  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.423  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.506  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.507  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.634  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.634  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.688  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.689  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.806  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.880  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:37.959  1654  1654 W Choreographer: Frame time is 0.047502  now = 2975997857081057  timestampNanos = 2975997857128559 ms in the future!  Check that graphics HAL is generating vsync timestamps using the correct timebase.
[... cut ...]
10-17 09:35:41.214  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:41.616  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:41.723  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:42.203  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:42.299  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb
10-17 09:35:42.299  1654  1685 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:626: Null anb

AbandonedCart, 2022-10-17:
Are you running any games or streaming apps in the background at the same time? 
`GuiExtAuxCheckAuxPath` is an issue often associated with one or both of those that results in low framerate. I am not entirely sure if it is something TagMo alone would be able to address.

schicho, 2022-10-17:
Nope. TagMo is the only app running.
As stated, it works completely fine when using the icon tile grid style.

I can also switch back and forth between the styles. It's only laggy when using the other styles.

AbandonedCart, 2022-10-17:
It's odd because the only difference between the tile grid and the other views are the visibility of text. *That* should actually be the slowest view.

AbandonedCart, 2022-10-17:
Other styles includes text only, correct? It sounds like it does, which means the issue must be somewhere in the text.

schicho, 2022-10-18:
That does not change anything in my tests.

AbandonedCart, 2022-10-18:
Optimized the views and how they are loaded, but I doubt it will have any impact. Most of what was done applied to images and the view as a whole, but the issue is only with loading text.

AbandonedCart, 2022-10-26:
Is this still the case or did all of the secondary optimization fix this one?

AbandonedCart, 2022-10-26:
I haven't seen it on either phone and I also tried with a few emulators, but they are typically a false positive.

AbandonedCart, 2022-10-31:
I have a feeling I know one of the causes for this and it dates back to long before I jumped into TagMo, which is also the reason I never really considered it. Working on an update to fix it now.

AbandonedCart, 2022-11-01:
The update with those changes should have hit Google Play yesterday.

schicho, 2022-11-01:
Used the latest github release. Still did not change a thing.

<spoiler>

TagMo 3.3.4 (GitHub Release) #0dc6446
Android S (12) - 7.31 GB RAM

--------- beginning of main
11-01 15:25:36.698 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:36.716 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:36.719 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:36.721 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:36.721 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:36.812 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:36.812 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:36.923 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:36.923 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.036 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.036 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.136 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.136 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.226 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.226 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.327 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.327 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.427 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.427 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.519 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.520 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.630 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.630 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.740 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.740 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.833 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.833 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.932 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:37.932 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:38.043 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:38.043 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:38.206 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:38.211 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:38.224 19598 19598 W View    : requestLayout() improperly called by androidx.appcompat.widget.AppCompatTextView{ba826b1 V.ED..... ......ID 216,43-402,124} during layout: running second layout pass
11-01 15:25:38.224 19598 19598 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{489796 VFED.V... ......ID 0,1999-1080,1999 #7f09010c app:id/foomiibo_list} during layout: running second layout pass
11-01 15:25:38.224 19598 19598 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{11c6f17 VFED.V... ......ID 0,0-1080,1999 #7f09005f app:id/amiibos_list} during layout: running second layout pass
11-01 15:25:40.224 19598 19915 E BLASTBufferQueue: BLASTBufferItemConsumer::onDisconnect()
11-01 15:25:41.901 19598 19915 E BLASTBufferQueue: BLASTBufferItemConsumer::onDisconnect()
11-01 15:25:43.769 19598 19598 W BottomSheetBehavior: Cannot set state: 5
11-01 15:25:44.002 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:44.010 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:44.015 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:44.019 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:44.022 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:44.024 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:44.024 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:44.024 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:44.024 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:44.024 19598 19915 E GPUAUX  : [AUX]GuiExtAuxCheckAuxPath:671: Null anb
11-01 15:25:44.025 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
11-01 15:25:44.029 19598 20109 E .tagmo.eightbi: == MALI DEBUG ===BAD ALLOC from gles_texture_egl_image_get_2d_template err is 0
[... cut ...]
11-01 15:25:45.206 19598 19598 W View    : requestLayout() improperly called by android.widget.LinearLayout{5f742d5 V.E..V... ........ 24,136-1020,177 #7f09011a app:id/group2} during second layout pass: posting in next frame
11-01 15:25:45.206 19598 19598 W View    : requestLayout() improperly called by android.widget.LinearLayout{6628aea V.E..V... ........ 24,95-1020,136 #7f090119 app:id/group1} during second layout pass: posting in next frame
11-01 15:25:45.206 19598 19598 W View    : requestLayout() improperly called by android.widget.LinearLayout{944a1db V.E..V... ........ 24,136-1020,177 #7f09011a app:id/group2} during second layout pass: posting in next frame

</spoiler>

AbandonedCart, 2022-11-01:
What type of device is it? I'm beginning to suspect it's an issue with the manufacturer.

schicho, 2022-11-01:
It's a OnePlus Nord 2. Also I just received the update to Android 12 and that did not change the issue.

AbandonedCart, 2022-11-01:
Looks like there is a trend of reports for any sort of swipe causing issues with OnePlus that dates back as far as 2020.

AbandonedCart, 2022-11-01:
I'm going to try disabling hardware acceleration for OnePlus and then adding an option to enable it. If the issue is broken GPU drivers, that would solve it. Normally I would say it comes at the cost of a little speed, but that gets cancelled out in this case. Having the option to enable hardware acceleration again will also prevent it from being destructive for those that didn't need it. If I remember right, most devices that reported similar issues were OnePlus and Redmi. Ironically, both are Chinese manufacturers. Samsung is Korean and Motorola is US.

AbandonedCart, 2022-11-10:
Haven't heard anything further about this, so I'm closing it for now. If this proves to still be an issue, we can revisit it later.

closed 2022-11-10 by AbandonedCart