HiddenRamblings/TagMo #598: [Bug]: Error writing 44 bins on latest version

Sora1995, 2022-10-23:
TagMo 3.2.9 (Google Play Release) #96a1e428
Android TIRAMISU (13) - 5.33 GB RAM

--------- beginning of crash
10-22 14:45:31.886 23811 23811 E AndroidRuntime: FATAL EXCEPTION: main
10-22 14:45:31.886 23811 23811 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 23811
10-22 14:45:31.886 23811 23811 E AndroidRuntime: java.lang.IllegalStateException: Fragment SettingsFragment{5d214fe} (4c2acc3f-e72a-47fb-ae3c-1a061efb6178) not attached to an activity.
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at androidx.fragment.app.Fragment.requireActivity(Fragment.java:995)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.buildSnackbar(SettingsFragment.java:624)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.lambda$parseCommitDate$35(SettingsFragment.java:642)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.$r8$lambda$2GSRyi1JUqwrtXcQaATbTveg1ZU(Unknown Source:0)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment$$ExternalSyntheticLambda34.run(Unknown Source:4)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:942)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:201)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:288)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7872)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
--------- beginning of main
10-23 03:17:32.780  5971  6385 E ViewRootImpl: Accessibility content change on non-UI thread. Future Android versions will throw an exception.
10-23 03:17:32.780  5971  6385 E ViewRootImpl: android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewRootImpl$SendWindowContentChangedAccessibilityEvent.runOrPost(ViewRootImpl.java:10646)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewRootImpl.postSendWindowContentChangedCallback(ViewRootImpl.java:9486)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewRootImpl.notifySubtreeAccessibilityStateChanged(ViewRootImpl.java:9666)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewGroup.notifySubtreeAccessibilityStateChanged(ViewGroup.java:3848)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewGroup.notifySubtreeAccessibilityStateChanged(ViewGroup.java:3848)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewGroup.notifySubtreeAccessibilityStateChanged(ViewGroup.java:3848)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewGroup.notifySubtreeAccessibilityStateChanged(ViewGroup.java:3848)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewGroup.notifySubtreeAccessibilityStateChanged(ViewGroup.java:3848)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewGroup.notifySubtreeAccessibilityStateChanged(ViewGroup.java:3848)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.ViewGroup.notifySubtreeAccessibilityStateChanged(ViewGroup.java:3848)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.view.View.notifyViewAccessibilityStateChangedIfNeeded(View.java:14367)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.widget.TextView.setText(TextView.java:6640)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.widget.TextView.setText(TextView.java:6454)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at android.widget.TextView.setText(TextView.java:6406)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:401)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda0.run(Unknown Source:4)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
10-23 03:17:32.780  5971  6385 E ViewRootImpl: 	at java.lang.Thread.run(Thread.java:1012)
10-23 03:17:32.801  5971  6385 W NullPointerException: java.lang.NullPointerException
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at java.io.FileInputStream.<init>(FileInputStream.java:152)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readTagFile(TagReader.java:52)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at com.hiddenramblings.tagmo.nfctech.TagArray.getValidatedFile(TagArray.java:274)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:405)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda0.run(Unknown Source:4)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
10-23 03:17:32.801  5971  6385 W NullPointerException: 	at java.lang.Thread.run(Thread.java:1012)
10-23 03:17:36.421  5971  5990 W Parcel  : Expecting binder but got null!
10-23 03:17:38.721  5971  6485 W Exception: java.lang.Exception: Data must start with 0x04.
10-23 03:17:38.721  5971  6485 W Exception: 	at com.hiddenramblings.tagmo.nfctech.TagArray.validateData(TagArray.java:179)
10-23 03:17:38.721  5971  6485 W Exception: 	at com.hiddenramblings.tagmo.nfctech.TagArray.getValidatedData(TagArray.java:267)
10-23 03:17:38.721  5971  6485 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:464)
10-23 03:17:38.721  5971  6485 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity.lambda$onNewIntent$0(NfcActivity.java:249)
10-23 03:17:38.721  5971  6485 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity.$r8$lambda$vP5NXNBqg2kDV8eg_wytd2g-zlQ(Unknown Source:0)
10-23 03:17:38.721  5971  6485 W Exception: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda0.run(Unknown Source:4)
10-23 03:17:38.721  5971  6485 W Exception: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
10-23 03:17:38.721  5971  6485 W Exception: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
10-23 03:17:38.721  5971  6485 W Exception: 	at java.lang.Thread.run(Thread.java:1012)
10-23 03:17:42.906  5971  5971 W View    : requestLayout() improperly called by androidx.appcompat.widget.AppCompatTextView{ca35dbd V.ED..... ......ID 189,38-353,109} during layout: running second layout pass
10-23 03:17:42.906  5971  5971 W View    : requestLayout() improperly called by myinnos.indexfastscrollrecycler.IndexFastScrollRecyclerView{591b97a VFED.V... ......ID 0,1818-1080,1925 #7f090111 app:id/foomiibo_list} during layout: running second layout pass
[... cut ...]
10-23 03:21:09.126  6648  7679 W NullPointerException: 	at com.hiddenramblings.tagmo.NfcActivity$$ExternalSyntheticLambda0.run(Unknown Source:4)
10-23 03:21:09.126  6648  7679 W NullPointerException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
10-23 03:21:09.126  6648  7679 W NullPointerException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
10-23 03:21:09.126  6648  7679 W NullPointerException: 	at java.lang.Thread.run(Thread.java:1012)
10-23 03:21:13.580  6648  6665 W Parcel  : Expecting binder but got null!

AbandonedCart, 2022-10-23:
Next time, just use the form. It does all the formatting for you.

AbandonedCart, 2022-10-23:
```
10-23 03:21:09.105  6648  7679 E ViewRootImpl: Accessibility content change on non-UI thread. Future Android versions will throw an exception.
10-23 03:21:09.105  6648  7679 E ViewRootImpl: android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
```

Fixed by 93ccf8c975a77981b2e6a04e881bb8bd68e518d8

AbandonedCart, 2022-10-23:
```
10-23 03:21:09.126  6648  7679 W NullPointerException: java.lang.NullPointerException
```

Fixed by 662c8a7445a2cc24ed3d9def27062d8d17587b66

AbandonedCart, 2022-10-23:
```
10-23 03:18:02.715  6648  6648 W BootstrapClass: reflect bootstrap failed:
10-23 03:18:02.715  6648  6648 W BootstrapClass: java.lang.reflect.InvocationTargetException
```

Fixed by d05fe1d4280a13effd67deeb8dd01cb2fcd3f1ae

AbandonedCart, 2022-10-23:
```
10-23 03:18:05.654  6648  6793 W TagLostException: android.nfc.TagLostException: Tag was lost.
```

That one is removing the tag from the NFC field, so there is nothing to address.

closed 2022-10-23 by AbandonedCart