HiddenRamblings/TagMo #600: [Issue]: Still unable to write 44 bins on latest version

Sora1995, 2022-10-23:
### Issue Description

I am using the latest version. The N2 is on a table that I lay my phone on top of. I have used this method every time to add amiibo. I can mass delete using this exact same method but I cannot mass write. I don't know if it's my Android version or my phone or something else. I can provide video evidence if you need to show that I'm not moving it and it's still erroring

### Logcat Output

```shell
TagMo 3.3.0 (Google Play Release) #1f559a46
Android TIRAMISU (13) - 5.33 GB RAM

--------- beginning of crash
10-22 14:45:31.886 23811 23811 E AndroidRuntime: FATAL EXCEPTION: main
10-22 14:45:31.886 23811 23811 E AndroidRuntime: Process: com.hiddenramblings.tagmo.eightbit, PID: 23811
10-22 14:45:31.886 23811 23811 E AndroidRuntime: java.lang.IllegalStateException: Fragment SettingsFragment{5d214fe} (4c2acc3f-e72a-47fb-ae3c-1a061efb6178) not attached to an activity.
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at androidx.fragment.app.Fragment.requireActivity(Fragment.java:995)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.buildSnackbar(SettingsFragment.java:624)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.lambda$parseCommitDate$35(SettingsFragment.java:642)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.$r8$lambda$2GSRyi1JUqwrtXcQaATbTveg1ZU(Unknown Source:0)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment$$ExternalSyntheticLambda34.run(Unknown Source:4)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:942)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:201)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:288)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7872)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
10-22 14:45:31.886 23811 23811 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
--------- beginning of main
10-23 12:15:10.108 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:15:12.324 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:15:29.758 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:15:31.276 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:15:43.291 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:15:43.291 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:15:50.600 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:15:50.600 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:07.663 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:16:07.690 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:16:36.739 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:16:37.343 24030 24053 W Parcel  : Expecting binder but got null!
10-23 12:16:50.259 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:50.259 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:50.578 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:50.578 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:50.846 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:50.846 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:51.162 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:51.162 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:51.909 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:51.909 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:54.176 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:16:54.176 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:17:14.663 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:17:14.663 24030 24030 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
10-23 12:17:31.861 24030 24041 W .tagmo.eightbit: Cleared Reference was only reachable from finalizer (only reported once)
10-23 12:17:37.849 24030 24053 W Parcel  : Expecting binder but got null!
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2022-10-23:
Errors fixed by 3a1c4cb8c136fcca696f5fafe97aa264859badf0 and 86dc5694751239650143bacf7e53456f9e21c167, but neither would cause a failed write.

Sora1995, 2022-10-23:
Truthfully I'm not sure what is going on. I haven't had other people try it to see if it's just me

AbandonedCart, 2022-10-23:
NFC is something that can only be tested on a live device. If one device or Android version is different, it's not something that can really be predicted until it breaks.

Sora1995, 2022-10-23:
Well, if there's anything I can do to help us figure this out, let me know

AbandonedCart, 2022-10-23:
It's trying to use files, but Google Play is documents. The fix will take a minute (plus however long Google spends in review)

AbandonedCart, 2022-10-24:
It's taking a bit longer than expected because the file is getting dropped along the way. It doesn't cause an error, so there isn't much to track. It is being addressed, though.

closed 2022-10-25 by AbandonedCart

AbandonedCart, 2022-10-26:
Went ahead and pushed this before all the UI stuff was finished because ugly and working is better than broken.