HiddenRamblings/TagMo #623: [Issue]: after update, when i go to n2elite page，it crush

richardc02009, 2022-11-13:
### Issue Description

after update, it said that no crypto key，then i choose no file and back，it said that crypto key loaded，but when i go to n2elite page it will crush

### Logcat Output

```shell
TagMo 3.4.2 (Google Play Release) #2ee178c1
Android S (12) - 9.95 GB RAM

java.lang.IllegalStateException: Fragment already added
	at androidx.fragment.app.Fragment.setInitialSavedState(Fragment.java:828)
	at androidx.viewpager2.adapter.FragmentStateAdapter.ensureFragment(FragmentStateAdapter.java:269)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onBindViewHolder(FragmentStateAdapter.java:175)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onBindViewHolder(FragmentStateAdapter.java:67)
	at androidx.recyclerview.widget.RecyclerView$Adapter.onBindViewHolder(RecyclerView.java:7254)
	at androidx.recyclerview.widget.RecyclerView$Adapter.bindViewHolder(RecyclerView.java:7337)
	at androidx.recyclerview.widget.RecyclerView$Recycler.tryBindViewHolderByDeadline(RecyclerView.java:6194)
	at androidx.recyclerview.widget.RecyclerView$Recycler.tryGetViewHolderForPositionByDeadline(RecyclerView.java:6460)
	at androidx.recyclerview.widget.RecyclerView$Recycler.getViewForPosition(RecyclerView.java:6300)
	at androidx.recyclerview.widget.RecyclerView$Recycler.getViewForPosition(RecyclerView.java:6296)
	at androidx.recyclerview.widget.LinearLayoutManager$LayoutState.next(LinearLayoutManager.java:2330)
	at androidx.recyclerview.widget.LinearLayoutManager.layoutChunk(LinearLayoutManager.java:1631)
	at androidx.recyclerview.widget.LinearLayoutManager.fill(LinearLayoutManager.java:1591)
	at androidx.recyclerview.widget.LinearLayoutManager.scrollBy(LinearLayoutManager.java:1395)
	at androidx.recyclerview.widget.LinearLayoutManager.scrollHorizontallyBy(LinearLayoutManager.java:1124)
	at androidx.recyclerview.widget.RecyclerView.scrollStep(RecyclerView.java:1969)
	at androidx.recyclerview.widget.RecyclerView$SmoothScroller.onAnimation(RecyclerView.java:12316)
	at androidx.recyclerview.widget.RecyclerView$ViewFlinger.run(RecyclerView.java:5564)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1108)
	at android.view.Choreographer.doCallbacks(Choreographer.java:866)
	at android.view.Choreographer.doFrame(Choreographer.java:792)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1092)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8751)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2022-11-13:
This is one of those errors where the "cause" isn't what caused it. Looks like it's trying to make the page twice, but that doesn't say why. I'll look into it.

AbandonedCart, 2022-11-14:
Added a bunch of double checking for this that should keep it from happening. It will be included in the next Google Play update.

closed 2022-11-14 by AbandonedCart