HiddenRamblings/TagMo #624: [Issue]: Tagmo won't start anymore

Saliakooh, 2022-11-14:
### Issue Description

I've used Tagmo successfully in the past, but haven't for the last few weeks. Now I tried to open the app and a weird issue came up.

Upon starting Tagmo, my phone's "share" menu will come up (as if I'm trying to share content to Telegram, Gmail, Discord etc.). It won't disappear no matter what I do/go away and immediately come back up.

Of course I tried the usual things like closing and reopening the app, reinstalling, restarting the phone. When I reinstall, I can open Tagmo once, then when I try to download the necessary keys, the issue starts again.

I downloaded from Google Play, if that helps. Does anyone else experience the same issue?

Thanks in advance.

Edit: I tried the GitHub version and the problem still occurs. Using Android 12.

### Logcat Output

```shell
TagMo 3.4.2 (Google Play Release) #2ee178c1
Android S (12) - 7.38 GB RAM

java.lang.SecurityException: Permission Denial: opening provider com.android.externalstorage.ExternalStorageProvider from ProcessRecord{fd65f87 16345:com.hiddenramblings.tagmo.eightbit/u0a332} (pid=16345, uid=10332) requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
	at android.os.Parcel.createExceptionOrNull(Parcel.java:2472)
	at android.os.Parcel.createException(Parcel.java:2456)
	at android.os.Parcel.readException(Parcel.java:2439)
	at android.os.Parcel.readException(Parcel.java:2381)
	at android.app.IActivityManager$Stub$Proxy.getContentProvider(IActivityManager.java:5985)
	at android.app.ActivityThread.acquireProvider(ActivityThread.java:7321)
	at android.app.ContextImpl$ApplicationContentResolver.acquireUnstableProvider(ContextImpl.java:3383)
	at android.content.ContentResolver.acquireUnstableProvider(ContentResolver.java:2560)
	at android.content.ContentResolver.query(ContentResolver.java:1219)
	at android.content.ContentResolver.query(ContentResolver.java:1161)
	at android.content.ContentResolver.query(ContentResolver.java:1117)
	at com.hiddenramblings.tagmo.amiibo.AmiiboDocument.listFiles(AmiiboDocument.java:90)
	at com.hiddenramblings.tagmo.amiibo.AmiiboDocument.listFiles(AmiiboDocument.java:66)
	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiiboDocuments(AmiiboManager.java:383)
	at com.hiddenramblings.tagmo.browser.BrowserActivity.lambda$loadAmiiboDocuments$39(BrowserActivity.java:1831)
	at com.hiddenramblings.tagmo.browser.BrowserActivity.$r8$lambda$I_9J5GuQIX3MODKF9w3mcj-fRno(Unknown Source:0)
	at com.hiddenramblings.tagmo.browser.BrowserActivity$$ExternalSyntheticLambda50.run(Unknown Source:6)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
	at java.lang.Thread.run(Thread.java:1012)
```

### Usage Requirements

- [X] I have read and understood the usage requirements for TagMo

AbandonedCart, 2022-11-14:
The share menu is the app trying to share the log. I added a "Toast" message that better explains this, since it seems to be causing some confusion. It's a failsafe for situations, such as yours, when the app can't start all the way up to allow you to report the error.

A handful of others have experienced the same issue, but the exact reason behind it is unknown. It is saying that the app does not have permission to access the files inside the folder you selected as your root folder. Ironically, though, the only way it would know the folder to use would be through getting the permission it claims to be missing.

AbandonedCart, 2022-11-14:
If you have the choice, the GitHub version is better overall. Honestly, using the "all files" permission would avoid this issue entirely and that is something Google Play wouldn't approve because the app loaded without the permission (despite how stupid a reason that is). It also gets updated much faster. The reason a Google Play version exists is for those that struggle with installing from GitHub, have paranoia about granting permission for files or installing updates, and needed that extra reassurance that the published source was not, in fact, malware-ridden piracy (again, a silly theory).

AbandonedCart, 2022-11-14:
https://github.com/HiddenRamblings/TagMo/commit/6d09ee54ff245e0ed724ec1fd3e4d96815ea2c36 should fix this one.

closed 2022-11-14 by AbandonedCart