HiddenRamblings/TagMo #644: [Logcat]: TagMo #3ecf7cd3

LogcatGitHub, 2022-11-29:
*Submitted by pyj08178@*****.com*

Description:
---

TagMo 3.4.4 (Google Play Release) #3ecf7cd3
Android TIRAMISU (13) - 5.44 GB RAM

java.lang.RuntimeException: Failure delivering result ResultInfo{who=null, request=680494097, result=-1, data=Intent { act=com.hiddenramblings.tagmo.eightbit.NFC_SCANNED (has extras) }} to activity {com.hiddenramblings.tagmo.eightbit/com.hiddenramblings.tagmo.NFCIntentFilter}: java.lang.NullPointerException: Attempt to read from field 'java.util.HashMap com.hiddenramblings.tagmo.amiibo.AmiiboManager.amiiboSeries' on a null object reference in method 'com.hiddenramblings.tagmo.amiibo.AmiiboSeries com.hiddenramblings.tagmo.amiibo.Amiibo.getAmiiboSeries()'
	at android.app.ActivityThread.deliverResults(ActivityThread.java:5347)
	at android.app.ActivityThread.handleSendResult(ActivityThread.java:5386)
	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67)
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2308)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7898)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
Caused by: java.lang.NullPointerException: Attempt to read from field 'java.util.HashMap com.hiddenramblings.tagmo.amiibo.AmiiboManager.amiiboSeries' on a null object reference in method 'com.hiddenramblings.tagmo.amiibo.AmiiboSeries com.hiddenramblings.tagmo.amiibo.Amiibo.getAmiiboSeries()'
	at com.hiddenramblings.tagmo.amiibo.Amiibo.getAmiiboSeries(Amiibo.java:92)
	at com.hiddenramblings.tagmo.browser.EliteBankFragment.updateAmiiboView(EliteBankFragment.java:833)
	at com.hiddenramblings.tagmo.browser.EliteBankFragment.lambda$new$14(EliteBankFragment.java:520)
	at com.hiddenramblings.tagmo.browser.EliteBankFragment.$r8$lambda$k7V-oAJpvA0CvC7RRtcV-_wQ9To(Unknown Source:0)
	at com.hiddenramblings.tagmo.browser.EliteBankFragment$$ExternalSyntheticLambda9.onActivityResult(Unknown Source:4)
	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:418)
	at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:375)
	at androidx.activity.ComponentActivity.onActivityResult(ComponentActivity.java:793)
	at androidx.fragment.app.FragmentActivity.onActivityResult(FragmentActivity.java:152)
	at android.app.Activity.dispatchActivityResult(Activity.java:8613)
	at android.app.ActivityThread.deliverResults(ActivityThread.java:5340)
	... 13 more

Device info:
---
<table>
<tr><td>App version</td><td>3.4.4</td></tr>
<tr><td>App version code</td><td>3440</td></tr>
<tr><td>Android build version</td><td>9080065</td></tr>
<tr><td>Android release version</td><td>13</td></tr>
<tr><td>Android SDK version</td><td>33</td></tr>
<tr><td>Android build ID</td><td>TP1A.221105.002</td></tr>
<tr><td>Device brand</td><td>google</td></tr>
<tr><td>Device manufacturer</td><td>Google</td></tr>
<tr><td>Device name</td><td>bluejay</td></tr>
<tr><td>Device model</td><td>Pixel 6a</td></tr>
<tr><td>Device product name</td><td>bluejay</td></tr>
<tr><td>Device hardware name</td><td>bluejay</td></tr>
<tr><td>ABIs</td><td>[arm64-v8a, armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (32bit)</td><td>[armeabi-v7a, armeabi]</td></tr>
<tr><td>ABIs (64bit)</td><td>[arm64-v8a]</td></tr>
</table>

Logcat:
---
```
TagMo 3.4.4 (Google Play Release) #3ecf7cd3
Android TIRAMISU (13) - 5.44 GB RAM

--------- beginning of main
11-29 17:42:08.378  9673  9673 W libc    : Access denied finding property "ro.vendor.oplus.market.name"
11-29 17:42:08.378  9673  9673 W libc    : Access denied finding property "ro.vendor.oplus.market.name"
11-29 17:42:08.378  9673  9673 W libc    : Access denied finding property "ro.vendor.oplus.market.name"
11-29 17:42:08.424  9673  9673 W BottomSheetBehavior: Cannot set state: 5
11-29 17:42:08.712  9673  9673 W View    : requestLayout() improperly called by androidx.appcompat.widget.AppCompatTextView{5771636 V.ED..... ......ID 189,38-374,109} during layout: running second layout pass
11-29 17:42:08.712  9673  9673 W View    : requestLayout() improperly called by androidx.recyclerview.widget.RecyclerView{3d1937 VFED.V... ......ID 0,222-1080,1993 #7f0900e5 app:id/elite_content} during layout: running second layout pass
[... 174 lines cut ...]
11-29 17:42:41.346 10024 10078 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readBankTitle(TagReader.java:90)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readTagTitles(TagReader.java:98)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at com.hiddenramblings.tagmo.nfctech.ScanTag.onTagDiscovered(ScanTag.java:87)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.lambda$onNewIntent$68(BrowserActivity.java:2755)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity.$r8$lambda$UKGWWKW8pM77me2SKr_RizzHR-c(Unknown Source:0)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at com.hiddenramblings.tagmo.browser.BrowserActivity$$ExternalSyntheticLambda19.run(Unknown Source:4)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
11-29 17:42:41.346 10024 10078 W TagLostException: 	at java.lang.Thread.run(Thread.java:1012)
11-29 17:42:41.346 10024 10078 W TagReader: Failed to parse banks!
11-29 17:42:42.555 10024 10034 W JavaBinder: BinderProxy is being destroyed but the application did not call unlinkToDeath to unlink all of its death recipients beforehand.  Releasing leaked death recipient: com.google.android.play.core.internal.zzai
11-29 17:42:51.142 10024 10024 W AssistStructure: Skipping window with dettached view: PopupWindow:eef3d94
11-29 17:42:54.348 10024 10024 W AssistStructure: Skipping window with dettached view: PopupWindow:eb3f64
11-29 17:43:00.337 10024 10024 W AssistStructure: Skipping window with dettached view: PopupWindow:7ab6ffb
11-29 17:43:02.072 10024 10024 W AssistStructure: Skipping window with dettached view: PopupWindow:5e569e1
11-29 17:43:04.027 10024 10024 W AssistStructure: Skipping window with dettached view: PopupWindow:fddc6b7
11-29 17:43:13.085 10024 10024 W androidtc: TextClassifier called on main thread
11-29 17:43:13.158 10024 10024 W androidtc: TextClassifier called on main thread
11-29 17:43:14.035 10024 10024 W androidtc: TextClassifier called on main thread
11-29 17:43:23.530 10024 10024 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
11-29 17:43:23.530 10024 10024 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
11-29 17:43:35.975 10024 10024 W AssistStructure: Skipping window with dettached view: PopupWindow:1af1e52
11-29 17:43:37.351 10024 10024 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
11-29 17:43:37.351 10024 10024 E SpannableStringBuilder: SPAN_EXCLUSIVE_EXCLUSIVE spans cannot have a zero length
```

AbandonedCart, 2022-11-29:
Looks like the N2 is being pulled away before it can finish scanning.

DirectorPoh, 2022-11-29:
> Looks like the N2 is being pulled away before it can finish scanning.

After erasing an amiibo the app seems to refuse to show anything after scanning the n2elite. But restarting the app works

DirectorPoh, 2022-11-29:
> Deleting the active bank is buggy with the N2 Elite, but that is how they designed the hardware.

I see thanks. Sorry for the terrible logs

AbandonedCart, 2022-11-29:
No worries. I'll look into trying to catch the issue without needing to reboot. Currently, it only catches it on the first run and that's likely why it has to be restarted.

closed 2022-11-29 by AbandonedCart